ipxe/ipxe #901: make bin-x86_64-efi/ipxe.efi  errors out

Unikat2000, 2023-02-25:
Hi, I'm unable to build using  **make bin-x86_64-efi/ipxe.efi** . Has some an idea whats's going wrong?

>   [AR] bin-x86_64-efi/blib.a
ar: creating bin-x86_64-efi/blib.a
  [VERSION] bin-x86_64-efi/version.ipxe.efi.o
  [LD] bin-x86_64-efi/ipxe.efi.tmp
bin-x86_64-efi/blib.a(librm.o): in function `pm_esp':
(.data.pm_esp+0x0): relocation truncated to fit: R_X86_64_32 against symbol `_estack' defined in .stack section in bin-x86_64-efi/blib.a(stack.o)
bin-x86_64-efi/blib.a(librm.o): in function `real_to_prot':
(.text16.real_to_prot+0x1): relocation truncated to fit: R_X86_64_PC16 against symbol `enable_a20' defined in .text16.early section in bin-x86_64-efi/blib.a(liba20.o)
(.text16.real_to_prot+0x8): relocation truncated to fit: R_X86_64_16 against symbol `rm_ds' defined in .text16.data.rm_ds section in bin-x86_64-efi/blib.a(librm.o)
(.text16.real_to_prot+0x22): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x2e): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x33): relocation truncated to fit: R_X86_64_16 against `.bss16.rm_virt_addrs'
(.text16.real_to_prot+0x39): relocation truncated to fit: R_X86_64_16 against `.data16.gdt'
(.text16.real_to_prot+0x58): relocation truncated to fit: R_X86_64_32 against `.text.real_to_prot'
bin-x86_64-efi/blib.a(librm.o): in function `r2p_pmode':
(.text.real_to_prot+0x10): relocation truncated to fit: R_X86_64_32 against `.data.pm_esp'
(.text.real_to_prot+0x17): relocation truncated to fit: R_X86_64_32 against symbol `idtr32' defined in .data section in bin-x86_64-efi/blib.a(librm_mgmt.o)
(.text.real_to_prot+0x1f): additional relocation overflows omitted from the output
ld: bin-x86_64-efi/blib.a(pxe_entry.o): in function `pxe_segments':
(.text16.data+0x26): undefined reference to `_data16_memsz'
ld: (.text16.data+0x2e): undefined reference to `_data16_memsz'
ld: (.text16.data+0x36): undefined reference to `_text16_memsz'
ld: (.text16.data+0x3e): undefined reference to `_text16_memsz'
ld: bin-x86_64-efi/blib.a(pxe_entry.o): in function `pxenv':
(.text16.data+0x76): undefined reference to `_data16_memsz'
ld: (.text16.data+0x82): undefined reference to `_data16_memsz'
ld: (.text16.data+0x86): undefined reference to `_text16_memsz'
make: *** [Makefile.housekeeping:1225: bin-x86_64-efi/ipxe.efi.tmp] Error 1
rm `bin-x86_64-efi/version.ipxe.efi.o`

[config.zip](https://github.com/ipxe/ipxe/files/10831166/config.zip)

mcb30, 2023-02-25:
You have tried to enable a feature that is specific to the BIOS platform, and cannot ever work on UEFI because it requires 16-bit real-mode code.

Start with the default configuration first, and only try enabling features that you specifically need.

closed 2023-02-25 by mcb30

Unikat2000, 2023-02-25:
thx for the help..... looks like PXE_CMD  was the culprit ...

SuibianP, 2024-06-19:
Would be great if the error message could be more intuitive or the build can fail right away if an impossible permutation is requested.