ipxe/ipxe #903: Zero-length initrd.magic

mcb30, 2023-02-26:
### Discussed in https://github.com/ipxe/ipxe/discussions/902

<div type='discussions-op-text'>

<sup>Originally posted by **cacheman** February 25, 2023</sup>
Hi....

I have run into a strange problem booting a static (i.e. no initrd) Linux 5.x or 6.x EFI kernel using iPXE 1.21.1. My menu entry is below:

`:TestLinux`
`kernel http://aaa.bbb.ccc.ddd/bzImage root=/dev/nfs nfsroot=aaa.bbb.ccc.ddd:/export/PC,vers=3,rsize=8192,wsize=8192,actimeo=9999 ip=dhcp nofb nomodeset nohugeiomap vga=0x317`
`boot || goto failed`
`goto start`

The kernel is one which I have built from source. Kernel versions v5.x and v6.x both work fine with iPXE v1.20.1.....and they have worked just fine for a long time....

However, with latest v1.21.1 the **_first_** time I select this in my PXE menu I get the following error:

`http://aaa.bbb.ccc.ddd/bzImage... ok`
`EFI stub: ERROR: Failed to load initrd: 0x8000000000000001`
`EFI stub: ERROR: efi_main() failed!`
`Could not boot: Error 0x7f048281 (https://ipxe.org/7f048281)`
`Booting failed.`
`Press key to continue`

....then when I 'Press key to continue' I get presented with my PXE menu again. The really strange thing is if I now for a **_second_** time select the same menu entry it gives a different error message, but it **_DOES_** work this time:

`http://aaa.bbb.ccc.ddd/bzImage... ok`
`EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path`
`_......kernel boots with usual Linux kernel boot messages....._`

Is there something new in v1.21.1 that I'm missing or perhaps doing incorrectly?
Or possibly a kernel build option I need to enable/disable?
Or might this be a bug?

Would be grateful for any help with this.

Thanks in advance.

Cacheman...

</div>

mcb30, 2023-02-26:
I've looked in to this.

The problem as observed arises because Linux (in its `efi_load_initrd_dev_path()`) expects the first call to `LoadFile()` to always fail with `EFI_BUFFER_TOO_SMALL`:
```c
	initrd->size = 0;
	status = efi_call_proto(lf2, load_file, dp, false, &initrd->size, NULL);
	if (status != EFI_BUFFER_TOO_SMALL)
		return EFI_LOAD_ERROR;
```

When the initrd is genuinely zero-length, iPXE will return success since the buffer is not too small to hold the (zero-length) file.  This causes Linux to immediately report a spurious failure.

We can change the logic in iPXE's `efi_file_load()` to always return `EFI_BUFFER_TOO_SMALL` if `Buffer==NULL` on entry.  The corner case of a zero-length file is not covered by the UEFI specification, and so either behaviour could be argued to be correct.

Unfortunately that would not fix the problem.  If we return `EFI_BUFFER_TOO_SMALL` and set the file length to zero, then Linux will call the boot services `AllocatePages()` method with a zero length.  In at least the EDK2 implementation, this will return `EFI_OUT_OF_RESOURCES`, and Linux will again report a failure.

This suggests that maybe we should not install the initrd device path handle at all if we have no initrd to offer.  Unfortunately this will fail in yet another corner case: if a previous bootloader has installed an initrd device path handle (e.g. to pass a boot script to iPXE) then we do not want to leave it in place, since then our loaded kernel will end up seeing the wrong initrd content.

The only way to cause Linux's `efi_load_initrd_dev_path()` to fail in a non-fatal way is to cause it to return `EFI_NOT_FOUND`:
```c
	status = efi_load_initrd_dev_path(&initrd, hard_limit);
	if (status == EFI_SUCCESS) {
		efi_info("Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path\n");
	...
	} else if (status == EFI_NOT_FOUND) {
		status = efi_load_initrd_cmdline(image, &initrd, soft_limit,
						 hard_limit);
```

The logic inside `efi_load_initrd_dev_path()` starts with:
```c
	dp = (efi_device_path_protocol_t *)&initrd_dev_path;
	status = efi_bs_call(locate_device_path, &lf2_proto_guid, &dp, &handle);
	if (status != EFI_SUCCESS)
		return status;

	status = efi_bs_call(handle_protocol, handle, &lf2_proto_guid,
			     (void **)&lf2);
	if (status != EFI_SUCCESS)
		return status;
```
which suggests that the cleanest fix will be to ensure that the handle is installed with the `EFI_DEVICE_PATH_PROTOCOL` instance present but with the `EFI_LOAD_FILE2_PROTOCOL` instance absent.

This matches the state that we deliberately put the handle into when uninstalling our load file protocol instance: we leave the handle intact for reasons explained in `efi_file_path_install()`:
```
 * The design choice in Linux of using a single fixed device path
 * makes this unfortunately messy to support, since device paths must
 * be unique within a system.  When multiple bootloaders are used
 * (e.g. GRUB loading iPXE loading Linux) then only one bootloader can
 * ever install the device path onto a handle.  Subsequent bootloaders
 * must locate the existing handle and replace the load file protocol
 * instance with their own.
```

mcb30, 2023-02-26:
A very quick hack test confirms that this approach does fix the problem:
```diff
--- a/src/interface/efi/efi_file.c
+++ b/src/interface/efi/efi_file.c
@@ -1084,7 +1084,7 @@ int efi_file_install ( EFI_HANDLE handle ) {
 
        /* Install Linux initrd fixed device path file */
        if ( ( rc = efi_file_path_install ( &efi_file_initrd_path.vendor.Header,
-                                           &efi_file_initrd.load ) ) != 0 ) {
+                                           NULL ) ) != 0 ) {
                goto err_initrd;
        }
 
```

With the above hack in place, a Linux kernel with no initrd will boot as expected.

cacheman, 2023-02-26:
Hi....

A couple of things to address here....hope thats OK. 

Firstly to respond to @mcb30, my name is Chris Bradshaw and my email address is cwbshaw@gmail.com. I see your fix/hack above and I will test as soon as I can. Below is a number of workaround steps I tried out and actually got it to work in the end (without the fix/hack) earlier today....

Apologies to @NiKiZe.....in the previous version of iPXE I had been using (v1.20.1) I had never seen this message:

EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path

....which I mistook to be an error, or at least a warning of some kind....clearly it doesn't have the word error anywhere in the message, so my bad.....

As regards using a minimal initrd line, I have tried this with a proper initrd.img file.....I don't get the first efi_main() failed error message.....I get the 'Loaded initrd from....' message above and it does boot, but at the point where the kernel pivots to mount a root filesystem, it just uses the initrd instead of the root=/dev/nfs nfsroot=XXXX I have defined on the kernel cmdline....

Just for the hell of it I gzip'ed an empty file and named it initrd.img to see what would happen....again, only message I get is the 'Loaded initrd from....' message above and it does boot, but at the point of mounting a root filesystem it just bombs out with a kernel panic, because there is no filesystem in my initrd.img, and again it seems to ignore the root=/dev/nfs nfsroot=XXXX kernel options.

I did however get it to work by using the kernel option 'noinitrd'. When I do this it boots first time, but it sits at the point shown below for about 30 seconds before moving on:

http://aaa.bbb.ccc.ddd/bzImage... ok
bzImage: 9531264 bytes [EFI] [SELECTED] "noinitrd root=/dev/nfs nfsroot=aaa.bbb.ccc.ddd:/export/PC,vers=3,rsize=8192,wsize=8192,actimeo=9999 ip=dhcp nofb nomodeset nohugeiomap vga=0x317"

I was then able to eliminate the 30 seconds wait by changing my menu entry to look like:

:TestLinux
initrd http://aaa.bbb.ccc.ddd/initrd-empty.img
kernel http://aaa.bbb.ccc.ddd/bzImage noinitrd root=/dev/nfs nfsroot=aaa.bbb.ccc.ddd:/export/PC,vers=3,rsize=8192,wsize=8192,actimeo=9999 ip=dhcp nofb nomodeset nohugeiomap vga=0x317
boot || goto failed
goto start

So I'm guessing this fools iPXE into loading a dummy initrd, but 'bzImage noinitrd' tells the linux kernel to ignore it.....?

In any case, this last workaround works and with no 30 second wait....

Hope this helps.

Cacheman.

cacheman, 2023-02-27:
Hi....

Just to let you know I've tried the fix listed above by @mcb30 and it does work.

It does seem to wait for 30 seconds before moving on from here:

`http://aaa.bbb.ccc.ddd/bzImage... ok`
`bzImage: 9531264 bytes [EFI] [SELECTED] "noinitrd root=/dev/nfs nfsroot=aaa.bbb.ccc.ddd:/export/PC,vers=3,rsize=8192,wsize=8192,actimeo=9999 ip=dhcp nofb nomodeset nohugeiomap vga=0x317"`

....but it does work.

I tried using the dummy initrd and noinitrd combination described at the end of my last comment, but this time it still waits 30 seconds.....

Hope this helps.

Cacheman

mcb30, 2023-02-27:
Thanks for testing.  I'm not sure where your 30 second delay is coming from, but whatever it is seems to be a separate issue.

What do you see as the last output that appears before the 30 second delay, and the first output that appears after the 30 second delay?

cacheman, 2023-02-27:
Hi....

@mcb30 humble apologies....I found the cause of the 30 seconds delay....I had put a 30sec delay into my iPXE script so I could record the bootup messages from imgstat and I forgot about it. 

Removed it just now, and your fix works fine....kernel boots with no delay. 

Again apologies....my bad....

Cacheman.

mcb30, 2023-02-28:
@cacheman Thanks for testing - the proper fix is now upstreamed as commit https://github.com/ipxe/ipxe/commit/04e60a278

closed 2023-02-28 by mcb30