ipxe/ipxe #910: Compilation fails for 808615bc.bin

niansa, 2023-03-05:
Hey!

I am attempting to compile `808615bc.bin` like this:

    make bin/808615bc.bin NO_WERROR=1 -j12

However, I am ending up with this error:

      [LD] bin/808615bc.tmp
    ld: warning: patch_cf.o: missing .note.GNU-stack section implies executable stack
    ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
    ld:--defsym:1: undefined symbol `_tmp_start' referenced in expression
    make: *** [Makefile.housekeeping:1189: bin/808615bc.tmp] Error 1

I am on Debian Testing, GCC version 12.2.0, binutils version 2.40.

stappersg, 2023-03-05:
I can reproduce that particular build error. For finding a _good_ for [_git bisect_](https://ipxe.org/howto/bisect), I tried `v1.21.1`, `v1.20.1` and `v1.0.0`. The first two have the same build error, `v1.0.0` fails with a completely different build error. That third attempt was my last attempt to find a _good_.

stappersg, 2023-03-05:
For what its worth: `make bin/808615bc.rom NO_WERROR=1` builds fine.

stappersg, 2023-03-06:
[Same PCI-ID](https://github.com/ipxe/ipxe/pull/911/files#diff-cfa32e513a91bad265dcbe306932291314e33904df6f3e6174a5b593b151ccee)

mcb30, 2023-03-06:
> I am attempting to compile `808615bc.bin` like this:

This is a non-existent build target.  The build system is correctly giving you an error.

See https://ipxe.org/appnote/buildtargets for a description of the available build targets, and please open a discussion if you need further help.

closed 2023-03-06 by mcb30

Krutonium, 2024-08-04:
> > I am attempting to compile `808615bc.bin` like this:
> 
> This is a non-existent build target. The build system is correctly giving you an error.
> 
> See https://ipxe.org/appnote/buildtargets for a description of the available build targets, and please open a discussion if you need further help.

https://ipxe.org/howto/romburning

According to this it *should* be. I'm running into the same error with `make bin/80861539.bin -j16`