ipxe/ipxe #919: ld:--defsym:2: undefined symbol `obj_rt8139' referenced in expression

perkelix, 2023-03-12:
Git as of 9e1f7a3659071004f4b8c76f2593da6287f0d575 gives me the following when building 'make bin/rt8139.lkrn':

...
  [HOSTCC] util/zbin
  [VERSION] bin/version.rt8139.lkrn.o
  [LD] bin/rt8139.lkrn.tmp
ld: warning: patch_cf.o: missing .note.GNU-stack section implies executable stack
ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
ld:--defsym:2: undefined symbol `obj_rt8139' referenced in expression
make: *** [Makefile.housekeeping:1225: bin/rt8139.lkrn.tmp] Error 1
rm bin/version.rt8139.lkrn.o

binutils: 2.40
GCC: 12.2.0

mcb30, 2023-03-12:
> Git as of [9e1f7a3](https://github.com/ipxe/ipxe/commit/9e1f7a3659071004f4b8c76f2593da6287f0d575) gives me the following when building 'make bin/rt8139.lkrn':

You are missing the `l` in `rtl8139`.

closed 2023-03-12 by mcb30

perkelix, 2023-03-12:
DOH! Indeed so. Sorry!