ipxe/ipxe #921: Legacy下USB2.0的USBLAN使用IPXE的rom无法进入PXE，但是USB3.0的USBLAN可以正常进入PXE

Donbdha, 2023-03-15:
使用最新的IPXE代码编译rom文件：make bin/ecm--ncm.rom
在Legacy下接入USB2.0的USBLAN，并从此设备启动，启动时跑完iPXE的ROM后没有进入PXE，而是继续遍历BIOS下的启动启动项
[image]

并出现No Configuration methods succeeded的打印。

将USB2.0的USBLAN更换成USB3.0的USBLAN，则可以在跑完iPXE的ROM后成功进入PXE环境

想请问下是否是iPXE的代码不支持USB2.0的设备，如果不支持的话，如何修改代码来兼容USB2.0的设备？

NiKiZe, 2023-03-15:
If you have a device with issues, you need to post the I'd of the device, preferable the output of lsusb, so we know which device it is.

I would also ask that you enter iPXE shell, run DHCP which will give you failure. And then the ifstat command.
This will show how many packets was sent and received, as well as errors.

The next step might be to check if there is any packets sent on the network, you will need tools such as tcpdump or wireshark.

Based on the output you have network issues or broken hardware. All this is covered in on the ipxe.org page which is linked in the error message.

mcb30, 2023-03-15:
> Would you like to ask whether the iPXE code does not support USB2.0 devices? If not, how to modify the code to be compatible with USB2.0 devices?

iPXE is compatible with both USB2 and USB3 devices, and the fact that you have got as far as seeing the `cdc-ecm` device listed on screen with a MAC address indicates that iPXE is successfully communicating with the USB2 device.

As @NiKiZe says, you will have to debug this as per any normal issue.

Donbdha, 2023-03-16:
USB2.0 USBLAN:
[image]

[image]

[image]

[image]

Does this mean that there is a problem with the DHCP configuration of the PXE server?
However, USB3.0 USBLAN uses the DHCP command to return OK. Use upxe.rom on the Intel official website to replace the compiled iPXE lom, and the USBLAN of USB2.0 can also enter the PXE service normally.

Is there any way to solve this problem by modifying the iPXE code?

NiKiZe, 2023-03-16:
Run dhcp then ifstat, we need ifstat data after dhcp.

I would run a packet dump on the network to see what is going on.

Donbdha, 2023-03-16:
[image]

NiKiZe, 2023-03-16:
Sent 4 packets, received 6827928, and 3159 receive errors.
Is this on a very active network?

You might need to get a test environment to narrow this down further, as I said use tcpdump to verify transmit path. And then build iPXE with debug enabled to see what is going on. But this will not be possible unless you have a test environment.

NiKiZe, 2026-03-05:
No feedback on the test environment, happy to have this re-opened if any new details comes in.