ipxe/ipxe #922: iPXE Boot Loop - menu.ipxe is not loaded

bloomcake, 2023-03-16:
I am trying to start `menu.ipxe` automatically. If the server is in PXE and has `undionly.kpxe` loaded, nothing else happens.... it boots this in a loop.

If I open the console (CRTL+B) and manually load the `menu.ipxe` via TFTP (imgload) and execute it (imgexec), it starts the menu.

(I tried in BIOS and EFI mode, but it does not work)

Does anyone see a mistake or did I misunderstand it to automatically look for the `menu.ipxe` in tftp? Otherwise, how can I tell it to run a script on boot?

tcpdump tftp (udp/69)
```
12:35:42.769352 IP (tos 0x0, ttl 64, id 822, offset 0, flags [none], proto UDP (17), length 71)
    172.16.18.214.1678 > ris-ffm.tftp: [udp sum ok]  43 RRQ "undionly.kpxe" octet blksize 1432 tsize 0
12:35:45.006311 IP (tos 0x0, ttl 64, id 873, offset 0, flags [none], proto UDP (17), length 71)
    172.16.18.214.41855 > ris-ffm.tftp: [udp sum ok]  43 RRQ "undionly.kpxe" octet blksize 1432 tsize 0
12:35:47.423234 IP (tos 0x0, ttl 64, id 856, offset 0, flags [none], proto UDP (17), length 71)
    172.16.18.214.1080 > ris-ffm.tftp: [udp sum ok]  43 RRQ "undionly.kpxe" octet blksize 1432 tsize 0
```

[image]

dhcpd.conf
```
authoritative;
default-lease-time 86400; # 24 hours
max-lease-time 86400; # 24 hours
allow booting;
deny bootp;

shared-network int {
    option broadcast-address 172.16.255.255;
    option domain-name-servers 1.1.1.1, 8.8.8.8;
    option ntp-servers 80.81.197.2;
    option routers 172.16.0.1;
    option time-offset 60;
    deny unknown-clients;

    class "UEFI-64-1" {
        match if substring(option vendor-class-identifier, 0, 20) = "PXEClient:Arch:00007";
        filename "ipxe.efi";
    }

    class "UEFI-64-2" {
        match if substring(option vendor-class-identifier, 0, 20) = "PXEClient:Arch:00008";
        filename "ipxe.efi";
    }

    class "UEFI-64-3" {
        match if substring(option vendor-class-identifier, 0, 20) = "PXEClient:Arch:00009";
        filename "ipxe.efi";
    }

    class "Legacy" {
        match if substring(option vendor-class-identifier, 0, 20) = "PXEClient:Arch:00000";
        filename "undionly.kpxe";
    }

    subnet 172.16.0.0 netmask 255.255.0.0 {
        next-server 172.16.0.99;
    }
}

include "/data/hosts/hosts.conf";
```

bloomcake, 2023-03-16:
Okay, I think I need to use embedded scripts to achieve what I want https://ipxe.org/embed

closed 2023-03-16 by bloomcake

mcb30, 2023-03-16:
For `ipxe.efi`, you should find that iPXE tries to automatically load the file `autoexec.ipxe` (not `menu.ipxe`).  This feature is not yet implemented for `undionly.kpxe`.

You could alternatively use a declaration such as:
```
    class "iPXE" {
        match if option user-class = "iPXE";
        filename "menu.ipxe";
    }
```
(though you'll need to make sure that this class declaration takes priority over the others, since iPXE will match on both).