ipxe/ipxe #923: DHCP error: "No configuration methods succeeded" with Broadcom BCM5720

EugenioSiciliano, 2023-03-17:
Hello all,
I tried booting with both UEFI and BIOS mode in a chainloading setup on a Dell R450 equipped with a Broadcom NetXtreme BCM5720 and an Intel Ethernet Controller X710 for 10GbE SFP+.

The Broadcom adapter fails at DHCP requests with both undionly.kpxe and ipxe.efi, although with some differences, while the Intel one always works OK instead.
I got same results even after updating BIOS and NIC firmwares to the last version provided by Dell.

All builds I tried are from commit 09e8a154, I only added `DEBUG=netdevice,tg3:3` to gather some logs.

### BIOS boot with undionly.kpxe
DHCP phase never succeeds at first try, it could work on 2nd or 3rd attempt when I hit Ctrl-B and issue `dhcp` manually. Then, even after network is correctly set, `autoboot` fails with a timeout. I managed to capture a messy output log (sorry for that, it comes from the IPMI console and has lots of escape characters, duplicate lines, etc, it is best viewed with `less -R`) and a PCAP dump of requests received at the DHCP server side, relevant packets can be filtered with `eth.addr==D0-8E-79-BA-6C-6E`.
[bios.log](https://github.com/ipxe/ipxe/files/11004454/bios.log)
[bios.pcap.zip](https://github.com/ipxe/ipxe/files/11004462/bios.pcap.zip)

### UEFI boot with ipxe.efi:
DHCP phase never works, even with manual commands. I only have the output log for this because at server side no DHCP packets were captured at all.
[uefi.log](https://github.com/ipxe/ipxe/files/11004463/uefi.log)

Thanks for your time.

NiKiZe, 2023-03-17:
If undionly.kpxe or snponly.efi/snp.efi fails then you have bugs in your BIOS or firmware code or the network stack/driver they provide. However also consider the issues mention on the ipxe.org error page, since it works on 2nd/3rd try, could it be STP blocking the port? (Reading on phone so have not checked your logs)

depending on how broadcom NICs was initialized they usually fail with the native driver. That is already well documented in other issues and that part makes this a duplicate.

EugenioSiciliano, 2023-03-23:
> depending on how broadcom NICs was initialized they usually fail with the native driver. That is already well documented in other issues and that part makes this a duplicate.

OK, indeed I'd like to understand whether some BIOS configuration could cause that.

> This seems to show multiple successful DHCP discover-offer-request-ack cycles taking place, and the packet capture confirms this. I do not see the problem that you describe.

Right, I did lots of tests and maybe I uploaded the wrong log for undionly case. However, in the last few days I cannot reproduce the same DHCP issue anymore, I think because in the meanwhile I tried different values in BIOS parameters, and now I don't remember what the exact BIOS configuration was.

> The log also seems to show that you are using iPXE to load iPXE to load iPXE to load iPXE, and that all of the iPXE builds are customised (since the DHCP user class is "Tinkerbell" rather than the standard "iPXE").

Sorry again for the messy logs, they had so many duplicate lines it looked like iPXE was loading many times, it always failed any download instead.
For the logs I am attaching now, I removed all the customizations (except for the ping command).

Despite the former DHCP issue does not happen anymore, I still observe this "Connection timed out" issue with both undionly.kpxe and ipxe.efi:
[image]
I don't remember what the exact BIOS configuration was when I first observed the DHCP issue, however now it is like this:
[image]
[image]

This time, I managed to capture neater logs:
[bios_session.txt](https://github.com/ipxe/ipxe/files/11054000/bios_session.txt)
[bios_session.pcap.zip](https://github.com/ipxe/ipxe/files/11054469/bios_session.pcap.zip)

[uefi_session.txt](https://github.com/ipxe/ipxe/files/11054571/uefi_session.txt)
[uefi_session.pcap.zip](https://github.com/ipxe/ipxe/files/11054595/uefi_session.pcap.zip)

Some observations:
- I enabled `tcp` DEBUG as per https://ipxe.org/err/4c0a60 advice
- network configuration set through DHCP looks good (ip/netmask/gateway)
- webserver in 172.17.2.114 is reachable (most of time a manual ping works)
- maybe I'm wrong, however in UEFI boot I was expecting something like "tg3" to be shown, it says `using 14e4-165f` instead

Thanks again.

mcb30, 2023-03-24:
> [image]
> 
> This time, I managed to capture neater logs: [bios_session.txt](https://github.com/ipxe/ipxe/files/11054000/bios_session.txt) [bios_session.pcap.zip](https://github.com/ipxe/ipxe/files/11054469/bios_session.pcap.zip)

Thank you for cleaning up the logs.  From the screenshot and files provided: your DHCP is completing successfully, but it appears that you do not have any web server listening at 172.17.2.114.  That IP address is responding to pings, but is ignoring connection attempts on port 80.  Your packet capture confirms this.

I am closing this issue since there is no problem with iPXE here.  Please feel free to start a discussion at https://github.com/ipxe/ipxe/discussions to ask for help in setting up your web server to serve your `auto.ipxe` file.

closed 2023-03-24 by mcb30