ipxe/ipxe #937: Stuck on "iPXE initialising devices" after chainloading ipxe.efi on `ice` driver E810 Intel NICs

nshalman, 2023-04-04:
I have some machines with Intel E810 NICs installed. They are able to boot into my current builds of ipxe.efi, however they don't handle chainloading from there into e.g. `http://boot.netboot.xyz` When that menu tries to chainload into ipxe.efi, the system will hang at `iPXE initialising devices`

I was able to reproduce this by trying to chainload from my production iPXE environment into my own debug build of the latest iPXE. I'm not totally certain where the bug is.

Example output from capturing `DEBUG=pci,ice`
```
iPXE initialising devices...0000:00:07.1 (1022:1484) has driver "bridge"
0000:00:07.1 has mem 10100 io 1f0 irq 255
0000:00:08.1 (1022:1484) has driver "bridge"
0000:00:08.1 has mem 20200 io 1f0 irq 255
0000:40:01.1 (1022:1483) has driver "bridge"
0000:40:01.1 has mem 414140 io 200001f0 irq 255
0000:41:00.0 (8086:159b) has driver "e810-xxv-sfp"
0000:41:00.0 has mem 92000000 io 0 irq 255
0000:41:00.0 device not enabled by BIOS! Updating PCI command 0146->0147
0000:41:00.0 latency timer is unreasonably low at 0. Setting to 32.
ICE 0x550ab0f0 PF 0 using port 0
ICE 0x550ab0f0 firmware v1/7.0.7 API v0/1.7.10
ICE 0x550ab0f0 VSI 0x0004 uplink 0x0000 func 0x0000
ICE 0x550ab0f0 has MAC address b4:96:91:dc:b1:7e
ICE 0x550ab0f0 speed 0x80 status 0xe1
0000:41:00.1 (8086:159b) has driver "e810-xxv-sfp"
0000:41:00.1 has mem 90000000 io 0 irq 255
0000:41:00.1 device not enabled by BIOS! Updating PCI command 0146->0147
0000:41:00.1 latency timer is unreasonably low at 0. Setting to 32.
ICE 0x550ac0f0 PF 1 using port 1
ICE 0x550ac0f0 firmware v1/7.0.7 API v0/1.7.10
ICE 0x550ac0f0 VSI 0x0005 uplink 0x0001 func 0x0001
ICE 0x550ac0f0 has MAC address b4:96:91:dc:b1:7f
ICE 0x550ac0f0 speed 0x80 status 0xe1
0000:40:03.1 (1022:1483) has driver "bridge"
0000:40:03.1 has mem 424240 io 1f0 irq 255
0000:40:03.2 (1022:1483) has driver "bridge"
0000:40:03.2 has mem 434340 io 1f0 irq 255
0000:40:03.3 (1022:1483) has driver "bridge"
0000:40:03.3 has mem 444440 io 1f0 irq 255
0000:40:03.4 (1022:1483) has driver "bridge"
0000:40:03.4 has mem 454540 io 1f0 irq 255
0000:40:07.1 (1022:1484) has driver "bridge"
0000:40:07.1 has mem 464640 io 1f0 irq 255
0000:40:08.1 (1022:1484) has driver "bridge"
0000:40:08.1 has mem 474740 io 1f0 irq 255
0000:80:01.1 (1022:1483) has driver "bridge"
0000:80:01.1 has mem 818180 io 1f0 irq 255
0000:80:01.2 (1022:1483) has driver "bridge"
0000:80:01.2 has mem 828280 io 1f0 irq 255
0000:80:07.1 (1022:1484) has driver "bridge"
0000:80:07.1 has mem 838380 io 1f0 irq 255
0000:80:08.1 (1022:1484) has driver "bridge"
0000:80:08.1 has mem 848480 io 1f0 irq 255
0000:c0:01.1 (1022:1483) has driver "bridge"
0000:c0:01.1 has mem c3c3c0 io 2000a1a0 irq 255
0000:c0:03.1 (1022:1483) has driver "bridge"
0000:c0:03.1 has mem c4c4c0 io 1f0 irq 255
0000:c0:03.2 (1022:1483) has driver "bridge"
0000:c0:03.2 has mem c5c5c0 io 1f0 irq 255
0000:c0:03.3 (1022:1483) has driver "bridge"
0000:c0:03.3 has mem c6c6c0 io 1f0 irq 255
0000:c0:03.4 (1022:1483) has driver "bridge"
0000:c0:03.4 has mem c7c7c0 io 1f0 irq 255
0000:c0:05.2 (1022:149a) has driver "bridge"
0000:c0:05.2 has mem c2c1c0 io 200001f0 irq 255
0000:c1:00.0 (1556:be00) has driver "bridge"
0000:c1:00.0 has mem da80da00 io c2c2c0 irq 255
0000:c0:07.1 (1022:1484) has driver "bridge"
0000:c0:07.1 has mem c8c8c0 io 1f0 irq 255
0000:c0:08.1 (1022:1484) has driver "bridge"
0000:c0:08.1 has mem c9c9c0 io 1f0 irq 255
```

This might be related to #624
See https://github.com/ipxe/ipxe/issues/624#issuecomment-1131975540

stappersg, 2023-04-04:
On Tue, Apr 04, 2023 at 12:43:20PM -0700, Nahum Shalman wrote:
> Example output from capturing `DEBUG=pci,ice`
> ```
> iPXE initialising devices...
> 0000:00:07.1 (1022:1484) has driver "bridge"
> 0000:00:07.1 has mem 10100 io 1f0 irq 255
> 0000:00:08.1 (1022:1484) has driver "bridge"
    ...
> 0000:c0:07.1 (1022:1484) has driver "bridge"
> 0000:c0:07.1 has mem c8c8c0 io 1f0 irq 255
> 0000:c0:08.1 (1022:1484) has driver "bridge"
> 0000:c0:08.1 has mem c9c9c0 io 1f0 irq 255

Idea:
* Boot into Linux
* `lspci`
* Check if there PCI devices after 0000:c0:08.1
* If so, tell which PCI ID they have, as intermediate report
* "blacklist" those PCI-IDs in iPXE
* Try again

Groeten
Geert Stappers

P.S.
$ grep 'has mem'  DEBUG=pci
0000:40:03.1 has mem 424240 io 1f0 irq 255
0000:40:03.2 has mem 434340 io 1f0 irq 255
0000:40:03.3 has mem 444440 io 1f0 irq 255
0000:40:03.4 has mem 454540 io 1f0 irq 255
0000:40:07.1 has mem 464640 io 1f0 irq 255
0000:40:08.1 has mem 474740 io 1f0 irq 255
0000:80:01.1 has mem 818180 io 1f0 irq 255
0000:80:01.2 has mem 828280 io 1f0 irq 255
0000:80:07.1 has mem 838380 io 1f0 irq 255
0000:80:08.1 has mem 848480 io 1f0 irq 255
0000:c0:01.1 has mem c3c3c0 io 2000a1a0 irq 255
0000:c0:03.1 has mem c4c4c0 io 1f0 irq 255
0000:c0:03.2 has mem c5c5c0 io 1f0 irq 255
0000:c0:03.3 has mem c6c6c0 io 1f0 irq 255
0000:c0:03.4 has mem c7c7c0 io 1f0 irq 255
0000:c0:05.2 has mem c2c1c0 io 200001f0 irq 255
0000:c1:00.0 has mem da80da00 io c2c2c0 irq 255
0000:c0:07.1 has mem c8c8c0 io 1f0 irq 255
0000:c0:08.1 has mem c9c9c0 io 1f0 irq 255
$ 

I would expect other "has mem" values. ( Those values
look odd to me.)
-- 
Silence is hard to parse

NiKiZe, 2023-04-04:
Does the behaviour change in any way if you chain from snp.efi into ipxe.efi, or the other way around?

stappersg, 2023-05-19:
Another idea  `make DEBUG=intel:3`.  That is based upon
```text
0000:41:00.0 (8086:159b) has driver "e810-xxv-sfp"
0000:41:00.0 has mem 92000000 io 0 irq 255
0000:41:00.0 device not enabled by BIOS! Updating PCI command 0146->0147
0000:41:00.0 latency timer is unreasonably low at 0. Setting to 32.
```
and 
```text
0000:41:00.1 (8086:159b) has driver "e810-xxv-sfp"
0000:41:00.1 has mem 90000000 io 0 irq 255
0000:41:00.1 device not enabled by BIOS! Updating PCI command 0146->0147
0000:41:00.1 latency timer is unreasonably low at 0. Setting to 32.
```
from the original report.

foyerunix, 2024-07-30:
Hello,

In our use case we use IPXe to boot on servers with Intel E810 25Gb SFP NIC connected to switches with LACP enabled. Our switches doesn't support any LACP fallback mode, so if IPXe doesn't establish the LACP session the install will fail.

With the current IPXe code we cannot establish a LACP session, therefore the install fail. If we disable LACP on the switches, the install complete as expected.

I can confirm that by commenting the following line, the installation will complete with LACP enabled:

https://github.com/ipxe/ipxe/blob/d2d194bc60f012569fa95ed54693cb6663beb5ce/src/drivers/net/ice.c#L963

Best Regards.