ipxe/ipxe #954: Misc. proxyDHCP issues

deliciouslytyped, 2023-05-23:
@mcb30 
 Attached is a PCAP file (renamed .txt for upload) with packets from the issue we were discussing on IRC.
Specifically packets 3 and 16, and the pcap was generated by running ipxe's dhcp command once for each of the two following dhcp server instances (the difference is the ip address added to dhcp-boot):

The first fails to load the tftp filename under proxydhcp/filename, the second does load it. (checked via the `config` command)

[1]
```
sudo /nix/store/6rnl636ayiy18927ivr3wvy9vwnyq9ak-dnsmasq-2.89/bin/dnsmasq -d --bind-interface --port=0 -l leases --dhcp-range=192.168.101.1,proxy --log-dhcp --log-queries --log-debug --enable-tftp --tftp-root="$(realpath ./files)" --pxe-service=254,dummy --dhcp-boot=pxelinux.0
```

iPXE output:
```
iPXE> dhcp
DHCP 0x222b4 entering discovery state
Configuring (net0 52:54:00:12:34:56)...DHCP 0x222b4 DHCPDISCOVER
DHCP 0x222b4 DHCPOFFER from 192.168.101.1:67 for 192.168.101.239
DHCP 0x222b4 DHCPOFFER from 192.168.101.1:67 proxy
DHCP 0x222b4 entering request state
DHCP 0x222b4 DHCPREQUEST to 192.168.101.1:67 for 192.168.101.239
.DHCP 0x222b4 DHCPACK from 192.168.101.1:67 for 192.168.101.239
DHCP 0x222b4 entering ProxyDHCP state
DHCP 0x222b4 ProxyDHCP REQUEST to 192.168.101.1
DHCP 0x222b4 DHCPACK from 192.168.101.1:4011
DHCP 0x222b4 ProxyDHCP REQUEST to 192.168.101.1
DHCP 0x222b4 DHCPACK from 192.168.101.1:4011
DHCP 0x222b4 ProxyDHCP REQUEST to 192.168.101.1
DHCP 0x222b4 DHCPACK from 192.168.101.1:4011
.DHCP 0x222b4 ProxyDHCP REQUEST to 192.168.101.1
DHCP 0x222b4 DHCPACK from 192.168.101.1:4011
.. ok
iPXE>
```

[2]
```
sudo /nix/store/6rnl636ayiy18927ivr3wvy9vwnyq9ak-dnsmasq-2.89/bin/dnsmasq -d --bind-interface --port=0 -l leases --dhcp-range=192.168.101.1,proxy --log-dhcp --log-queries --log-debug --enable-tftp --tftp-root="$(realpath ./files)" --pxe-service=254,dummy --dhcp-boot=pxelinux.0,,192.168.101.1
```
[virbr2--2023-05-23--10-50-51.pcap.txt](https://github.com/ipxe/ipxe/files/11541739/virbr2--2023-05-23--10-50-51.pcap.txt)

deliciouslytyped, 2023-05-23:
I failed to notice that option 54 is set everywhere, however has_pxe_opts specifically checks for `dhcppkt->dhcphdr->siaddr.s_addr`
https://github.com/ipxe/ipxe/blob/master/src/net/udp/dhcp.c#L308

deliciouslytyped, 2023-05-23:
Right, I mostly figured that out.

I'm explicitly trying to not care about having a boot menu set (because it doesn't make sense?), so that's just something dnsmasq adds (IDK if its by spec or just something they do).

So I guess my question is what can be done about siaddr not being set?

This looks like a dnsmasq (documentation) bug?
Per the quote in the documentation above it looks like it should be set right?

mcb30, 2023-05-23:
> So I guess my question is what can be done about siaddr not being set?
> This looks like a dnsmasq (documentation) bug? Per the quote in the documentation above it looks like it should be set right?

Yes, the dnsmasq documentation suggests that it should be setting `siaddr` to its own IP address.  You'd have to dig through the dnsmasq code to figure out precisely what it's doing here.

deliciouslytyped, 2023-05-23:
The relevant code seems to be around these areas but I can't tell if they really just forgot to set a default.

https://github.com/imp/dnsmasq/blob/770bce967cfc9967273d0acfb3ea018fb7b17522/src/rfc2131.c#L2440
https://github.com/imp/dnsmasq/blob/770bce967cfc9967273d0acfb3ea018fb7b17522/src/option.c#L3966