ipxe/ipxe #956: Initrd autodection breaks Ubuntu's "hwe" initrd on bare metal

drlamb, 2023-05-25:
Background: Ubuntu provides updated kernels and initrds for their LTS releases by way of the "[hardware enablement stack](https://wiki.ubuntu.com/Kernel/LTSEnablementStack)." In order to ensure the broadest hardware support and to avoid the duplication of effort by updating the system after installation we had been PXE booting using the hwe kernel and initrd for well over a year at this point. 

After I rebuilt iPXE from master recently I noticed that our Ubuntu 20.04.X PXE attempts started failing to boot with the error "Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)."  

As shown in the following iPXE script snippet _I am_ defining the initrd= imgarg to the provided kernel file that iPXE pulls down. 

```
#!ipxe

### iPXE Configuration ###
set pxe-server [Redacted]
set dns [Redacted]

## Network Configuration ##
# Attempt DHCP connection until successful 
:retry_dhcp_connection 
dhcp && echo DHCP CONNECTION SUCCESSFUL && goto menu || goto retry_dhcp_connection

## Main Menu ##
:menu
menu Linux Installers
item
item --gap -- Ubuntu
item
item focal 20.04.6
item
item --gap -- iPXE Utilities 
item
item shell Enter iPXE shell
choose os && goto ${os}

:focal
echo -n Enter the asset type : && read asset-type
echo -n Enter the owner of the asset : && read asset-owner
echo -n Enter the last five digits of the asset tag : && read asset-tag
echo -n Enter the asset class : && read asset-class
kernel tftp://${pxe-server}/ubuntu/amd64/focal/pxe/hwe-vmlinuz || read void
initrd tftp://${pxe-server}/ubuntu/amd64/focal/pxe/hwe-initrd || read void
imgargs hwe-vmlinuz initrd=hwe-initrd ip=dhcp url=http://${pxe-server}/ubuntu/amd64/focal/iso/ubuntu-20.04.6-live-server-amd64.iso boot=casper ignore_uuid nomodeset modprobe.blacklist=nouveau rd.driver.blacklist=nouveau autoinstall ds=nocloud-net;s=http://${pxe-server}/ubuntu/amd64/focal/ cloud-config-url=/dev/null hostname=L-${asset-type}T-${asset-owner}${asset-tag}-${asset-class}.[REDACTED]
boot || goto fail 

:shell
shell
set timeout 0
set submenu-timeout 0 
goto menu

:fail
echo PXE BOOTING FAILED
```

Messing with it I was able to both confirm that **non-hwe** kernels/initrds still work as expected and from bisection that this issue stems from [6a004be0cceab5d669eedb5a2e6ee2feab31d5bd](https://github.com/ipxe/ipxe/commit/6a004be0cceab5d669eedb5a2e6ee2feab31d5bd). I've attempted removing the initrd= declaration as well as trying to rely on the initrd.magic variable instead of static file names to no change. The kicker is that I'm unable to replicate this issue locally with libvirt. Building iPXE straight from master and using the above script (embedded in bin-x86_64-efi-sb/ipxe.efi) works in tianocore on a virtual machine. (SHA1 hashes of the boot files match)

Regardless of NIC (intel or realtek) bare metal hosts fail in the aforementioned way. Our bare metal PXE infrastructure is handled by dnsmasq running on RHEL 8 in _proxydhcp_ mode serving the built iPXE efi executable to UEFI-hosts only. 

For now I've got iPXE working again by using the commit prior to the above but would appreciate any insights you may have. Thank you for your time.

NiKiZe, 2023-05-25:
So this fails on EFI, can you put imgstat before your boot command and post the output?
While at it, also grab which kernel version is used.

Have you tried without initrd= in your imgargs?

Thank you for a well written and detailed but still balanced issue.

drlamb, 2023-05-25:
> So this fails on EFI, can you put imgstat before your boot command and post the output? While at it, also grab which kernel version is used.

(imgargs not necessary to boot Ubuntu's installer omitted)

```
Backdrop6.png = 27087 bytes
hwe-vmlinuz = 11458952 bytes [EFI] [SELECTED] "initrd=hwe-initrd ip=dhcp url=http://10.10.126.3/ubuntu/amd64/focal/iso/ubuntu-20.04.6-live-server-amd64.iso
hwe-initrd = 126393731 bytes
```

Ubuntu 20.04.6 as deployed above is using Kernel `5.15.0-67-generic`

> Have you tried without initrd= in your imgargs? 

Yes, I've tried omitting the declaration entirely.

stappersg, 2023-05-25:
...
> hwe-vmlinuz = 11458952 bytes [EFI] [SELECTED] "initrd=hwe-initrd ip=dhcp url=http://10.10.126.3/ubuntu/amd64/focal/iso/ubuntu-20.04.6-live-server-amd64.iso"

Those are **not** all the arguments as supplied in the original posting. (Just saying.)

NiKiZe, 2023-05-25:
So you have a PNG in there, try an imgfree before your kernel line,

NiKiZe, 2023-05-25:
As the commit states that you linked to, from kernel 5.7 initrd is not needed, so if you will only be using newer versions of iPXE, skip that arg and leave initrd stuff to the iPXE magic.
Gotcha being to supply valid initrd.

If this still fails with the imgfree we might need to look into DEBUG= compile time options

drlamb, 2023-05-25:
> So you have a PNG in there, try an imgfree before your kernel line,

This solved it at least when paired with **no** initrd= declaration.  I don't have the corporate branding present locally so that explains why it worked in libvirt. (That's what I get for chasing aesthetics)

What's curious to me is how _normal_ **non-hwe** ubuntu works just fine _without_ imgfree being there and the branding image included. Regardless I'm able to boot Ubuntu HWE kernels/initrds directly from firmware netboot using iPXE built from master.

**I cannot thank you enough for your rapid help.**

closed 2023-05-25 by drlamb

NiKiZe, 2023-05-25:
Which kernel version do you have in the nonhwe case

drlamb, 2023-05-25:
> Which kernel version do you have in the non hwe case?

Double-checking, the standard kernel is the _ancient_ 5.4.0-144 so it was only working _because of_ the initrd= declaration. If it was > 5.7 it would've encountered the same conflict with the branding. 

I'm all set now. Thanks again.

NiKiZe, 2023-05-25:
Thanks for verifying the version. Will be helpful to know if someone else comes along with something similar.