ipxe/ipxe #965: Configure ipxe on libvirt/QEMU with SMBIOS OEM strings

joernheissler, 2023-06-04:
Hello,

ipxe provides "[VMware GuestInfo](https://ipxe.org/buildcfg/vmware_settings)" support, allowing to set arbitrary ipxe settings directly from VMware.

However, I use QEMU + libvirt, not VMware. I'd like to propose a new feature that achieves the same on QEMU, using [SMBIOS OEM strings](https://libvirt.org/formatdomain.html#smbios-system-information).

Example domain config for libvirt:
```xml
<domain type='kvm'>
  ...
  <os>
    ..
    <smbios mode='sysinfo'/>
  </os>
  <sysinfo type='smbios'>
    <oemStrings>
      <entry>ipxe.ip=192.168.1.8</entry>
      <entry>ipxe.scriptlet=ifopen net0 &amp;&amp; chain ${filename}</entry>
    </oemStrings>
  </sysinfo>
</domain>
```

libvirt turns this into a call to:
`/usr/bin/qemu-system-x86_64 ... -smbios type=11,value=ipxe.ip=192.168.1.8,value=ipxe.scriptlet=ifopen net0 &amp;&amp; chain ${filename}`

`dmidecode` in the virtual machine reports:
```
Handle 0x0E00, DMI type 11, 5 bytes
OEM Strings
    String 1: ipxe.ip=192.168.1.8
    String 2: ipxe.scriptlet=ifopen net0 && chain ${filename}
```

I attached a working (but probably very bad) patch for ipxe that loads those settings.

Would this feature be interesting for ipxe? If so, could someone who knows their way around the ipxe source code implement it properly?

[ipxe-smbios-oem-strings.patch.txt](https://github.com/ipxe/ipxe/files/11646339/ipxe-smbios-oem-strings.patch.txt)

NiKiZe, 2023-06-04:
This might very well be implemented as an ipxe script using the existing smbios settings. With the challenge to cut strings properly.

Otherwise fantastic, especially the libvirt and qemu smbios examples.

mcb30, 2023-06-09:
> ```
> <domain type='kvm'>
>   ...
>   <os>
>     ..
>     <smbios mode='sysinfo'/>
>   </os>
>   <sysinfo type='smbios'>
>     <oemStrings>
>       <entry>ipxe.ip=192.168.1.8</entry>
>       <entry>ipxe.scriptlet=ifopen net0 &amp;&amp; chain ${filename}</entry>
>     </oemStrings>
>   </sysinfo>
> </domain>
> ```
> 
> Would this feature be interesting for ipxe? If so, could someone who knows their way around the ipxe source code implement it properly?

This would definitely be interesting.  The implementation would end up sharing a lot of code with https://github.com/ipxe/ipxe/blob/master/src/arch/x86/interface/vmware/guestinfo.c, to the extent that most of `guestinfo.c` should be abstracted out to provide a framework for an external mechanism that allows for global and per-netdevice settings to be retrieved externally from a text-based key-value store, where the key comprises `ipxe[.<network device>].<setting name>[.<setting type>]`.

aragilar, 2026-08-20:
From what I can see, the "standard" (or at least common) form that is used is `<owner>:<key>=<value>`, with `<owner>` either being the project name (i.e. `ipxe`) or the reverse domain name (e.g. `org.ipxe`). I think it makes sense to use `ipxe` over `org.ipxe`. I wonder if a first step could be to loop over the OEM strings and put any starting with `ipxe` under `smbios_oem/`, and then if any items matched standard iPXE settings copy them over? Would there be any concerns about adding that?

The other configuration system that's available on QEMU is [fw_cfg](https://www.qemu.org/docs/master/specs/fw_cfg.html), though that's obviously QEMU specific.

Doing some searching on other hypervisors, HyperV has its own protocol like VMWare's guestinfo (though from the docs it's looks much more heavyweight) but no SMBIOS support it appears, Xen can set SMBIOS OEM strings (but I can't see a guestinfo equivalent), and Virtualbox has its guestinfo equivalent via its "Guest Additions" (which I think is undocumented) and not much interest in arbitrary SMBIOS OEM strings (based on the forums there). Of the "cloud hypervisors" which are not cloud platforms (that I'm aware of, there may be others), firecracker doesn't seem to have anything (but I could be wrong), and cloud-hypervisor has SMBIOS. So parsing SMBIOS OEM as key-value pairs independent of networking gives a fair bit of value for providing configuration (and presumably this could be opt-in for VM configs if there were space concerns).

Thoughts? When I have some free time I could add SMBIOS OEM string parsing support?