ipxe/ipxe #973: Stuck on "iPXE initialising devices" HP Elitebook 10Gen

flagoda, 2023-06-16:
Hi,
Somebody success to boot iPXE on HP Elitebook 10 Generation ? (640, 650, 840 etc)
I try latest vesion, same problem i stuck on `iPXE initialising devices`
I'm not an expert on iPXE i don't know where to search

NiKiZe, 2023-06-16:
I assume you are booting in EFI mode?
Which of ipxe.efi, snponly.efi and snp.efi have you tried?
And what is the pciid (vendor and device) of the nic in the machine?

NiKiZe, 2023-06-20:
I'm asking you to test to get data points. Not to switch permanently.

8086:1a1e is a intel NIC.
No device IDs was given for the usb NIC, have you tried booting both with, and without dock?

Hammarskjold, 2023-06-20:
Just ordered an 840 so should be able to test/verify and then plan a fix for this.

flagoda, 2023-06-20:
> I'm asking you to test to get data points. Not to switch permanently.
> 
> 8086:1a1e is a intel NIC. No device IDs was given for the usb NIC, have you tried booting both with, and without dock?

Thanks. It works with snp.efi and snponly.efi.
Do you think we can easily switch ipxe.efi to snp.efi ? Have lot of models Dell, HP

NiKiZe, 2023-06-20:
If you are only installing desktops/laptops and only expect one nic I would usually go with snponly.efi if ipxe.efi gives any issues.

If you want vlans or other things that might change, but test snponly.efi, inform users about the change and document how to switch back if needed.

Did ipxe.efi hang regardless of dock connected or not?
Do you have any remote management configured (in firmware) on the machine?

flagoda, 2023-06-21:
> and only expect one nic
What do you mean by "expect one nic"

> Did ipxe.efi hang regardless of dock connected or not?
ipxe.efi hang regardless if dock connected or not.

> Do you have any remote management configured (in firmware) on the machine?
i don't think

flagoda, 2023-06-21:
> If you are only installing desktops/laptops and only expect one nic I would usually go with snponly.efi if ipxe.efi gives any issues.
What do you mean by "expect one nic"

> Did ipxe.efi hang regardless of dock connected or not? Do you have any remote management configured (in firmware) on the machine?

ipxe.efi hang regardless if dock connected or not.
I don't use remote management firmware like AMT.

NiKiZe, 2023-06-21:
> > If you are only installing desktops/laptops and only expect one nic I would usually go with snponly.efi if ipxe.efi gives any issues.
> > Did ipxe.efi hang regardless of dock connected or not? Do you have any remote management configured (in firmware) on the machine?
> 
> -What do you mean by "expect one nic" ?

snponly should only show the nic it was chained thru.
If you have a machine with multiple NICs, snponly will only ever show the one that was used to network boot, even if it has 2, or 4 ports. (Think server NICs with multiple ports)

flagoda, 2023-06-22:
Many thanks for your help :)

closed 2023-06-22 by flagoda

rexberg, 2023-08-18:
> Just ordered an 840 so should be able to test/verify and then plan a fix for this.

Any success with the 840 model?