ipxe/ipxe #974: iPXE failed to compile in Ubuntu 22.04.2 LTS container

RikiReal, 2023-06-17:
I am unable to compile iPXE (master branch; last commit https://github.com/ipxe/ipxe/commit/4fa4052c7ebb59e4d4aa396f1563c89118623ec7) on Ubuntu 22.04.2 LTS Docker container

Is my compiler version too old? I'd much appreciate any help :smiley: 
I've also unsuccessfully tried compiling iPXE using https://github.com/ipxe/ipxe/releases/tag/v1.21.1 
If you need more info, please let me know and I'll tell you ASAP.

My command is:
```
make -j 4 NO_WERROR=1 EMBED=../../boot-x86_64.ipxe bin-x86_64-efi/ipxe.efi
```
gcc --version: 
```
gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0
```
make --version:
```
GNU Make 4.3
Built for x86_64-pc-linux-gnu
```

The error is:
```
  [VERSION] bin-x86_64-efi/version.ipxe.efi.o
  [AR] bin-x86_64-efi/blib.a
ar: creating bin-x86_64-efi/blib.a
In file included from include/ipxe/features.h:6,
                 from core/version.c:33:
core/version.c:49:32: error: expected expression before ',' token
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      |                                ^
include/ipxe/dhcp.h:548:58: note: in definition of macro 'DHCP_OPTION'
  548 | #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
      |                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:9: note: in expansion of macro '__DHCP_FEATURE'
   71 |         __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
      |         ^~~~~~~~~~~~~~
include/ipxe/features.h:69:9: note: in expansion of macro '_DHCP_FEATURE'
   69 |         _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
      |         ^~~~~~~~~~~~~
include/ipxe/features.h:106:9: note: in expansion of macro 'DHCP_FEATURE'
  106 |         DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
      |         ^~~~~~~~~~~~
core/version.c:49:1: note: in expansion of macro 'FEATURE_VERSION'
   49 | FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
      | ^~~~~~~~~~~~~~~
core/version.c:64:48: error: expected expression before ';' token
   64 | const int product_major_version = VERSION_MAJOR;
      |                                                ^
cc1: note: unrecognized command-line option '-Wno-dangling-pointer' may have been intended to silence earlier diagnostics
make: *** [Makefile.housekeeping:1212: bin-x86_64-efi/version.ipxe.efi.o] Error 1
make: *** Waiting for unfinished jobs....
```

RikiReal, 2023-06-17:
It was a bug on my side.
I added iPXE as a git submodule and populated it. Then I forgot about that.
Inside my Dockerfile I also downloaded iPXE separately using curl.

So I had two different iPXE versions inside my Container and this caused issues.

I'm closing this issue.

closed 2023-06-17 by RikiReal

NiKiZe, 2023-06-17:
Nice that you found the issue.
I would have asked that you do a clean checkout with no local modifications and simple build command, and then if that works change one thing at a time.

None full checkouts, incompatible configs, and weird build commands is some common sources for issues.
Other issues involves incompatible versions of GCC.

stappersg, 2023-06-18:
> I added iPXE as a git submodule and populated it. Then I forgot about that.

That is #534 

> It was a bug on my side.
> I added iPXE as a git submodule and populated it. Then I forgot about that.

That is #534

stappersg, 2023-06-18:
> It was a bug on my side.
> I added iPXE as a git submodule and populated it. Then I forgot about that.

That is #534