ipxe/ipxe #975: Keyboard on Surface Laptop go 2 does not work.

keithga, 2023-06-19:
https://www.microsoft.com/en-us/d/surface-laptop-go-2/8PGLPV76MJHN

After booting to iPXE, attached physical on Laptop does not work. 
However, as a work-around, the "On-Screen" keyboard for the touch screen does work, as does any attached USB keyboard. 

My assumption is that the attached keyboard is not enumerated as a standard USB device, instead requiring some kind of intermediate driver. 

Attached is a list of all "HID" PNP device ID's on the Surface Laptop go 2: 

[surfacelist.txt](https://github.com/ipxe/ipxe/files/11792945/surfacelist.txt)

NiKiZe, 2023-06-19:
If you boot directly to the uefi shell, does the keyboard work there?
Can you boot Linux and figure out how it is connected?

Without details on how to fix this (or paying for commercial support) it is unlikely to get any iPXE based fixed. More likely there can be a firmware based fix released.

mcb30, 2023-06-22:
Built-in laptop keyboards are typically wired in as PS/2 keyboards, since the wire protocol consumes less power than USB.  iPXE does not get involved in driving the hardware for these, but will rely solely on the UEFI firmware's `ConIn` device to read keystrokes.

As @NiKiZe suggested, try booting into the UEFI shell to see if it works there.

keithga, 2023-12-12:
Sorry, there is NO EFI shell on surface go devices (thanks Microsoft).

NiKiZe, 2023-12-12:
Wait, isn't the keyboard on those Bluetooth!?

Boot efi shell binary from USB?

keithga, 2023-12-13:
No, this is a Surface *LAPTOP* Go 2, the keyboard is attached at all times. 

I was surprised, just ran on my test machine, and the attached keyboard DOES work with the UEFI shell from:
https://github.com/tianocore/edk2/blob/edk2-stable201903/ShellBinPkg/UefiShell/X64/Shell.efi

NiKiZe, 2023-12-13:
Which iPXE binary are you using? Have you built it yourself? Does it have USB support enabled (if it does it takes over all USB from firmware, including keyboard)

keithga, 2023-12-14:
We are most likely using the 2Pint snponly_drv_x64.efi binary which has support for USB networking drivers:
o	CDC-ECM USB Ethernet driver
o	USB RNDIS Ethernet driver
o	CDC-NCM USB Ethernet driver
I will try to re-run using the 2Pint snponly_x64.efi binary, when I get a chance.

NiKiZe, 2023-12-14:
In that case only USB devices should be disconnected, you will need to be able to build and run debug builds of iPXE to try and figure out if this is USB related or not.

keithga, 2024-01-08:
We were able to get a hold of a Surface Laptop Go 2 today, and ran some tests.

When booted from our "snponly_x64.efi" file, we *ARE* able to type using the build-in keyboard.
When booted from our "snponly_usb_x64.efi" file, we are NOT able to type using the build-in keyboard.
The snponly_usb_x64.efi program is built with the following extra drivers:
    --ecm--acm--ncm 

So, yes it's entirely possible that these USB Network drivers are somehow interfering with the on-board USB Keyboard. 
We will use the non-USB Network driver "snponly_x64.efi" ipxe program on the Surface Laptop Go 2 as a work around for now.

Thanks

NiKiZe, 2024-01-09:
Full lsusb -v might be helpful here (from Linux)

aretsan, 2024-02-18:
Also encountered the same problem here. 

ipxe.efi: keyboard works properly.
ipxe--ecm.efi: keyboard does not work.
ipxe--acm.efi: keyboard does not work.
ipxe--ncm.efi: keyboard does not work.
ipxe--ecm--ncm--acm.efi: keyboard does not work.

and by the way this is the keyboard in my laptop:
```
Bus 001 Device 002: ID 0b05:1866 ASUSTek Computer, Inc. N-KEY Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1866 N-KEY Device
  bcdDevice           40.22
  iManufacturer           1 ASUSTek Computer Inc.
  iProduct                2 N-KEY Device
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     331
          Report Descriptors:
            ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
Device Status:     0x0001
  Self Powered
```

Sisterjohng, 2025-03-29:
May be to late on this or you may have already found it, but could be worth checking out https://github.com/linux-surface & their repos. Sounds like from what I've seen you could solve some of your issues.