ipxe/ipxe #977: Freeze during "initialising devices" in Virtualbox when chainloading

mcb30, 2023-06-22:
| Same problem with Virtualbox 7.0.8 and all Intel 1000 cards in EFI mode

_Originally posted by @Reonaydo in https://github.com/ipxe/ipxe/issues/624#issuecomment-1591551656_

mcb30, 2023-06-22:
Testing with some extra debug shows that the freeze occurs when we attempt to open the PCI I/O protocol with `BY_DRIVER` attributes:

```c
	/* Open PCI device */
	if ( ( rc = efipci_open ( device, ( EFI_OPEN_PROTOCOL_BY_DRIVER |
					    EFI_OPEN_PROTOCOL_EXCLUSIVE ),
				  efipci ) ) != 0 ) {
		DBGC ( device, "EFIPCI %s could not open PCI device: %s\n",
		       efi_handle_name ( device ), strerror ( rc ) );
		DBGC_EFI_OPENERS ( device, device, &efi_pci_io_protocol_guid );
		goto err_open;
	}
````

[image]

The debug messages show that even after disconnection, the VirtualBox "E1000 network interface card Driver" still has the handle open with `BY_DRIVER` attributes.  This suggests that the freeze is caused by the platform firmware being stuck in a loop attempting to repeatedly disconnect the driver, but the driver is failing each time to close its `BY_DRIVER` handle.

mcb30, 2023-06-22:
This failure of the VirtualBox "E1000 network interface card Driver" to disconnect properly can be reproduced without any iPXE code using the UEFI shell:

[image]

mcb30, 2023-06-22:
I have found the root cause bug within the VirtualBox VBoxPkg E1kNetDxe driver.

The VirtualBox driver's `Stop()` method will do absolutely nothing unless called with a non-zero number of children:

```c
STATIC
EFI_STATUS
EFIAPI
E1kNetDriverBindingStop (
  IN EFI_DRIVER_BINDING_PROTOCOL *This,
  IN EFI_HANDLE                  ControllerHandle,
  IN UINTN                       NumberOfChildren,
  IN EFI_HANDLE                  *ChildHandleBuffer
  )
{
  if (NumberOfChildren > 0) {
    //
    // free all resources for whose access we need the child handle, because
    // the child handle is going away
    //
    EFI_STATUS                  Status;
    EFI_SIMPLE_NETWORK_PROTOCOL *Snp;
    E1K_NET_DEV                 *Dev;
    EFI_TPL                     OldTpl;

    ASSERT (NumberOfChildren == 1);

    ......

      gBS->CloseProtocol (
             ControllerHandle,
             &gEfiPciIoProtocolGuid,
             This->DriverBindingHandle,
             ControllerHandle
             );

    ......
  }

  return EFI_SUCCESS;
}
```

This relies on the driver's `Start()` method having opened the underlying PCI I/O protocol handle with `BY_CHILD_CONTROLLER` attributes on behalf of its newly created SNP protocol handle, so that the UEFI firmware can be aware of the parent-child relationship and therefore pass in a list of child handles to the `Stop()` method.  However, the VirtualBox driver's `Start()` method never performs this `BY_CHILD_CONTROLLER` open.

This bug can be observed directly in the UEFI shell's list of devices:

[image]

Note that device `AB` (the SNP protocol handle) is **not** shown as a child of device `9E` (the PCI I/O protocol handle).

The net effect is that the VirtualBox driver's `Stop()` method is a pure no-op: it will never close the PCI I/O protocol handle as required.

veeyoung, 2024-03-10:
solved it by chaging network adapter type from intel pro 1000 to pcnet fast

neitsab, 2025-01-14:
> solved it by chaging network adapter type from intel pro 1000 to pcnet fast

I tried the 6 difference interface types offered, and only those 3 had PXE capability in EFI mode: `Intel PRO/1000 MT Desktop (82540EM)`, `Intel Server PRO/1000 MT (82545EM)` and `virtio-net`. None of them went further than

[image]

This is with VBox 7.1.4 on Arch Linux connected via Ethernet. The VM interface is configured in bridge mode so as to appear as a regular machine on the network and use the company's IPXE server, which we use to image physical machines (chainloading to FOG project).

[Lots of EFI/PXE issues with the latest VBox](https://forums.virtualbox.org/viewtopic.php?t=112419), dang it.