ipxe/ipxe #983: Operation not supported yet HTTP protocol is listed

alonsocamaro, 2023-06-30:
Hi I´m using the following embedded ipxe script loaded as initrd from grub

```
#!ipxe
dhcp
initrd --name initrd http://api.openshift.com/api/assisted-images/images/1a120a66-8881-42a4-9ba0-a8ccfe4a752b/pxe-initrd?arch=x86_64&image_token=XYZversion=4.12
kernel http://api.openshift.com/api/assisted-images/boot-artifacts/kernel?arch=x86_64&version=4.12 initrd=initrd coreos.live.rootfs_url=http://api.openshift.com/api/assisted-images/boot-artifacts/rootfs?arch=x86_64&version=4.12 random.trust_cpu=on rd.luks.options=discard ignition.firstboot ignition.platform.id=metal console=tty1 console=ttyS1,115200n8 coreos.inst.persistent-kargs="console=tty1 console=ttyS1,115200n8"
boot
```

DHCP succeeds but then it fails to load from HTTP, see next:

[image]

The grub menu entry I´m using is the following which seems alright:

```
menuentry "iPXE" {

        linux /ipxe.lkrn
        initrd /openshift.ipxe
}
```

where the ipxe.lkrn is meant to support INITRD according to https://ipxe.org/embed so I´m unsure what I'm missing here

I´ve tried ipxe.lkrn from RHEL 8.3 and from the current git with the same result. 

Given that I´m using pretty standard features I can imagine this is a user error but the error message doesn´t make sense I thought it was worth reporting.

I will be opening a support case with RH as well

NiKiZe, 2023-06-30:
The error and URL in image is Operation not supported https://ipxe.org/3c092003
From the features flag we can see that you have HTTP support.
Please verify which protocol is used.

Could it be that you have a redirect to https ?
If not, please break up the steps in your script and try them one by one, at least the part around the initrd.

Also follow the above error URL, and try to follow the steps there.

alonsocamaro, 2023-06-30:
I created a local repository and worked like a charm. 

It could be redirect which is happening

```
> Host: api.openshift.com
> User-Agent: curl/7.61.1
> Accept: */*
> 
< HTTP/1.1 302 Found
< content-length: 0
< location: https://api.openshift.com/api/assisted-images/boot-artifacts/rootfs?arch=x86_64&version=4.12
< cache-control: no-cache
< 
```

but checking the code it is meant to be supported. What flag do you suggest to recompile with to provide more diagnostics?

NiKiZe, 2023-06-30:
Thanks for confirming
You need https://ipxe.org/buildcfg/download_proto_https for https.

closed 2023-06-30 by NiKiZe