ipxe/ipxe #987: MFS set too low on Intel 40GBE?

srhb, 2023-07-03:
Hi

I'm a little unclear on whether this behaviour is expected, so please bear with me as I try to explain.

We recently upgrade from 70995397e5bdfd3431e12971aa40630c7014785f (good) to ae435cb4cc78183814f867686d278885db2988f8 (bad) and somewhere between them, our intel 40GBE NICs ended up with a too low MFS set, even if our ipxe script is just a fall-through "exit 1" which immediately proceeds to boot whatever is next in the boot order.

Linux, once booted after the "bad" rev of ipxe was briefly loaded, reports

```
Jul 03 12:44:33 container-p01 kernel: i40e 0000:01:00.1:
MFS for port 1 has been set below the default: 600
```

(we expected to be able to run jumbo frames at this point)

Everything behaves as expected on the "good" rev.

I've been trying to bisect the issue more completely, but unfortunately most of the revs simply crash so it's unclear what's going on. If I simply add a `printf("-----MFS: %d------", intelxl->mfs);` here: https://github.com/ipxe/ipxe/blob/master/src/drivers/net/intelxl.c#L1414-L1416 (feel free to mentally insert "i have no idea what I'm doing" doge meme here)

I can tell it's being set to 1536 on both the good and bad rev, so I have no idea where 600 is coming from!

Now, this issue only arises because ipxe is configuring via dhcp and we did not set a MTU (before or after), with eg dhcp setting mtu to 9000 the problem also goes away (mfs is set to 9216 according to the printf, and persists into the OS) -- but it still seems so weird that the MFS somehow is different between the two versions, and that it survives into the booted OS after ipxe despite the fact that we are manually configuring an MTU (static config of the NIC in the OS) but I suppose that may be a bug in the linux driver itself.

Please let me know if I can do anything else to explain this behaviour. It was a bit of a surprise, but I'm not sure that what ipxe is doing is actually wrong.

mcb30, 2023-12-28:
> I can tell it's being set to 1536 on both the good and bad rev, so I have no idea where 600 is coming from!

The `600` is presumably a hex value reported without a `0x` prefix, since `0x600 == 1536`.

I don't honestly know what the "correct" behaviour would be.  iPXE assumes that in a non-virtual environment (specifically: when using a NIC that isn't a PCI virtual function or a virtual NIC such as `netfront`) then the MTU is 1536 unless specified otherwise.  We therefore allocate only 1536-byte RX buffers, and so configure the NIC to use only that buffer size.

The Intel 40GbE does have two separate places where MFS is configured: `intelxl_admin_mac_config()` (which sets the maximum frame length that will be accepted as a valid packet by the MAC) and `intelxl_context_rx()` (which sets the size of receive descriptor buffers).  It's possible that everything would still work if we were to set only the latter, and use the default value (whatever that is) in `intelxl_admin_mac_config()` (or just omit that function entirely).  You could try that and see if it fixes the problem, and report back.

ErwanAliasr1, 2024-02-27:
Just faced the issue here under Linux where some i40e devices detect the MFS is too low but don't set it :/
This ipxe behavior broke the driver behavior assuming the MFS is properly set. I'll send a patch to the kernel to avoid accepting setting up the MTU if its greater than the MFS. 
I really wonder why iPXE tries to enforce the MFS which is the greatest L2 packet to be received. Why not assuming the firmware set it properly ?

mcb30, 2024-03-19:
> That said, I still don't understand the _why_ iPXE has to reset the MFS to a smaller value than the default (firmware). This is not part of the commit message. I wonder if the previous code using PRTGL was really changing the configuration.

The `PRTGL_SA[HL]` register holds the 48-bit MAC address and 16-bit MFS in the same 64-bit register, so the code that wrote to `PRTGL_SA[HL]` was definitely setting the MFS since it had no way to avoid doing so.

The commit was part of a series to add support for the Intel 100Gig NICs.  The `PRTGL_SA[HL]` registers are not present in the same form on the 100Gig NICs, but the relevant admin queue commands are identical for both 40Gig and 100Gig, and so by switching to using the admin queue commands instead we can share code between the two drivers.

iPXE needs to set the RX buffer length, so that the hardware doesn't attempt to DMA packets beyond the end of the allocated RX buffer.  We also potentially need to set the MFS if we are attempting to use jumbo frames, since we don't know what value the MFS will have when we first start driving the card.

If the MFS is higher than the RX buffer length then the hardware may attempt to split packets across multiple RX descriptors, which the iPXE driver will not support.  I'd have to recheck the datasheet to figure out what would actually happen in this scenario.  It's possible that we'd just see the fragments as separate packets.

I'm not sure why the Linux (or other) drivers would **not** set the MFS.  There seems to be no particular reason to assume that it has a sensible power-on default value, since the correct value is a property of the network infrastructure to which the NIC is attached, not a property of the NIC itself.

ErwanAliasr1, 2024-03-19:
Yeah I agree the Kernel should also ensure the proper size of the MFS, trying to figure out how to properly handle it in the Kernel. They consider the firmware set properly, which is said in the x710 spec to be the default (9728).

For now, my patch which prevents to set jumbo frames when iPXE sets the MFS to a lower value than the default is about to be merged. At least, users will avoid falling into this issue. 

Thx for the explanation about why not considering the default was not considered, I was lacking this information in the commit message.

mcb30, 2024-03-19:
I've looked through the relevant parts of the XL710 spec.  If we do not set MFS, and a jumbo frame is received, then it will be split across multiple RX buffers.  iPXE will interpret the last fragment of the packet as a truncated packet (which will be discarded as nonsensical by the rest of the stack), and will interpret any preceding fragments in an undefined way since it will assume that various flags in the RX writeback descriptor are valid (which is not the case for non-final fragments).

We could potentially reassemble fragments into complete packets using `iob_concatenate()`, but I'm not sure it's worth doing.  We allocate very few RX buffers and so there's a good chance that portions of a packet would be lost.  The datasheet does not seem to describe how this situation is handled.  It's possible that the hardware doesn't implement low latency RX cut-through and so won't start consuming RX descriptors until the whole packet length is known, in which case it would probably cleanly drop the whole packet, but I don't see anything in the spec that directly addresses this.

We could fairly easily add logic in `intelxl_poll_rx()` to drop fragmented packets: whenever a completed descriptor does not have the EOP bit set, then both that descriptor and the next completed descriptor should be treated as receive errors.  (This would require maintaining a state flag in `struct intelx_ring`, since the next completed descriptor may not be processed in the same invocation of `intelxl_poll_rx()`.)

This would allow us to function correctly even if we leave the MFS with its default value: any received frames above our expected MTU would then be cleanly treated as (multiple) receive errors.  However, in my opinion it would be **adding** a bug to iPXE to leave the MFS untouched since (as I mentioned above) we have no guarantees that it already holds a sensible value.

It may be worth adding this bug for the sake of compatibility with other operating systems, if every other operating system handles MFS incorrectly and will continue doing so.

Zariel, 2025-08-25:
Is it possible to fix this after booting via ipxe?

NiKiZe, 2025-08-26:
> Is it possible to fix this after booting via ipxe?

If some driver in some OS does not deal with this case, it should be considered a bug in that driver. It should not be assumed that the firmware has a valid setting for the specific network.

With that said @mcb30 any chance there could be a MFS register restore on iPXE driver exit?