ipxe/ipxe #996: hanging sanboot iscsi:xxx.xxx.xxx.xxx:::5:iqn.xxxx-xx.com.xxx:xx.xx.xx

hab-dad, 2023-07-12:
From time to time we have a problem booting a system.

the boot would hang printing the following on screen when booting disk1:

Booting from SAN device 0x80

After analysis and testing we found that in ipxe the following command was being executed:
sanboot iscsi:xxx.xxx.xxx.xxx:::5:iqn.xxxx-xx.com.xxx:xx.xx.xx || goto failed

this would hang forever. 

when rebooting and going into ipxe and explicitly giving a filename like below:

sanboot --filename \EFI\redhat\shimx64.efi \
          iscsi:xxx.xxx.xxx.xxx:::5:iqn.xxxx-xx.com.xxx:xx.xx.xx

the system would normally boot. For some reason sanboot is not able to find the file or ?

we found a workaround :
The way to get the system booting again is to boot to another disk (disk0) and 
searching for the entry in efibootmgr that is for disk1. when this entry is delete then the system would boot normally.

I have the idea that something goes wrong in the firmware. I found a comment online which says:

(sanboot is just sanhook with the extra "tell firmware to boot it", so should be no difference)

can somebody please give me a hint.
I have already checked the /sys/firmware/efi/efivars (did a hex dump). for the vars when booting was possible and when booting was not possible anymore. I didn't see a difference.

NiKiZe, 2023-07-12:
As you say, "sanboot is just telling firmware to boot it"
If you have a intel based x64 machine the default boot path is "\EFI\BOOT\BOOTX64.EFI", does this file exist on your disks?
If you want to boot anything else, such as "\EFI\redhat\shimx64.efi" you need to specify that.

I would however expect your firmware to return with something like "file not found" if bootx64.efi fails, but maybe it is that binary itself that fails?
Have you checked anything about what is going on at the wire? (tcpdump/wireshark)
You might also want to consider building iPXE with debug enabled to figure out what your firmware is trying to do, But that sometimes gets more confusing than helpful in this scenario.

hab-dad, 2023-07-12:
Thank you very much for your quick reply.

I will check if \EFI\BOOT\BOOTX64.EFI is available. 
Do you know why it would boot if we remove the entry from the efibootmgr.
I assumed that sanboot searches  in the nvram for the efi entries and that it for some reason gets stuck in the 
entry that was revering to the lun that we want to boot.

NiKiZe, 2023-07-12:
iPXE asks firmware to boot the device.
So, if firmware has an entry for the device, then it might use those settings, while iPXE itself does not use settings/nvram from firmware.

hab-dad, 2023-07-13:
I tried using --filename "\EFI\BOOT\BOOTX64.EFI" and this did not work. 
while \EFI\redhat\shimx64.efi works. 
To me it looks like I have a problem in the bios.
Thank you very much for your help.

mcb30, 2023-07-18:
The default boot filename for UEFI x86_64 is `\EFI\BOOT\BOOTX64.EFI`.  This is the filename that will be tried by UEFI on removable disks, or on unknown fixed disks.

UEFI does allow for boot entries to be created and stored in non-volatile firmware variable storage.  These can point to different boot filenames on the disk.  Older distros tended to use these to point to the distro's choice of boot filename (e.g. `\EFI\redhat\shimx64.efi`).  Newer distros tend to simply install the boot file as the default filename (i.e. installing `shimx64.efi` as the file `\EFI\BOOT\BOOTX64.EFI`), since this means that the disk will be bootable even if the associated non-volatile variable storage is missing, e.g. if the disk is physically moved to a new machine.

A SAN disk is logically a removable disk, since it is not permanently attached to the machine.  iPXE will ignore any boot entries defined in the non-volatile variable storage, and will attempt to boot the removable disk default boot filename `\EFI\BOOT\BOOTX64.EFI`.  This will work as expected for newer distros, which will always install their bootloader using this filename.

As documented at https://ipxe.org/cmd/sanboot you can use the `--filename` option to override this choice of boot filename in order to boot older distros that do not install their bootloader using the default boot filename.

Closing since everything is working as designed.  :slightly_smiling_face:

closed 2023-07-18 by mcb30