ipxe/ipxe #997: make target bin-x86_64-efi/ipxe.usb fails with Error: operand size mismatch for `push'

jstncno, 2023-07-14:
Hello,

I have been building ipxe from source using `make bin-x86_64-efi/ipxe.usb`. Lately I've been receiving the following errors:

```
$ make bin-x86_64-efi/ipxe.usb
  ...
  [BUILD] bin-x86_64-efi/bootsector.o
arch/x86/image/bootsector.c: Assembler messages:
arch/x86/image/bootsector.c:85: Error: operand size mismatch for `push'
make: *** [Makefile.housekeeping:964: bin-x86_64-efi/bootsector.o] Error 1
```

Could anybody help me troubleshoot this issue? Thanks

NiKiZe, 2023-07-14:
If this has worked previously, please use instructions at https://ipxe.org/howto/bisect to identify commits introducing issues.
Also make sure you don't have any local modifications.

If you can't find a working commit at all, please make sure to include versions of your toolchain, especially gcc.

boot.ipxe.org produces builds every hour, so there shouldn't be anything that could break for everyone being commited to the tree.

jstncno, 2023-07-14:
Thanks @NiKiZe for the quick response. It seems it has to do with my toolchain as I was able to successfully build in a vanilla Ubuntu environment.

My working env has gcc v10.2.1 while the non-working env has gcc v10.4.0. Is there a way to check for the correct toolchain?

mcb30, 2023-07-18:
This looks like a gcc bug.  There are precisely three `push` instructions within that inline assembly (including the implicit `push $1f` created by the `REAL_CODE()` macro), and none of them are problematic.

From your description of this as being fixed by using a "vanilla Ubuntu environment", I'm guessing that you may have a locally built `gcc` in your original environment, in which case that is most likely to be the source of the problem.

Closing: please reopen if you are able to reproduce the error in some kind of "standard" environment.

jstncno, 2023-07-18:
@mcb30 thanks for your response.

With some help from some colleagues, we were able to determine the root cause.

It seems that this is likely a change in the `binutils` package (specifically the assembler, `as`) rather than `gcc`.

In my working environment (I mistakenly mentioned "vanilla Ubuntu" instead of "vanilla Debian"), running `as --version` shows version `2.35.2`:
```
GNU assembler (GNU Binutils for Debian) 2.35.2
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `x86_64-linux-gnu'.
```

In my non-standard, non-working environment, `as --version` shows version `2.40.50.20230611`:
```
GNU assembler (GNU Binutils for Debian) 2.40.50.20230611
Copyright (C) 2023 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `x86_64-linux-gnu'.
```

It makes to keep this ticket closed since this is still a "non-standard" environment, but wanted to post this info if this issue ever propagates.

Thanks again!

sonic42, 2023-08-03:
Hi
The problem persists with new released binutils package 2.41 https://www.gnu.org/software/binutils/
 as --version
GNU assembler (GNU Binutils) 2.41
Copyright (C) 2023 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `x86_64-pc-linux-gnu'``

NiKiZe, 2023-08-03:
Is this on one machine or multiple?
What is needed to reproduce so we can create a VM and test?

sonic42, 2023-08-03:
After to upgrading a LFS based distri toolchain with fresh released binutils  2.41 / gcc 13.2 and glibc 2.38, some packages fails build. I'm using a Debian based VM for development. This are the first tests with this new  binutils, gcc and glibc releases. After some testing use of previous versions of gcc and glibc to compile ipxe , the error raises the new binutils package.  The error is currently on one machine. Tests in a second manjaro / arch based build environment will follow...

sonic42, 2023-08-04:
The build are successful.  Thanks.

jstncno, 2023-08-04:
Can this issue be reopened, since this reproducible with `as` v2.40+ ?

NiKiZe, 2023-08-04:
Is there any bug or documentation in/from vinutils on this?

ipaqmaster, 2023-08-05:
That patch resolved it for me. Thanks @jstncno

mcb30, 2023-09-04:
I am unable to reproduce the issue.  I get a successful build using binutils 2.41:
```console
$ as --version
GNU assembler version 2.41-5.fc40
```

I'm happy to look into this further, but only if someone can provide me with a simple way to reproduce the problem.

c---, 2023-09-04:
Building using Arch Linux I get the error:
```
$ git clone https://git.ipxe.org/ipxe.git
$ cd ipxe/src
$ make bin-x86_64-efi/ipxe.efi

  [BUILD] bin-x86_64-efi/bootsector.o
arch/x86/image/bootsector.c: Assembler messages:
arch/x86/image/bootsector.c:85: Error: operand size mismatch for `push'
make: *** [Makefile.housekeeping:964: bin-x86_64-efi/bootsector.o] Error 1
```

```
$ as --version
GNU assembler (GNU Binutils) 2.41.0
```

mcb30, 2023-09-05:
OK, I have reproduced the problem.  My fault for using a plain `make` to test, which doesn't exercise the 64-bit build path.  :man_facepalming: 

I have a simplified test case, independent of iPXE:
```shell
as -o /dev/null <<'EOF'
    .text
    push $1f
1:
EOF
```

I have bisected this down to binutils commit https://github.com/bminor/binutils-gdb/commit/5cc007751cdb8ea713c98294e37cd447c12c9bab ("x86: further adjust extend-to-32bit-address conditions").  I don't fully understand what's going on in that commit, but it definitely does relate to operand sizes.  That commit first appeared in binutils-2.41, which matches the issue reports.

PR #1036 (2023-09-05): [librm] Use explicit operand size when pushing a label address
We currently use "push $1f" within inline assembly to push the address of the real-mode code fragment, relying on the assembler to treat this as "pushl" for 32-bit code or "pushq" for 64-bit code.

As of binutils commit 5cc0077 ("x86: further adjust extend-to-32bit- address conditions"), first included in binutils-2.41, this implicit operand size is no longer calculated as expected and 64-bit builds will fail with

  Error: operand size mismatch for `push'

Fix by adding an explicit operand size to the "push" instruction.

Fixes: # [... cut]
closed 2023-09-05 by mcb30