ipxe/ipxe #999: The machine is fully equipped with network cards, but selecting one of the ports in BootManager Menu failed to boot IPxe

ccBeHappy, 2023-07-17:
Hello，
We found compatibility issues with the ipxe version and some network cards. When the number of PXE boot entries exceeds a certain number, it can cause some network cards to experience ipxe boot exceptions. Hope our experts can help with the analysis. Thank you very much...

For example:
the boot options are one hdd + 124 pxe，XP380 slot8 can not boot， but one hdd + 90 pxe， XP380 slot8 can boot.

The error printed content is as follows：
net60: fe80::3209:f9ff:fe22:43c1/64 (inaccessible)
net61: fe80::3209:f9ff:fe22:43c2/64 (inaccessible)
net62: 192.168.20.216/255.255.255.0 gw 192.168.20.1
net62: 2001:db8:f00f:cafe:daff:c3ff:fe80:c5c/64 gw fe80::6c2:c60f:8acf:29c1
net62: fe80::daff:c3ff:fe80:c5c/64
net63: fe80::daff:c3ff:fe80:c5d/64 (inaccessible)
Next server: 192.168.20.1
Filename: http://192.168.20.1/real_boot_script.php
http://192.168.20.1/real_boot_script.php.................. Connection timed out (https://ipxe.org/4c0a6092)
No more network devices

We have tried two versions of ipxe and both have issues，the versions of ipxe we have tried are :
(1)one version was got from community  ,timeline:2023-07-13 08:00 1.0M
(2)the other is iPXE 1.21.1+ (gd3c89).

The issue is quite urgent and we look forward to receiving your reply. 
Thank you very much.

NiKiZe, 2023-07-17:
Exact duplicate of #998 (but improvement with text instead of picture)
As stated there, don't open duplicate issues, and if there is issues, at least provide output of ifstat, after following the error URL.

In this case https://ipxe.org/4c0a6092 makes it clear that the TCP connection fails, is the server listening properly?, does tcpdump or similar show retries?

You have provided which version of iPXE you have tried, but not which binary. But let's start with ifstat.

You can reach out for commercial support thru https://ipxe.org/contact

ccBeHappy, 2023-07-17:
Sorry, I didn't notice that other colleagues also raised this issue. I have closed this one. Thank you very much

NiKiZe, 2023-07-17:
Closing until relevant information is provided, and the OP has checked what happens with their TCP stream.

ccBeHappy, 2023-07-19:
1) iPXE  binary name:
ipxe-gc30b7.efi
ipxe-gd3c89.efi
2)  another error information：
[2023-07-15 15:50:08]  net46: fe80::daff:c3ff:fe80:c5d/64 (inaccessible)
[2023-07-15 15:50:08]  Next server: 192.168.1.54
[2023-07-15 15:50:09]  Filename: ipxe.cfg
[2023-07-15 15:50:10]  tftp://192.168.1.54/ipxe.cfg.................. Connection timed out (https://ipxe.org/4c126092)
[2023-07-15 15:50:26]  No more network devices

It means TFTP transfer has timed out...

NiKiZe, 2023-07-19:
And if you use snponly.efi or snö.efi instead?

ccBeHappy, 2023-07-20:
Using the binary of snp.efi or snponly.efi, network ports that were previously not bootable can now be booted. Thank you~

ifstat log is as follow:
[root@localhost Desktop]# ifstat -tT -i ens4f0np0
  Time        ens4f0np0             Total       
HH:MM:SS   KB/s in  KB/s out   KB/s in  KB/s out
17:28:14      0.00      0.00      0.00      0.00
17:28:15      0.00      0.00      0.00      0.00
17:28:16      0.00      0.00      0.00      0.00
17:28:17      0.08      0.00      0.08      0.00
17:28:18      0.00      0.00      0.00      0.00
17:28:19      0.00      0.00      0.00      0.00
17:28:20      0.00      0.00      0.00      0.00
17:28:21      0.08      0.00      0.08      0.00
17:28:22      0.18      0.00      0.18      0.00
17:28:23      0.00      0.00      0.00      0.00
17:28:24      0.00      0.00      0.00      0.00
17:28:25      0.00      0.00      0.00      0.00
17:28:26      0.13      0.00      0.13      0.00
17:28:27      0.00      0.00      0.00      0.00
17:28:28      0.00      0.00      0.00      0.00
17:28:29      0.00      0.00      0.00      0.00
17:28:30      0.00      0.00      0.00      0.00
17:28:31      0.13      0.00      0.13      0.00
17:28:32      0.38      0.05      0.38      0.05
17:28:33      0.00      0.33      0.00      0.33
17:28:34      0.00      0.05      0.00      0.05
17:28:35      0.00      0.05      0.00      0.05
17:28:36     42.04   1037.89     42.04   1037.89
17:28:37      0.08      0.06      0.08      0.06
17:28:38      0.00      0.00      0.00      0.00
17:28:39      0.00      0.00      0.00      0.00
17:28:40      0.00      0.00      0.00      0.00
17:28:41      0.00      0.00      0.00      0.00
17:28:42      0.13      0.05      0.13      0.05
17:28:43      0.06      0.10      0.06      0.10
17:28:44      0.00      0.05      0.00      0.05
17:28:45      0.00      0.00      0.00      0.00
  Time        ens4f0np0             Total       
HH:MM:SS   KB/s in  KB/s out   KB/s in  KB/s out
17:28:46      0.00      0.00      0.00      0.00
17:28:47      0.00      0.00      0.00      0.00
17:28:48      0.00      0.00      0.00      0.00
17:28:49      0.00      0.00      0.00      0.00
17:28:50      0.00      0.00      0.00      0.00
17:28:51      0.00      0.00      0.00      0.00
17:28:52      0.00      0.00      0.00      0.00
17:28:53      0.00      0.00      0.00      0.00
17:28:54      0.00      0.00      0.00      0.00
17:28:55      0.00      0.00      0.00      0.00
17:28:56      0.00      0.00      0.00      0.00
17:28:57      0.00      0.00      0.00      0.00
17:28:58      0.00      0.00      0.00      0.00
17:28:59      0.00      0.00      0.00      0.00
17:29:00      0.00      0.00      0.00      0.00
17:29:01      0.00      0.00      0.00      0.00
17:29:02      0.00      0.00      0.00      0.00
17:29:03      0.00      0.00      0.00      0.00
17:29:04      0.00      0.00      0.00      0.00
17:29:05      0.00      0.00      0.00      0.00
17:29:06      0.00      0.00      0.00      0.00
17:29:07      0.00      0.00      0.00      0.00
17:29:08      0.00      0.00      0.00      0.00
17:29:09      0.00      0.00      0.00      0.00
17:29:10      0.00      0.00      0.00      0.00
17:29:11      0.00      0.00      0.00      0.00
17:29:12      0.00      0.00      0.00      0.00
17:29:13      0.00      0.00      0.00      0.00
17:29:14      0.00      0.00      0.00      0.00
17:29:15      0.00      0.00      0.00      0.00
[... 225 lines cut ...]
17:32:47      0.00      0.00      0.00      0.00
17:32:48      0.00      0.00      0.00      0.00
17:32:49      0.00      0.00      0.00      0.00
17:32:50      0.00      0.00      0.00      0.00
17:32:51      0.00      0.00      0.00      0.00
17:32:52      0.00      0.00      0.00      0.00
17:32:53      0.00      0.00      0.00      0.00
17:32:54      0.00      0.00      0.00      0.00
17:32:55      0.00      0.00      0.00      0.00
17:32:56      0.00      0.00      0.00      0.00
17:32:57      0.00      0.00      0.00      0.00
17:32:58      0.00      0.00      0.00      0.00
17:32:59      0.00      0.00      0.00      0.00
17:33:00      0.00      0.00      0.00      0.00
17:33:01      0.00      0.00      0.00      0.00
  Time        ens4f0np0             Total       
HH:MM:SS   KB/s in  KB/s out   KB/s in  KB/s out
17:33:02      0.00      0.00      0.00      0.00
17:33:03      0.00      0.00      0.00      0.00
17:33:04      0.00      0.00      0.00      0.00
17:33:05      0.00      0.00      0.00      0.00
17:33:06      0.00      0.00      0.00      0.00
17:33:07      0.00      0.00      0.00      0.00
^C
[root@localhost Desktop]#

NiKiZe, 2023-07-20:
That is not the output of ifstat (the command in iPXE)

ccBeHappy, 2023-07-20:
Using the binary of snp.efi or snponly.efi, network ports that were previously not bootable can now be booted.  --- Can this indicate that  maybe there is a bug in the nic driver in ipxe.efi ?

ccBeHappy, 2023-07-20:
Using the binary of snp.efi or snponly.efi, network ports that were previously not bootable can now be booted. --> Our problem should be related to the nic driver in ipxe. efi. What should we do next ？

ccBeHappy, 2023-07-20:
ifstat log ，we will retrieve the log again.

ccBeHappy, 2023-07-20:
iPXE> ifstar t
net0: 5c:6f:69:2a:93:c2 using 14e4-16D7 on 0000:17:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net1: 5c:6f:69:2a:93:c3 using 14e4-16D7 on 0000:17:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net2: 60:7e:cd:5a:3f:de using x550t2 on 0000:3d:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net3: 60:7e:cd:5a:3f:df using x550t2 on 0000:3d:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net4: 34:73:79:2d:57:bd using i350 on 0000:3f:00.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net5: 34:73:79:2d:57:be using i350 on 0000:3f:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net6: 34:73:79:2d:57:bf using i350 on 0000:3f:00.2 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net7: 34:73:79:2d:57:c0 using i350 on 0000:3f:00.3 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net8: e8:eb:d3:ba:ad:8e using ConnectX-5 on 0000:63:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net9: e8:eb:d3:ba:ad:8f using ConnectX-5 on 0000:63:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net10: 34:73:79:2d:59:51 using i350 on 0000:99:00.0 (Ethernet) [closed]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net11: 34:73:79:2d:59:52 using i350 on 0000:99:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net12: 34:73:79:2d:59:53 using i350 on 0000:99:00.2 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net13: 34:73:79:2d:59:54 using i350 on 0000:99:00.3 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net14: 10:70:fd:c3:f5:9c using ConnectX-6DX on 0000:ab:00.0 (Ethernet) [open]
  [Link:up, TX:12 TXE:7 RX:35 RXE:1]
  [TXE: 1 x "Network unreachable (https://ipxe.org/28086090)"]
  [TXE: 6 x "Error 0x2a376089 (https://ipxe.org/2a376089)"]
  [RXE: 1 x "Invalid argument (https://ipxe.org/1c056082)"]
net15: 10:70:fd:c3:f5:9d using ConnectX-6DX on 0000:ab:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net16: e4:3d:1a:fd:4b:9c using 14e4-1657 on 0001:17:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net17: e4:3d:1a:fd:4b:9d using 14e4-1657 on 0001:17:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net18: e4:3d:1a:fd:4b:9e using 14e4-1657 on 0001:17:00.2 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net19: e4:3d:1a:fd:4b:9f using 14e4-1657 on 0001:17:00.3 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net20: 34:73:79:a2:66:f6 using ConnectX-4Lx on 0001:18:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net21: 34:73:79:a2:66:f7 using ConnectX-4Lx on 0001:18:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net22: 34:73:79:2d:54:39 using i350 on 0001:3d:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
[... 81 lines cut ...]
net50: 34:73:79:2c:75:4e using ConnectX-4Lx on 0001:cf:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net51: 34:73:79:2c:75:4f using ConnectX-4Lx on 0001:cf:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net52: 30:09:f9:22:43:bf using NII on NII-0000:2a:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net53: 30:09:f9:22:43:c0 using NII on NII-0000:2a:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net54: 30:09:f9:22:43:c1 using NII on NII-0000:2a:00.2 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net55: 30:09:f9:22:43:c2 using NII on NII-0000:2a:00.3 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net56: d8:ff:c3:80:0c:5c using SNP on SNP-0000:bd:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net57: d8:ff:c3:80:0c:5d using SNP on SNP-0000:bd:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
iPXE>

NiKiZe, 2023-07-20:
Are you building a switch or something?
You seem to have every kind of 10g+ nic available.
Still your ifstat output does not show net62 which was mentioned in the initial error report.

I don't see that we can do much here.

ccBeHappy, 2023-07-20:
It's net14.  
Full log is as follow:

BIOS Startup is completed @ 2023.7.20 18:25:6
EfiPxeLoadFile!

>>Booting EFI Network for IPv4(1070FDC3F59C) - SLOT8 Port1
 Pxe Timeout Setting: 4

.
  Station IP address is 192.168.20.237

  Server IP address is 192.168.20.1
  NBP filename is ipxe.efi
  NBP filesize is 1026560 BytesEfiPxeLoadFile!
EfiPxeLoadFile!

 Downloading NBP file...

  NBP file downloaded successfully.
TFTP error code 256
iPXE initialising devices...SnpReceiveFilters [BD:00:00] return Success
SnpReceiveFilters [BD:00:00] return Success
SnpReceiveFilters [BD:00:00] return Success
SnpReceiveFilters [BD:00:00] return Success
SnpReceiveFilters [BD:00:00] return Success
SnpReceiveFilters [BD:00:00] return Success
Snp shutdown
Set port disable
Set vport disable
Flush io resource.
Free tx queue resources
Free rx queue resources
Free cqe resources vir: 0x5B08A000 phy 0x5B08A000.
Free rx resources vir: 0x5B09A000 phy 0x5B09A000 depth 2048.
Free qp ctxts.
Free qps
Snp shutdown device[BD:00:00] returns Success, snm state 1
Snp stop
Snp stop returns Success [BD:00:00] snm state Success
Snp start
Snp start returns Success [BD:00:00]
SnpReceiveFilters [BD:00:01] return Success
SnpReceiveFilters [BD:00:01] return Success
SnpReceiveFilters [BD:00:01] return Success
SnpReceiveFilters [BD:00:01] return Success
SnpReceiveFilters [BD:00:01] return Success
SnpReceiveFilters [BD:00:01] return Success
Snp shutdown
Set port disable
Set vport disable
Flush io resource.
Free tx queue resources
Free rx queue resources
Free cqe resources vir: 0x5A728000 phy 0x5A728000.
Free rx resources vir: 0x5A738000 phy 0x5A738000 depth 2048.
Free qp ctxts.
Free qps
Snp shutdown device[BD:00:01] returns Success, snm state 1
Snp stop
Snp stop returns Success [BD:00:01] snm state Success
Snp start
Snp start returns Success [BD:00:01]
ok

iPXE 1.21.1+ (gc1834) -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP iSCSI TFTP VLAN SRP AoE EFI Menu

Press Ctrl-B for the iPXE command line...                                         net14: 10:70:fd:c3:f5:9c using ConnectX-6DX on 0000:ab:00.0 (Ethernet) [open]
  [Link:up, TX:0 TXE:1 RX:0 RXE:0]
[... 217 lines cut ...]
net50: 34:73:79:2c:75:4e using ConnectX-4Lx on 0001:cf:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net51: 34:73:79:2c:75:4f using ConnectX-4Lx on 0001:cf:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net52: 30:09:f9:22:43:bf using NII on NII-0000:2a:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net53: 30:09:f9:22:43:c0 using NII on NII-0000:2a:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net54: 30:09:f9:22:43:c1 using NII on NII-0000:2a:00.2 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net55: 30:09:f9:22:43:c2 using NII on NII-0000:2a:00.3 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net56: d8:ff:c3:80:0c:5c using SNP on SNP-0000:bd:00.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
net57: d8:ff:c3:80:0c:5d using SNP on SNP-0000:bd:00.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (https://ipxe.org/1a086194)]
iPXE>

mcb30, 2023-07-20:
Why does this hardware have so many different NICs installed?!  (There's nothing technically wrong with that, but it's certainly bizarre.)

NiKiZe, 2023-07-20:
Since it is unlikely that anyone else will have this setup, can you narrow it down to exactly which nic this happens on, or is this a scenario where to much resources just kills the iPXE network stack?

ccBeHappy, 2023-07-21:
Not only does a certain nic have this problem, but some nic ports also have similar phenomena.，such as BCM957414N4140C, MCX623106AN-CDAT, and XP380, and the number of boot items reached by each nic is different.

ccBeHappy, 2023-07-24:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/C00013~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/C00013~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

host   server 网线直连 使用官网最新的ipxe.efi | 1硬盘启动项+90   pxe启动项 | 1硬盘启动项+88   pxe启动项
-- | -- | --
MCX623106AN-CDAT | slot8：ipxe无法boot | ipxe可以正常boot
  |   |  
ixpe srerver接交换机--用官网最新的ipxe.efi | 1硬盘启动项+124pxe启动项 | 1硬盘启动项+88 pxe启动项
XP380  slot8 | ipxe无法boot |  
3S930  slot9 | ipxe可以boot |  
  |   |  
ixpe srerver接交换机--用官网最新的ipxe.efi (gc30b7) | 1硬盘启动项+124pxe启动项 | 1硬盘启动项+90 pxe启动项
BCM957414N4140C OCP port2 | ipxe无法boot |  
XP380  slot8 | ipxe无法boot | ipxe无法boot
MCX623106AN-CDAT slot8 |   | ipxe无法boot
  |   |  
ixpe srerver接交换机--用旧的ipxe.efi    (gd3c89) | 1硬盘启动项+124pxe启动项 | 1硬盘启动项+90 pxe启动项
BCM957414N4140C OCP port2 | ipxe可以boot |  
XP380  slot8 | ipxe无法boot | ipxe可以boot
MCX623106AN-CDAT slot8 |   | ipxe可以boot

</body>

</html>

ccBeHappy, 2023-07-24:
Could you please help me take a look? Looking forward to your reply

mcb30, 2023-07-25:
> Could you please help me take a look? Looking forward to your reply~ Thank you ~

Given the sizes (and costs) of the kind of equipment you're using, I think this has to be a commercial enquiry.  Please reach out to vendor-support@ipxe.org to discuss costs.

closed 2023-10-16 by zube[bot]