ipxe/ipxe #1016: intel driver skips ARP requests on i210

MichaelRiss, 2023-08-09:
TL;DR: I believe the "intel" driver has a bug which makes it skip ARP requests and use the Ethernet broadcast address instead of the proper Ethernet destination address.

---

I have a problem with iPXE network traffic not being routed beyond its local subnet.

Situation: the iPXE machine with IP address 10.222.215.8 (static IP configuration, no DHCP) tries to send ping packets through the gateway 10.222.215.1 to e.g. 192.168.14.1 as destination (it's a nested NAT setup).
While the machine can ping the gateway 10.222.215.1 which is in its subnet, pinging the other destination (192.168.14.1) does not work. Wireshark on the router tells me that the ping packets arrive at the router, but they don't get routed to the destination.

The router is a Linux machine running a current Fedora which might be to blame. However, if I do the ping from 10.222.215.8 with a Linux installation instead of iPXE then the ping to 192.168.14.1 works.

I looked at the differences of the ping packets iPXE <-> Linux and the most striking difference that I could find was that the ping packets from iPXE arrive with ff:ff:ff:ff:ff:ff as Ethernet destination address whereas the ping packets from Linux arrive with the proper MAC address of the router.

Looking at the Wireshark traces I can see that Linux sends out an ARP request before sending the ping packets, iPXE does not.

I then checked whether this is a general behavior of iPXE and repeated the test in a VM with the virtio-net driver. Here, iPXE sends out an ARP request and then uses the proper MAC address for the ping packets. As a result the ping packets get properly routed through the Fedora machine.

Hardware:
Asrock Rack B650D4U motherboard, with Intel i210 network controller, running BIOS 1.10.

PS. I was only mentioning ping packets because ping was what my debugging settled down to, but the issue started for me with http accesses beyond the local subnet and the http packets did get swallowed by the router analog to the ping packets.

mcb30, 2023-08-09:
It's vanishingly unlikely that a driver bug could cause this.  iPXE drivers are (by design) fairly dumb: there is simply no support for any TCP/IP offload capabilities that could be the cause of behaviour like this.

On the failing machine at the point where you could issue the `ping` command, what do you get as output from `ifstat` and `route`?

MichaelRiss, 2023-08-09:
Good call, this throws a range of error messages:
[image]

MichaelRiss, 2023-08-09:
Oh no! This was my mistake, I did set the netmask to 255.255.255.255 due to a typo. Once I set the correct netmask 255.255.255.0 everything works again. So sorry for the noise and thank you for taking time for me anyway.

closed 2023-08-09 by MichaelRiss