ipxe/ipxe #1023: ipxe.efi hangs at downloading large files over HTTP

elvinup, 2023-08-12:
[image]

I've been trying to boot to WinPE which requires me to download a boot.wim file at ~400MB. It always stops after the first block it downloads (could be 2%, 16%, 29% etc), and never progresses further afterwards.

Interestingly, I am able to download the same boot.wim file over TFTP with the latest ipxe source code, but i definitely had some data corruption issues on that boot.wim file since i'd run into this [Red screen of death](https://community.hpe.com/t5/proliant-servers-ml-dl-sl/dl360gen9-x64-exception-type-0d-general-protection-exception/td-p/7176368) half the time, and the other half it would boot properly. Perhaps because TFTP is using UDP instead of TCP and is dropping/corrupting some packets? Not sure, but anyways...

I spent a couple days wondering what the issue could be trying to get HTTP working, and said screw it eventually and tried to compile the ipxe.efi file off of [commit 1295b4acff1f2014261c40d9f9d2107ffd668d92](https://github.com/ipxe/ipxe/commit/1295b4acff1f2014261c40d9f9d2107ffd668d92) instead after reading [issue 155](https://github.com/ipxe/ipxe/issues/155). 

The ipxe.efi file i compiled from that commit now allows the boot.wim file to download over HTTP in < 1 second and boots straight to WinPE properly. Also, i don't get a red screen of death half the time, anymore it's consistent when downloading over HTTP instead of TFTP! 

I have no clue what happened between now and 2020 when that commit was made, but I just wanted to shed light on this issue that downloading a large file from HTTP seems to be busted currently.

If someone makes a patch to try to address this, I am happy to recompile that ipxe.efi file with it to test :). Until then, i'll keep using this older version.

NiKiZe, 2023-08-13:
I say it works perfectly fine as is. (That is, it works for me and many others on the hardware we have)
There is several details you are leaving out.

* Which binary are you using (ipxe.efi, snponly efi, snp.efi), as well as exact version?
* What nic is used? 10gb intel? 
* What hardware/machine is used? HPE?
* Have you done a tcpdump to check what is going on at the wire?
* Have you tried a git bisect to verify if it is the given commit, or something else that can be attributed to the issues you are seeing.

Since a few versions back of wimboot you no longer need boot.sdi or BCD

elvinup, 2023-08-14:
@NiKiZe I understand, I figure it likely hasn't been mentioned yet in the git issues lately because mostly people aren't running into the same problem with my specific HW perhaps? To answer your questions:

- I'm using ipxe.efi, compiled myself off the latest commit c1834f3 using `make bin-x86_64-efi/ipxe.efi`
- This is the exact [NIC used](https://buy.hpe.com/us/en/options/adapters/host-adapters/proliant-host-adapters/hpe-ethernet-10gb-2-port-flr-t-bcm57416-adapter/p/817721-b21), seems to be a 10gb one.
- Yes, it's an HPE machine
- I have done a tcpdump and didn't really see anything of relevance, packets seems normal until 
- Haven't done a git bisect to narrow down the issue. I plan to eventually but it's gonna take some time since quite a lot of commits have seemed to come through since then.

Thanks for the pointer on wimboot, i did not know that! Just getting my feet wet with pxe booting in general in the last couple weeks.

At the end of the day, the ipxe.efi i compiled after doing these steps made the difference 

```
git clone git://git.ipxe.org/ipxe.git
cd ipxe/src
git checkout 1295b4acff1f2014261c40d9f9d2107ffd668d92
make bin-x86_64-efi/ipxe.efi
```
And it worked perfectly and allowed me to fully download my boot.wim file from my web server.

So going back in time in git history definitely shows something went wrong somewhere, and so I doubt it has much to do with my machine, but rather some regression that's happened since then that maybe nobody noticed until now? 🤷‍♂️ Just brought up the issue so that at least someone else can see it in case they run into the same thing, or if the problem might be obvious to anyone familiar with the codebase.

Meantime I can try to figure out which commit this starts to break to pinpoint it, but until then if you or anyone else has an idea, happy to hear it!

elvinup, 2023-08-15:
Some more info. This is my [exact NIC](https://devicehunt.com/view/type/pci/vendor/14E4/device/16D8)

Also simply reverting that commit 059c4dc makes downloading over HTTP work without issues, from the latest master commit. 

But looking at the diff I don't really see what the issue could really be 😕

mcb30, 2023-09-04:
> Also simply reverting that commit [059c4dc](https://github.com/ipxe/ipxe/commit/059c4dc688cea6d9398d63ca10f5c00e94ae2159) makes downloading over HTTP work without issues, from the latest master commit.
> 
> But looking at the diff I don't really see what the issue could really be 😕 , specifically with device id 16D8 like mine. Is it possible this affected all BNXT nics?

That is strange.  There is nothing of substance in that commit that should affect behaviour.

Could you check the output of `ifstat` in known-good and known-bad builds of `ipxe.efi`?  I'm wondering if somehow the commit is causing the device to fail to be detected by the driver, causing iPXE to fall back to using the NIC's existing UNDI/NII driver.

euthuppan, 2023-09-06:
@mcb30 Here's the output of an ifstat on a 

"good" ipxe.efi file
[image]

and a "bad" ipxe.efi file
[image]

I do see a slight change in that top line using `NII` vs `14e4-16D8`. Perhaps that's the clue to the problem here?

mcb30, 2023-09-06:
> I do see a slight change in that top line using `NII` vs `14e4-16D8`. Perhaps that's the clue to the problem here?

Yes, that's a major difference.  :slightly_smiling_face: 

For some reason, the commit is causing the NIC to be recognised by the `bnxt` driver in iPXE.  This driver then seems to suffer from the problem that you have described.

In the earlier commit, the NIC is for some reason not recognised by the `bnxt` driver (or fails in some way during the device probe).  iPXE then falls back to using the `NII` driver (i.e. using API calls into the driver provided by the platform firmware, instead of driving the hardware directly).

euthuppan, 2023-09-06:
Thanks for the explanation, things are starting to make sense! 
So my machine spits this out when checking for the NIC:
```
# lspci | egrep -i --color 'network|ethernet'
5d:00.0 Ethernet controller: Broadcom Inc. and subsidiaries BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller (rev 01)
5d:00.1 Ethernet controller: Broadcom Inc. and subsidiaries BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller (rev 01)
```

It sounds like the current logic is somehow more correct, but for some reason the bnxt driver just doesn't want to work, right? I have updated the NIC firmware before and still shows the same problem, so not sure if it's solely a problem with this specific driver or bnxt ones in general that iPXE tries to handle specifically.

Do you think all bnxt drivers should just fall back to `NII` if it works anyways?

NiKiZe, 2023-09-06:
The broadcom driver has issues on many machines. Often it depends on the firmware in the machine (on Macs) where it works in older, but not newer.

Consider using snponly.efi or snp.efi, at least on these machines.
But of course if we can figure out how to fix the driver that would be great. (But this should now probably be a duplicate of that existing issue)

elvinup, 2023-09-07:
@NiKiZe Thanks for the suggestion, snp.efi seems to work just fine, now realizing that it was an option 😁 

Feel free to mark this as a duplicate then to another bad driver issue.

sploders101, 2023-10-01:
I'm not sure how useful this is, but I am also experiencing this issue. I have an AsRock Rack motherboard with the BCM57416 (same chip from the card linked earlier). Reverting https://github.com/ipxe/ipxe/commit/059c4dc688cea6d9398d63ca10f5c00e94ae2159 fixed it for me as well from master (currently https://github.com/ipxe/ipxe/commit/8b14652e506d99499cfbeaed0df07d6a83ec029e). Once I finish my redundant server setup, I'd be happy to do some testing and maybe contribute. I'm not *great* with C, but I can kind of kludge something together and take comments in a review to clean it up. It seems like that commit is not that complicated, so I think I can handle it if I understand what exactly is happening.

This commit doesn't seem too complicated, but I'm missing some of the context. If someone who understands it could point me in the right direction I'd love to give it a go. Otherwise, I might circle back to it later.

NiKiZe, 2023-10-01:
Could you grab ifstat from iPXE , both with and without that commit.

Unless there was changes to which devices are "supported" by that commit we wouldn't expect any changes.

sploders101, 2023-10-16:
Sorry for the late response. I haven't had a good opportunity to take my server down for troubleshooting lately without making someone upset. Here are the results of running `ifstat` on the latest build before and after DHCP. The port it's connected to is a trunk port with multiple VLANs, but I don't see why traffic for an undeclared VLAN would trigger an *error*.

[image]
[image]

NiKiZe, 2023-10-16:
Reported errors from ifstat is common, and often not an issue.

The interesting part is comparing ifstat output between the working and non working iPXE builds.
This is the one with issues? What about ifstat from the working build?

NiKiZe, 2023-12-05:
Did you have a chance to compare ifstat between working and non working builds?

ErwanAliasr1, 2024-01-18:
I had booting issue with a BCM57414 (14e4:16d7) where depending on the server I'm loading a large ramfs (+500MB) from, ipxe remains stuck on a given (random) percentage of the file.

It was reproducible so started on top on the upstream version (26d3ef062b33e43e076b7ecef20c4ec3f9441860), I enabled debug.
The trace always ends like :

- RX Stat Total 1955 Good 1955 Drop err 0 LB 0 VLAN 0
- CQ Type (rx) cid 27
  RX desc_idx 3 PktLen 60
- RX Stat Total 1956 Good 1956 Drop err 0 LB 0 VLAN 0
- CQ Type (rx) cid 29
- CQ Type (rx) cid 29
- CQ Type (rx) cid 29
- CQ Type (rx) cid 29
- CQ Type (rx) cid 29
- CQ Type (rx) cid 29
- CQ Type (rx) cid 29
- CQ Type (rx) cid 29
[...]

If appear that bnxt_rx_complete() https://github.com/ipxe/ipxe/blob/master/src/drivers/net/bnxt/bnxt.c#L484, returns NO_MORE_CQ_BD_TO_SERVICE in loop.

This makes the driver looping on the same packet and not reading new packets and blocks the boot process. 

It's unclear to me if it's a firmware or driver issue.

I tried a workaround (attached to this comment) and by reducing the number of RX buffers, all my servers were fine at booting large file. This patch can affect the download speed when operating in perfect conditions but solved my issue here.

If some want to test it and make a feedback I'd be happy about it, I can also offer a PR but I'd love having more comments on it first.

[0004-bnxt.txt](https://github.com/ipxe/ipxe/files/13977746/0004-bnxt.txt)

For the reference, please find my card info 
  FW Version         : 226.0.145.0
  cmd timeout        : 5000
  hwrm_max_req_len   : 128
  hwrm_max_ext_req   : 384
  chip_num           : 16d7
  chip_id            : 1010000
  Port Number        : 0
  fid                : 0x0001
  PF MAC             : 14:23:f2:c3:e4:20
  min_hw_ring_grps   : 169
  max_hw_ring_grps   : 169
  min_tx_rings       : 170
  max_tx_rings       : 170
  min_rx_rings       : 241
  max_rx_rings       : 241
  min_cq_rings       : 295
  max_cq_rings       : 295
  min_stat_ctxs      : 235
  max_stat_ctxs      : 235
  ordinal_value      : 0
  stat_ctx_id        : 0
  num_cmpl_rings     : 1
  num_tx_rings       : 1
  num_rx_rings       : 1
  num_ring_grps      : 1
  num_stat_ctxs      : 1 

I also emailed the original driver author to inform him about the issue and my workaround, we'll see how it goes.

jw14812, 2024-02-12:
Hi, I tried reproducing this issue by downloading a 2.5GB test file using tftp and http, but I was not able to observe the hang.  Is the boot.wim or large ramfs available to download so that I can test that on my setup directly

ErwanAliasr1, 2024-02-13:
I cannot share the RAMFS I'm using in production but it's a 600MB one. Please note that's on a real production network infrastructure implicating several switches and routing between the server and the client.

mcb30, 2024-02-13:
@jw14812 thanks for testing!

@ErwanAliasr1 can you try simplifying the setup by e.g. trying a different (and public) large image to download, or by using a direct connection that eliminates the variety of switches and routers from the scenario?

agrevtsev, 2024-02-25:
Hey guys! Thanks for clues!
Were trying to boot Supermicro server with H12SSL-NT m/b, and such NICs
```
45:00.0 Ethernet controller: Broadcom Inc. and subsidiaries BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller (rev 01)
45:00.1 Ethernet controller: Broadcom Inc. and subsidiaries BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller (rev 01)
```
When i used `ipxe.efi`, booting over HTTP were stuck while downloading initrd (around 30Mb), with error [Error: No buffer space available](https://ipxe.org/err/2a2c40).
As soon as i rebuild `ipxe.efi` with @ErwanAliasr1 (thanks mate!) fix - server could download initrd and boot (download were veeeery slow - but successful)

Sorry for lack of debug logs - unfortunately have only IPMI access, can't copy-paste.

Br, Alexey

ErwanAliasr1, 2024-02-26:
So it's not an issue with my infra but a common issue between many. My
patch only workaround the real issue I have no clue but should be useful
for the author to dig the code/fw.
Glad it helped one at least

mczka, 2024-06-25:
```
10.10.104.11 - - [24/Jun/2024:15:00:09 +0200] "GET /POZCEN-FWGI01/pozcen-fwgi01-comp01-kvm/boot/vmlinuz-lts HTTP/1.1" 200 10599488 "-" "iPXE/1.0.0+"
10.10.104.11 - - [24/Jun/2024:15:00:10 +0200] "GET /POZCEN-FWGI01/pozcen-fwgi01-comp01-kvm/boot/initramfs-lts HTTP/1.1" 200 26564111 "-" "iPXE/1.0.0+"
```

ipxe remains stuck on a given (random) percentage of the file.  I'm using Dell PowerEdge 7525 + NICs:

```
pozcen-fwgi01-comp01-kvm:~$ lspci -nn | grep -i eth
41:00.0 Ethernet controller [0200]: Mellanox Technologies MT2892 Family [ConnectX-6 Dx] [15b3:101d]
41:00.1 Ethernet controller [0200]: Mellanox Technologies MT2892 Family [ConnectX-6 Dx] [15b3:101d]
41:00.2 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
41:00.3 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
41:00.4 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
41:00.5 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
41:00.6 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
41:00.7 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
63:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller [14e4:16d8] (rev 01)
63:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller [14e4:16d8] (rev 01)
a1:00.0 Ethernet controller [0200]: Mellanox Technologies MT2892 Family [ConnectX-6 Dx] [15b3:101d]
a1:00.1 Ethernet controller [0200]: Mellanox Technologies MT2892 Family [ConnectX-6 Dx] [15b3:101d]
a1:00.2 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
a1:00.3 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
a1:00.4 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
a1:00.5 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
a1:00.6 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
a1:00.7 Ethernet controller [0200]: Mellanox Technologies ConnectX Family mlx5Gen Virtual Function [15b3:101e]
e1:00.0 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]
e1:00.1 Ethernet controller [0200]: Broadcom Inc. and subsidiaries NetXtreme BCM5720 Gigabit Ethernet PCIe [14e4:165f]

iPXE NIC is eth0:
pozcen-fwgi01-comp01-kvm:~$ ethtool -i eth0
driver: bnxt_en
version: 6.1.75-0-lts
firmware-version: 223.0.205.0/pkg 22.31.13.70
expansion-rom-version:
bus-info: 0000:63:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no

```

How to reproduce:  download zip from github.com/ipxe, unzip ; cd ipxe/src; NO_WERROR=1 make bin-x86_64-efi/ipxe.efi EMBED=netboot.comp01.ipxe;  I use alpinelinux netboot 3.18.6;

WORKAROUND: restart the server until the kernel and initframfs is loads correctly by iPXE binary; UEFI mode.

msteggink, 2024-07-18:
This also fixed it for me (HPE DL380+Broadcom BCM57416 NetXtreme-E 10GBase-T ):
```
-#define NUM_RX_BUFFERS                          8
+#define NUM_RX_BUFFERS                          2
```

ErwanAliasr1, 2024-07-18:
considering how beneficial this patch I suggest to get in as it helps a lot.

warfery, 2024-07-19:
We are also facing the described problem (Intel NIC). We think that the issue can be related with the change in this commit: https://github.com/ipxe/ipxe/commit/2d180ce233f6eae8d0df279e5cbf134f547f604c
It is possible that problematic NICs can't process data fast enough from a buffer and the buffer overflows. With previous value it was not a problem since there was 256kB window frame (not tested yet). We discovered that that communication crashed when the difference between data sent by server and data ack by client was about 400kB.

mldailey, 2024-08-14:
Seeing this same issue on Broadcom BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller. Decreasing NUM_RX_BUFFERS to 2 in src/drivers/net/bnxt/bnxt.h as above fixed the issue for us in ipxe head yesterday (3b4d0cb).

Oddly, we have four identical systems with identical hardware, and were having this issue only on one of the systems, but it was consistently failing on this one.

ErwanAliasr1, 2024-08-19:
https://github.com/tinkerbell/ipxedust/pull/171 solved this issue, we can probably close this one too

NiKiZe, 2024-08-19:
> @elvinup, [tinkerbell/ipxedust#171](https://github.com/tinkerbell/ipxedust/pull/171) solved this issue, we can probably close this one too

The short answer is probably no https://github.com/ipxe/ipxe/issues/1170#issuecomment-2125466371

This issue is similar to #1170 in behaviour

> We are also facing the described problem (Intel NIC). We think that the issue can be related with the change in this commit: [2d180ce](https://github.com/ipxe/ipxe/commit/2d180ce233f6eae8d0df279e5cbf134f547f604c) It is possible that problematic NICs can't process data fast enough from a buffer and the buffer overflows. With previous value it was not a problem since there was 256kB window frame (not tested yet). We discovered that that communication crashed when the difference between data sent by server and data ack by client was about 400kB.

Intel NIC and commit 2d180ce is probably more relevant to #1170

achristianson, 2024-08-26:
We have some `ROMED8-T2/BCM` nodes in our cluster and are experiencing the same issue: stalled large initramfs download at random percentages.

This is using the integrated 10G `BCM57416` NIC. Oddly, we only noticed the issue once we plugged it into a 10G port, and like @mldailey  mentioned, it is only on one specific server even though some other servers are running identical hardware.

We applied the @mczka patch to the current master branch commit as of today (`bdb5b4aef46ed34b47094652f3eefc7d0463d166`) and all our problems went away instantly.

HotelMoted, 2024-09-19:
Changing `NUM_RX_BUFFERS` from 8 to 2 also fixed my issue of HTTP stalling on my BCM57412 NICs. I had to apply this fix manually. Could this change be merged into the main branch?

mcb30, 2024-09-20:
> Changing `NUM_RX_BUFFERS` from 8 to 2 also fixed my issue of HTTP stalling on my BCM57412 NICs. I had to apply this fix manually. Could this change be merged into the main branch?

No, not unless someone has done the debugging work to figure out **why** this change fixes the problem.  Otherwise, we're just storing up unexplained problems for the future.

ErwanAliasr1, 2024-09-20:
I agree on the principles but the original code writer (Broadcom) do no
answer, code is impossible to read, no specs and driver is reported buggy
by many now.
I did as far as I could to report traces to narrow the issue.
As maintainer is ghosting us, i'd prefer a working code, even if not
satisfying, instead of a broken code that cause us issues.
So yes, boot is slower but works.
I'd advocate to get the pr being accepted and reverted once Broadcom will
help us understanding/maintaining this code.

NiKiZe, 2024-09-20:
Could those with issues again verify with `ifstat` that they are using the broadcom driver (14e4-xxxx), and not `NII`

If that is the case you could consider using `snp.efi` or `snponly.efi` instead.

Since the broadcom driver is unmaintained, and as mentioned (my paraphrasing) a mess, the better solution is probably to pull that driver from the codebase, or at least pull it from default ipxe builds, and have it needed to be enabled explicitly just as we do for USB based NICs?

NiKiZe, 2024-09-20:
@jw14812 any chance you could take a look at this again

`16d7`, `16d8` seems to be the ones mentioned most, with 10gb links, http downloads, any largeish files.

Delay might play a part, or maybe firmware used?

https://github.com/ipxe/ipxe/issues/1023#issuecomment-1898585257 might be a start?

agrevtsev, 2024-09-20:
> Seeing this same issue on Broadcom BCM57416 NetXtreme-E Dual-Media 10G RDMA Ethernet Controller. Decreasing NUM_RX_BUFFERS to 2 in src/drivers/net/bnxt/bnxt.h as above fixed the issue for us in ipxe [...]

Hey @mldailey could you check firmware versions for those BCMs across your systems? It could help to understand if it's firmware problem.
Cheers!

mldailey, 2024-09-20:
@agrevtsev The four systems I had trouble with last month all have identical hardware and firmware. ethtool shows this on all the NICs:
`firmware-version: 228.0.116.0/pkg 228.1.111.0`
I don't know if there's a better way to get the version, but hopefully this is enough to go on.

I should also mention that all the testing I did was at 10Gbps. Unfortunately, all these systems are now in production, so I can't do any further testing at this time.

HotelMoted, 2024-09-20:
I went from v216.0.333.11 to v230.1.116.0 on my BCM57412 to try and see if that would fix the issue, it wasn't until I changed the NUM_RX_BUFFERS did HTTP stop stalling out.

jw14812, 2024-09-20:
My apologies for not chiming in earlier.  I will try to look into this issue and the possible solution proposed

mcb30, 2024-09-22:
> My apologies for not chiming in earlier.  I will try to look into this issue and the possible solution proposed

@jw14812 Thank you - much appreciated! Please let me know if there's any help I can provide.

daberkow, 2024-09-24:
I have Broadcom `BCM57414` running firmware `229.2.52.0/pkg 22.92.06.10` which is the latest for a Dell R660. I am also running 10gb/s on a 10/25gb nic. It auto went to the `14e4-16d7` driver. I was seeing the same issue where it would start booting then work about 60% of the time, sometimes stop loading part way through the transfer over http. Sometimes it would seem to happen in a row, then sometimes be fine for a bit.

I did the `NUM_RX_BUFFERS` fix, and then its a bit slower to load, but seems to always work. Hope this helps, if you have any questions let me know!

ErwanAliasr1, 2024-09-25:
Thanks for the reporting. The exact same story. I hope we'll get a clear
fix soon as a large set of servers are affected the same.

jw14812, 2024-10-24:
@ErwanAliasr1 - I am still having problems reproducing this error.  If possbile, can you share some general details about your setup?

ErwanAliasr1, 2024-10-24:
I'm using a 25g link and download 600MB initramfs from an http server.

jw14812, 2024-10-24:
Is the connection going through multiple switches?  Or the client/server reside on the same switch with minimal other traffic?

ErwanAliasr1, 2024-10-25:
> Is the connection going through multiple switches? Or the client/server reside on the same switch with minimal other traffic?
No there are at least 4/5 hops between the two.

daberkow, 2024-10-25:
If it helps, same... I think I have 3 switches. But all have anywhere from 10gb/s to 40gb/s interconnecting them and no where near saturated.

ErwanAliasr1, 2024-10-25:
I agree with Dan, our networking stack is pretty robust, not saturated, and
proved to work with all other brands & servers brands.
This is the unique case I'm aware at our scale (40K servers, 11
datacenters) where we have servers refusing to boot like this.
At least I have isolated the part of the code were we are stuck, this is
maybe enough to give ideas of what gone wrong, I don't know.

jw14812, 2024-11-21:
@ErwanAliasr1 - would it be possible for you to collect a tcpdump  if you observe the failure again.  Thanks

ErwanAliasr1, 2024-11-21:
I'm sorry but I can't tcpdump in our infra...

ErwanAliasr1, 2024-11-21:
That's why I ended up trying to understand the issue from the host side and
instrument the ipxe code to end in that portion of the code.

ErwanAliasr1, 2026-01-07:
@mcb30 This issue has been open for 2.5 years and still affects users.

My workaround was reported to be functional by many and is used in numerous downstream builds to make our infrastructure work.

Until @jw14812 provides a fix for this, he's the only one to be able to do it, as there is no documentation to understand what the code is doing/expecting from the firmware. Can you please merge this patch so all users can download large files on Broadcom nics (which are very common)?

HotelMoted, 2026-01-07:
I permanently fixed this in my environment by going to src/drivers/net/bnxt/bnxt.c and commenting out:
 	PCI_ROM( 0x14e4, 0x16d6, "14e4-16D6", "14e4-16D6", 0 ),
	PCI_ROM( 0x14e4, 0x16d7, "14e4-16D7", "14e4-16D7", 0 ),

It forces the NIC to load through SNP mode instead of the native driver. I have literally never thought of this problem ever again until I just got pinged this morning from this thread in my email. The RX Buffers fix did help a bit, but commenting out those lines fixed it permanently in my environment.

jw14812, 2026-01-07:
Are you able to reproduce this with the tip of the tree?

JosephGarrone, 2026-03-19:
@jw14812 that is correct. Was encountering issues with the 1.21.1 build that the FOGProject uses (with some minor customisations that they apply). I then tried a local build using the FOGProject's modifications, but against the latest master commit in this repository, and had the same issues. I then repeated that again but with the `sed` modification I listed above injected into FOGProject build script and it worked without issue. I have attached my build script that I used to generate a working file (separate from the FOGProject's build script), this should run on a blank Ubuntu 24.04 server install.

[build-ipxe.sh](https://github.com/user-attachments/files/26106395/build-ipxe.sh)

jsuchenia, 2026-09-09:
@mcb30 — following up on your [request to @JosephGarrone](https://github.com/ipxe/ipxe/issues/1023#issuecomment-4090120266)
above to test with a completely unmodified iPXE from current `master`: we hit this same failure independently on a **CompuLab Fitlet-3** (same hardware family reported above), and built an isolated point-to-point lab specifically to get clean, unmodified-build evidence, since our first attempts were confounded by our own debug tooling (not reported here, happy to share separately if useful, but it's not an iPXE-upstream issue).

Onboard NIC is `8086:4ba1`/`4bb1`, bound to the **NII** driver (no native iPXE PCI driver exists for this chip). Setup: a spare NIC port on the Fitlet-3 wired directly to a second identical unit, no switch, no other traffic on the segment.

Also relevant to the more detailed mechanism discussion in [#1170](https://github.com/ipxe/ipxe/issues/1170) — cross-posting a pointer there.

## Setup

- iPXE `master` (~`ff6e520`), `HEAP_SIZE` already at 4MB (`a8c89276c`),  default `TCP_MAX_WINDOW_SIZE` (2MB) unless noted otherwise.
- `DEBUG=http,tcp,malloc,iobuf,settings` build; console output captured  via screen recording (no serial console available on this hardware) and  packet capture on the server's NIC.
- Capture interface had `tso`/`gso`/`gro` disabled before capturing, per  the guidance earlier in this thread about offload corrupting captures.
- `set syslog` was **not** active during any of the runs described below  (we found, separately, that enabling it interacts badly with our own  debug build in a way that produces a different, self-inflicted failure - not reporting that part here since it's not an iPXE-upstream issue,  happy to share details if useful).

## Findings

**1. The known `TCP_MAX_WINDOW_SIZE=256KB` workaround also fixes this hardware.** Downloads that reliably stalled with the default 2MB window completed successfully every time with a 256KB window.

**2. The already-merged `HEAP_SIZE=4MB` fix (`a8c89276c`) does *not* fix it on this hardware** with 4MB heap and the default 2MB window, the stall still reproduces. Might be a useful data point since 4MB should comfortably hold a 2MB window.

**3. The stall point is deterministic, not random.** Two independent connections (different boot cycles, different TCP ports/ISNs) fetching the same file both stalled at the **exact same byte offset** (1,068,831 bytes). This doesn't look like ordinary random packet loss.

**4. Real link-level loss is negligible.** `ip -s link` on the server showed only 8 RX drops total across several hours and a dozen+ repro
attempts on this otherwise-idle point-to-point segment. A single lost segment is sufficient to trigger a permanent stall here. This isn't a lossy-link scenario.

**5. SACK is negotiated but never used, even once, in a stalled connection**. Despite the server having sent ~560KB more data past the point the client stopped acknowledging (confirmed via a clean, offload-disabled capture). We can't tell from the capture alone whether that data reached the client and was silently reneged, or never arrived at all.

**6. This is a permanent deadlock, not degraded throughput.** After the first lost segment, the server retransmitted it three times with correct exponential backoff over 2+ minutes. **Every retransmission was ignored**, the client's cumulative ACK never advanced, not even by one byte, until iPXE's own connection timeout gave up. We'd expect reneging-under-memory-pressure to degrade throughput, not permanently wedge the connection against a correctly-retransmitted, in-window segment.

**7. We reviewed `tcp.c`'s RX/ACK/window/SACK/discard logic against this exact scenario** (a retransmission landing precisely at `rcv_ack`) and could not find a code path that would reject or silently drop it. `tcp_rx_enqueue()`'s discard conditions don't fire for it, `xfer_window()` defaults to unlimited and neither `httpcore.c` nor `downloader.c` overrides it (so the window-replenish path looks fine for a plain HTTP download), and `tcp_rx_ack()`/`tcp_rx_opts()` have no rejection path that applies. We may well have missed something, but at face value this doesn't look like a `tcp.c` logic bug.

**8. `ifstat` at the point of failure shows the driver's own RX error counter is almost completely silent**.  1 `RXE` across 1,681 RX
operations for the whole boot, despite the permanent stall. If this is consistent with what's been seen before: it suggests the loss is happening somewhere the NII/UNDI firmware itself doesn't report as an error at all (rather than a driver-detected-and-mishandled failure), which would match the concern already raised in this thread about this class of driver having "no visibility into RX buffer exhaustion."

If you need any more details or evidence, just let me know. We're interested in a signed, Secure-Boot-enabled fix for this, so we're happyto help gather further evidence or develop fixes, e.g. having a stalled/failed download resume via an HTTP `Range` request from the last successfully-delivered byte, since the HTTP layer already constructs `Range` headers for block-device reads in `httpblock.c`.

Full evidence (packet capture, console captures, `ifstat` output) is available here: **https://drive.google.com/drive/folders/1DXi-29j6ef2rKnyzyh5YoEfHqySiIELH**

mcb30, 2026-09-11:
@jsuchenia Thank you for testing.  There is unfortunately too much information in what you have written for me to do anything beyond a cursory check.  I note that you have `DEBUG=http,tcp,malloc,iobuf,settings` enabled.  This seems like quite an intrusive level of debug: I would expect this amount of debugging to change behaviour to the point of becoming no longer meaningful.

jsuchenia, 2026-09-13:
@mcb30 Yes, I used some LLM tooling to summarise it - TL;DR:

> After the first lost segment, the server retransmitted it three times with correct exponential backoff over 2+ minutes. Every retransmission was ignored,

So when we lost at least one fragment, the retransmission happen - and it's never handled correctly by the client. From a network (server) perspective (documented in PCAP files) - it looks like ignored.  In that setup it's almost impossible to download a large images over http(s) connections. I suspect that it's affecting all devices with Aptio BIOS for Intel Gigabit devices.

As we are interested in secure-boot iPXE, we had to throttle it on a server side (to around 1MB/s) - to avoid congestion - and it's some kind of solution.   Another tested solution is to get back to a small TCP_MAX_WINDOW_SIZE to 256KB - however it will force us to introduce custom keys, and that's not an option at the moment

Could you consider decrease TCP_MAX_WINDOW_SIZE to 256KB as was stated by some of the other ppl here in this thread as a solution on iPXE side? I know that it's not a bug in iPXE, however it could be mitigated in this way.