ipxe/ipxe #1027: Loading grub from ipxe on ubuntu

edgood1, 2023-08-18:
I can load grub fine from ipxe via `kernel file://EFI/EFI/grub/grubx64.efi` , however the ubuntu installation of grub expects that grub can read a `grub.cfg` file, which is essentially a script of grub commands from the same location on the EFI partition (i.e. `//EFI/EFI/grub/grub.cfg`).

Booting the grub bootstub directly works, grub reads the grub.cfg file without issue and launches the commands, but when loading grub in ipxe it just drops me to a grub shell, presumably because it can't read the `grub.cfg` file.

It tried loading that `grub.cfg` via `imgload` but that didn't help. I'm hoping someone has ran into this before and has a fix.

mcb30, 2023-08-18:
iPXE provides any loaded EFI image with a virtual filesystem (via the standard `EFI_SIMPLE_FILE_SYSTEM_PROTOCOL` abstraction) containing all other files that were downloaded alongside the image.

Unfortunately GRUB has no support for reading from a standard EFI filesystem abstraction, and will instead try to find a block device containing a supported filesystem.  GRUB is therefore unable to read any of the files within the virtual filesystem provided by iPXE.

The solution in this case is to use iPXE to boot Ubuntu directly, and to avoid using GRUB altogether.

closed 2023-08-18 by mcb30

NiKiZe, 2023-08-18:
You could hack around this by loading all needed files in iPXE and then chain into wimboot, which does provide a virtual block filesystem. But the bug is in grub.

edgood1, 2023-08-21:
My workaround was to create a custom grub image with an embedded bootstrap script that includes the same commands that the `grub.cfg` in the EFI partition would have.  Having it embedded into the image means that iPXE doesn't need to read it from the EFI partition. This is accomplished with something like:
`grub-mkimage -O x86_64-efi -o /boot/efi/EFI/grub/grubx64.efi -p "/boot/grub" -c /tmp/grub_commands $DESIRED_GRUB_MODULES_LIST`