ipxe/ipxe #1042: Issue with persistent source TCP Port on UEFI iPXE

gauchim, 2023-09-21:
Hello,

We've identified an issue with your software on an Intel NUC PC (model nuc11atk) that only supports UEFI boot. We use iPXE for booting and rely on HTTP for file downloads during the boot process. The problem arises when we need to restart the PC during this process. It appears that our firewall/router drops the new attempts to establish a connection for the HTTP protocol.

Upon inspecting the network traffic with Wireshark, we've noticed that the socket created to communicate with the web server consistently uses the same source port. Since the source and destination IPs are allways the same, and the destination port is fixed at 80, then we believe that the source port should ideally change with each PC restart, allowing our firewall to recognize it as a new communication attempt and not drop it.

Could you please help us understand why the software is not selecting a random source port when establishing a TCP connection? We appreciate your assistance.

Thank you.

NiKiZe, 2023-09-21:
Your firewall have a bug, a new SYN packet should always be accepted regardless of parameters.

But I agree that source port should, if possible, be "random"
Connections are reused, and the firewall should, without issue, handle the existing behaviour.

gauchim, 2023-09-21:
Excellent explanation @mcb30!!! 

From my complete ignorance of the source code, it seems that the option of using the **rdtsc** function is easily implementable. Is that so? Are they few lines of code?

Thank you so much!!

mcb30, 2023-09-21:
> From my complete ignorance of the source code, it seems that the option of using the **rdtsc** function is easily implementable. Is that so? Are they few lines of code?

The cleanest abstraction would probably be to use the existing `profile_timestamp()` to read the TSC counter, since this will also work to read the equivalent high-resolution timer on non-x86 platforms.  To maintain compatibility with pre-Pentium CPUs, the profiling code would have to be updated to check for TSC presence (and do so while adding zero overhead to the `profile_timestamp()` call itself).

gauchim, 2023-09-21:
The tsc function may already be used on i386 architecture, but not on x86_64 architecture?

src/arch/i386/include/bits/profile.h
```
profile_timestamp ( void ) {
	uint64_t tsc;

	/* Read timestamp counter */
	__asm__ __volatile__ ( "rdtsc" : "=A" ( tsc ) );
	return tsc;
}
```

src/arch/x86_64/include/bits/profile.h
```
profile_timestamp ( void ) {
	uint32_t eax;
	uint32_t edx;

	/* Read timestamp counter */
	__asm__ __volatile__ ( "rdtsc" : "=a" ( eax ), "=d" ( edx ) );
	return ( ( ( ( uint64_t ) edx ) << 32 ) | eax );
}
```

gauchim, 2023-09-21:
Very sorry. I'll leave writing code to people who know! :_)

I propose to implement this improvement in one of the next releases.

Thank you!

gauchim, 2023-09-22:
Hi again,

Today, we conducted further testing using different hardware (specifically an Intel NUC model nuc5cpyh) which supports both BIOS and UEFI boot options. Through network captures with Wireshark, we observed a behavior similar to what we encountered yesterday.

When booting via UEFI, the system continues to reuse the same supposed random source port with each restart. However, with BIOS boot, the source port does change on every restart.

Do you think it's possible for iPXE in UEFI mode to exhibit the same behavior as iPXE in BIOS mode? Your insights on this matter would be greatly appreciated.

Thank you.

NiKiZe, 2023-10-03:
> We ran into this issue as well. As a Noob I've come up with the following patch. It compiles fine. `undionly.kpxe` grows only 5 bytes and the size of `ipxe.efi` doesn't change.
> 
> I haven't got a test-lab set up to try (yet). Would this work?
> 
> ```diff
> diff --git a/src/core/random.c b/src/core/random.c
> index 975a03cf..165731a2 100644
> --- a/src/core/random.c
> +++ b/src/core/random.c
> @@ -7,7 +7,7 @@
>  FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
>  
>  #include <stdlib.h>
> -#include <ipxe/timer.h>
> +#include <ipxe/time.h>
>  
>  static int32_t rnd_seed = 0;
>  
> @@ -31,7 +31,7 @@ long int random ( void ) {
>         int32_t q;
>  
>         if ( ! rnd_seed ) /* Initialize linear congruential generator */
> -               srandom ( currticks() );
> +               srandom ( (unsigned int)time_now() );
>  
>         /* simplified version of the LCG given in Bruce Schneier's
>            "Applied Cryptography" */
>            ```
> ```

How will this behave on systems without a rtc, such as Raspberry Pi?

mcb30, 2023-10-03:
> We ran into this issue as well. As a Noob I've come up with the following patch.  It compiles fine. `undionly.kpxe` grows only 5 bytes and the size of `ipxe.efi` doesn't change.
> 
> I haven't got a test-lab set up to try (yet). Would this work?

You must have some feature enabled that already requires the real-time clock (eg HTTPS), which is why you don't see a substantial code size change with your configuration. In the default configuration, the impact will be much higher.

hkoof, 2023-10-03:
It was the unmodified configuration of [ today's master branch](https://github.com/ipxe/ipxe/commit/8b14652e506d99499cfbeaed0df07d6a83ec029e).

hkoof, 2023-10-03:
> How will this behave on systems without a rtc, such as Raspberry Pi?

My assumption would be `time_now()` always returns `0`, which would in essence hardly be different from always the same number of ticks.

gauchim, 2023-10-04:
Based on the change proposed by @hkoof, I have tried to implement the second option proposed by @mcb30 (use of the rdtsc instruction). I think we still need to check that we have support for TSC. These are the changes:

```
diff --git a/src/core/random.c b/src/core/random.c
index 975a03cf..09231f16 100644
--- a/src/core/random.c
+++ b/src/core/random.c
@@ -7,7 +7,6 @@
 FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );

 #include <stdlib.h>
-#include <ipxe/timer.h>
 
 static int32_t rnd_seed = 0;
 
@@ -29,9 +28,11 @@ void srandom ( unsigned int seed ) {
  */
 long int random ( void ) {
        int32_t q;
+       uint64_t tsc_value;
 
+       asm volatile("rdtsc" : "=A" (tsc_value));
        if ( ! rnd_seed ) /* Initialize linear congruential generator */
-               srandom ( currticks() );
+               srandom ( tsc_value );
 
        /* simplified version of the LCG given in Bruce Schneier's
           "Applied Cryptography" */
``` 

After making network captures with Wireshark, I have verified that both the changes proposed by hkoof and those proposed by me change the source port randomly at each PC reboot!

Is it possible to propose a change commit in the main branch of the project?

hkoof, 2023-10-05:
TSC is only random by coincidence and not available on all hardware. Since the issue has only surfaced on EFI and EFI systems have an RTC, how about seeding using RTC only on EFI for the source port?

```diff
diff --git a/src/interface/efi/efi_pxe.c b/src/interface/efi/efi_pxe.c
index 843ebb5e..2dce0f2a 100644
--- a/src/interface/efi/efi_pxe.c
+++ b/src/interface/efi/efi_pxe.c
@@ -38,6 +38,7 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 #include <ipxe/open.h>
 #include <ipxe/dhcppkt.h>
 #include <ipxe/udp.h>
+#include <ipxe/time.h>
 #include <ipxe/efi/efi.h>
 #include <ipxe/efi/efi_snp.h>
 #include <ipxe/efi/efi_pxe.h>
@@ -1067,6 +1068,7 @@ efi_pxe_udp_write ( EFI_PXE_BASE_CODE_PROTOCOL *base, UINT16 flags,
                /* The API does not allow for a sensible concept of
                 * binding to a local port, so just use a random value.
                 */
+               srandom( (unsigned int)time_now() );
                src.st.st_port = ( random() | htons ( 1024 ) );
                if ( src_port )
                        *src_port = ntohs ( src.st.st_port );
```

mcb30, 2023-10-06:
> It was the unmodified configuration of [ today's master branch](https://github.com/ipxe/ipxe/commit/8b14652e506d99499cfbeaed0df07d6a83ec029e).

Interesting.  It turns out that commit https://github.com/ipxe/ipxe/commit/41f786cc0 causes `rtc_time.o` to be dragged in to the default build anyway in order to provide the `${unixtime}` setting.  With this being the case, there is no reason not to use `time()` to seed the default entropy pool.

PR #1052 (2023-10-06): [libc] Use wall clock time as seed for the (non-cryptographic) RNG
We currently use the number of timer ticks since power-on as a seed for the non-cryptographic RNG implemented by random().  Since iPXE is often executed directly after power-on, and since the timer tick resolution is generally low, this can often result in identical seed values being used on each cold boot attempt.

As of commit 41f786c ("[settings] Add "unixtime" builtin setting to expose the current time"), the current wall-clock time is always available within the default build of iPXE.  Use this time instead, to introduce v [... cut]
closed 2023-10-06 by mcb30