ipxe/ipxe #1048: iPXE fails to run DHCP with 56cc61a

RenaudWasTaken, 2023-10-04:
Hello there!

We recently spent a few days debugging boot issues with our nodes and managed to zero out on the [latest commit](https://github.com/ipxe/ipxe/commit/8b14652e506d99499cfbeaed0df07d6a83ec029e) as the source of our issues.

When we boot in iPXE we initialize our machine with iPXE we expect it to go through the DHCP boot process:
1. Discover
2. Offer
3. Request
4. Ack

When we ran tcpdump we confirmed that the discover and offer did arrive on the DHCP server but we never saw the Request packet coming in.

I ended up compiling the latest version of iPXE and printing all the debug logs in net/udp/dhcp and observed that the Request packet seemed to be sent from there. I never managed to get a tcpdump from our switches but I suspect that the packet never gets sent.

What's odd is that I went into the PXE shell and I repeatedly ran dhcp and I could see the Discover & Offer packets sent and received every single time. But the Ack never went through.

I bisected a little bit more and observed that the DHCP flow works with the (HEAD - 1 commit)[https://github.com/ipxe/ipxe/commit/56cc61a168820c7cbbe23418388129ec11699a8c]

I'm absolutely not familiar with EAP and EAPoL so I'd love to have more details.

aospan, 2023-10-08:
Hello,

@RenaudWasTaken thanks for reporting this issue!
We observing the exact same issue with [this](https://github.com/ipxe/ipxe/commit/8b14652e506d99499cfbeaed0df07d6a83ec029e) commit on our bare metal platforms. 

Please let me know if any additional information needed to root cause this issue.
Thank you!

NiKiZe, 2023-10-09:
Could you build iPXE with DEBUG=eapol:3 and see if it explains anything about what you are seeing?

aospan, 2023-10-09:
@NiKiZe 

Please find console log below for ipxe built with `DEBUG=infiniband:3,eapol:3,dhcp:3`

```
IBDEV  creating queue pair
IBDEV  created queue pair 0xb121cd68 (0xb1219f08) with QPN 0x0
IBDEV  QPN 0x0 has 64 send entries at [0xb121cea8,0xb121d0a8)
IBDEV  QPN 0x0 has 64 receive entries at [0xb121d0a8,0xb121d2a8)
IBDEV  creating completion queue
IBDEV  created 128-entry completion queue 0xb121bc88 (0xb1219f28) with CQN 0x64a
net1: 7c:d3:0a:da:02:61 using ConnectX-4Lx on 0000:21:00.0 (Ethernet) [open]
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
DHCP 0xb1222028 entering discovery state
Configuring (net1 7c:d3:0a:da:02:61)...DHCP 0xb1222028 DHCPDISCOVER
EAPOL net1 transmitting Start
DHCP 0xb1222028 DHCPOFFER from 192.168.1.1:67 for 192.168.1.129
.DHCP 0xb1222028 DHCPDISCOVER
.EAPOL net1 transmitting Start
.DHCP 0xb1222028 entering request state
DHCP 0xb1222028 DHCPREQUEST to 192.168.1.1:67 for 192.168.1.129
DHCP 0xb1222028 DHCPREQUEST to 192.168.1.1:67 for 192.168.1.129
DHCP 0xb1222028 DHCPREQUEST to 192.168.1.1:67 for 192.168.1.129
.EAPOL net1 transmitting Start
DHCP 0xb1222028 DHCPREQUEST to 192.168.1.1:67 for 192.168.1.129
..EAPOL net1 transmitting Start
DHCP 0xb1222028 DHCPREQUEST to 192.168.1.1:67 for 192.168.1.129
..EAPOL net1 transmitting Start
..EAPOL net1 transmitting Start
DHCP 0xb1222028 DHCPREQUEST to 192.168.1.1:67 for 192.168.1.129
..EAPOL net1 transmitting Start
..EAPOL net1 transmitting Start
..EAPOL net1 transmitting Start
..EAPOL net1 transmitting Start
 ok
net0: fe80::7ed3:aff:feda:260/64 (inaccessible)
net1: fe80::7ed3:aff:feda:261/64
net2: fe80::7ed3:aff:feda:25e/64 (inaccessible)
net3: fe80::7ed3:aff:feda:25f/64 (inaccessible)
Nothing to boot: No such file or directory (https://ipxe.org/2d03e18e)
IBDEV  destroying QPN 0xc4b
IBDEV  destroying completion queue 0x64a
```

Looks like DHCP discovery was successful, but a DHCP request never went through (and no DHCP request packets were seen on the server side).

jhujasonw, 2023-10-11:
I am experiencing something similar, and had an older build  `g0aa2e`   That older version works on my ConnectX 6 ethernet card.  Booting with a newer version, `gff0f8` fails in the same way or similar to @aospan's output.

jhujasonw, 2023-10-11:
One other interesting test, comment out line 42 in `config/general.h` that reads

```//#define       NET_PROTO_EAPOL         /* EAP over LAN protocol */```

Which effectively disables the new EAPOL code, and it seems to work for me.  Looking at the code, I don't 100% see what is happening here except that if when it's sending the start packets, it's locking a buffer or something and new incoming stuff isn't allowed in or out.... that's just pure speculation though, I haven't studied the code to see what it's doing.

noslin005, 2023-10-13:
I'm experiencing the same issue with the latest iPXE v1.21.1+ (gff0f8) and Mellanox ConnectX-4, even after trying the workaround of commenting the NET_PROTO_EAPOL. The previous version, iPXE v1.21.1 (g988d), seems to function correctly. 

[image]
[image]

stappersg, 2023-10-20:
On _IRC_ in the _iPXE channel_ was this
```text
16:57 < redacted> got it working. should have looked through the github issues days ago. 
              https://github.com/ipxe/ipxe/issues/1048
```

jhujasonw, 2023-10-20:
> On _IRC_ in the _iPXE channel_ was this
> ```text
> 16:57 < redacted> got it working. should have looked through the github issues days ago. 
>               https://github.com/ipxe/ipxe/issues/1048
> ```

Is this acknowledgement of the bug?  I don't see any commits to main that are new to fix this.

kent-hydra, 2023-11-03:
> > On _IRC_ in the _iPXE channel_ was this
> > ```
> > 16:57 < redacted> got it working. should have looked through the github issues days ago. 
> >               https://github.com/ipxe/ipxe/issues/1048
> > ```
> 
> Is this acknowledgement of the bug? I don't see any commits to main that are new to fix this.

This was my comment in IRC. Yes this was an acknowledgement of the bug. I am fairly new to iPXE in general and dont quite know how to resolve the issue. My work around was using `56cc61a168820c7cbbe23418388129ec11699a8c` to build from source. After that my machine had no issues loading iPXE. The machine I was testing against had a Mellanox ConnectX-6D interface attached. With tcpdump I could see the DHCP DISCOVER requests were making it to my DHCP server but nothing else after that.

mcb30, 2023-11-07:
Should now be fixed: please let me know if not.

closed 2023-11-07 by mcb30

bdelano, 2023-12-06:
I have been struggling wit this issue and I am still seeing the same behaviour with the latest version [f22879c](https://github.com/ipxe/ipxe/commit/f22879ca994aedd2667bb0c73ebce505bf5f8cef) 
if I downgrade to https://github.com/ipxe/ipxe/commit/8b1d34badf321668f830c78e6803d718446b57ef it begins working
[image]

stappersg, 2024-12-30:
@bdelano 

Regarding email Wed, Dec 06, 2023 at 01:42:34PM -0800:
> I have been struggling wit this issue and I am still seeing the same behaviour with the latest version [f22879c](https://github.com/ipxe/ipxe/commit/f22879ca994aedd2667bb0c73ebce505bf5f8cef)
> if I [...]

Raise that in a fresh issue.

Regarding
Geert Stappers
While updating a closed issue.
-- 
Silence is hard to parse

PR #1073 (2023-11-07): Fix EAPoL-Start behaviour to allow slow authentication exchanges to succeed
EAP exchanges may take a long time to reach a final status, especially when relying upon MAC Authentication Bypass (MAB).  Our current behaviour of sending EAPoL-Start every few seconds until a final status is obtained can prevent these exchanges from ever completing.

Fix by suppressing EAPoL-Start once EAP is in progress, and by limiting the total number of EAPoL-Start packets sent per authentication attempt.

Fixes: #1048