ipxe/ipxe #1060: Feature Request: Add parameter to mask or hide input from "read" command.

matbe, 2023-10-17:
Today when using the read command in a iPXE menu the input is show in clear text.
https://ipxe.org/cmd/read

I would like to request to add an optional parameter to the read command that masks the input (with stars '*') in the same way that the password is masked when using the login command.

Use case is that we use the read command to ask the end user to provide a simple pin code instead of a full username/password.
Since we are using the read command the pin is shown in cleartext on the screen.

For example:
read --maskinput pincode
Should mask the input with stars and put the result in the "pincode" variable.

Expected end result:
[image]

NiKiZe, 2023-10-17:
Some alternative ways that this could maybe be implemented in:

What if login cmd had options to only request password? It would also require to have title of fields changable?

Would read without any output be acceptable? (That is no stars) which would be same as most *nix prompts.

srett, 2023-10-18:
We're using something like that, here's our patch (probably not up to snuff when it comes to quality/style etc. regarding a PR?)

```diff
diff --git a/src/hci/commands/login_cmd.c b/src/hci/commands/login_cmd.c
index c9e19643..6ed76c6f 100644
--- a/src/hci/commands/login_cmd.c
+++ b/src/hci/commands/login_cmd.c
@@ -23,6 +23,7 @@
 
 #include <string.h>
 #include <stdio.h>
+#include <getopt.h>
 #include <ipxe/command.h>
 #include <ipxe/parseopt.h>
 #include <ipxe/login_ui.h>
@@ -36,10 +37,16 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
  */
 
 /** "login" options */
-struct login_options {};
+struct login_options {
+	/** No Username */
+	int nouser;
+};
 
 /** "login" option list */
-static struct option_descriptor login_opts[] = {};
+static struct option_descriptor login_opts[] = {
+	OPTION_DESC ( "nouser", 'n', no_argument,
+			   struct login_options, nouser, parse_flag ),
+};
 
 /** "login" command descriptor */
 static struct command_descriptor login_cmd =
@@ -61,7 +68,7 @@ static int login_exec ( int argc, char **argv ) {
 		return rc;
 
 	/* Show login UI */
-	if ( ( rc = login_ui() ) != 0 ) {
+	if ( ( rc = login_ui( opts.nouser ) ) != 0 ) {
 		printf ( "Could not set credentials: %s\n",
 			 strerror ( rc ) );
 		return rc;
diff --git a/src/hci/tui/login_ui.c b/src/hci/tui/login_ui.c
index 3c55325d..56fc2fa9 100644
--- a/src/hci/tui/login_ui.c
+++ b/src/hci/tui/login_ui.c
@@ -48,12 +48,12 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 #define EDITBOX_COL		( ( COLS / 2U ) - 10U )
 #define EDITBOX_WIDTH		20U
 
-int login_ui ( void ) {
+int login_ui ( int nouser ) {
 	char username[64];
 	char password[64];
 	struct edit_box username_box;
 	struct edit_box password_box;
-	struct edit_box *current_box = &username_box;
+	struct edit_box *current_box = nouser ? &password_box : &username_box;
 	int key;
 	int rc = -EINPROGRESS;
 
@@ -76,11 +76,15 @@ int login_ui ( void ) {
 	color_set ( CPAIR_NORMAL, NULL );
 	erase();
 	attron ( A_BOLD );
-	mvprintw ( USERNAME_LABEL_ROW, LABEL_COL, "Username:" );
+	if ( ! nouser ) {
+		mvprintw ( USERNAME_LABEL_ROW, LABEL_COL, "Username:" );
+	}
 	mvprintw ( PASSWORD_LABEL_ROW, LABEL_COL, "Password:" );
 	attroff ( A_BOLD );
 	color_set ( CPAIR_EDIT, NULL );
-	draw_editbox ( &username_box );
+	if ( ! nouser ) {
+		draw_editbox ( &username_box );
+	}
 	draw_editbox ( &password_box );
 
 	/* Main loop */
@@ -94,11 +98,15 @@ int login_ui ( void ) {
 			current_box = &password_box;
 			break;
 		case KEY_UP:
-			current_box = &username_box;
+			if ( ! nouser ) {
+				current_box = &username_box;
+			}
 			break;
 		case TAB:
-			current_box = ( ( current_box == &username_box ) ?
-					&password_box : &username_box );
+			if ( ! nouser ) {
+				current_box = ( ( current_box == &username_box ) ?
+						&password_box : &username_box );
+			}
 			break;
 		case KEY_ENTER:
 			if ( current_box == &username_box ) {
diff --git a/src/include/ipxe/login_ui.h b/src/include/ipxe/login_ui.h
index 313e0734..07b1dd49 100644
--- a/src/include/ipxe/login_ui.h
+++ b/src/include/ipxe/login_ui.h
@@ -9,6 +9,6 @@
 
 FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 
-extern int login_ui ( void );
+extern int login_ui ( int nouser );
 
 #endif /* _IPXE_LOGIN_UI_H */
```

NiKiZe, 2024-06-21:
You can now do things like this in, https://ipxe.org/cmd/present

Closing as completed, if I'm wrong then I'm happy to reopen.

closed 2024-06-21 by NiKiZe