ipxe/ipxe #1061: String manipulation methods

kingsumos, 2023-10-20:
Hi,

Looks like we don't have any possibility for string manipulation inside iPXE scripts. I found this pull request which was rejected for some reason:
https://github.com/ipxe/ipxe/pull/95
The proposal above will be very helpful, e.g.:
show smbios/product:string:0:8
show smbios/uuid::0:24
echo ${ifname::3:4}

Other user looking for this feature:
https://forum.ipxe.org/showthread.php?tid=18573&pid=36076
(tried adding a post in forum 1st, but new users aren't allowed for registration)

Basically what I want is to use ${smbios/asset} for passing boot parameters to the script (ip/netmask/gateway/next-server/filename).

kingsumos, 2023-10-20:
It would be nice if iPXE can mimic bash-scripting parameter substitution functionality, reference link:
https://tldp.org/LDP/abs/html/parameter-substitution.html

NiKiZe, 2023-10-20:
String manipulations will not be added to iPXE, even if there is good usecases you are almost always better of sending strings to a server and let more competent languages handle these things.

closed 2023-10-20 by NiKiZe

kingsumos, 2023-10-21:
Ok no problem Mr. Lennart Poettering
I'm indeed very competent, good luck with your project

NiKiZe, 2023-10-21:
Better description then mine https://github.com/ipxe/ipxe/discussions/326#discussioncomment-629914