ipxe/ipxe #1069: x509 failure handling

hramrach, 2023-10-28:
Hello,

I have a server from which I cannot boot using iPXE due to x509 failure.

Building with DEBUG=x509 I get certificate dump on the VGA console of which I see only the final lines which is not immensely helpful.

After some discussion on irc the probable cause was identified as one of the intermediate certificated in the chain setting 'Certificate Name Constraints' as critical extension.

hramrach, 2026-03-05:
So if it's well documented what is likely to provide the relevant information and cut out the irrelevant this would be likely resolved.

hramrach, 2026-03-05:
Enabling debug as instructed provides no information because the information does not fit on the screen, as already reported.

The problem was diagnosed to be with intermediate certificate, not the server certificate. That means examining the server certificate will likely NOT show the extenstion that iPXE complains about.

Also if I as a user get unsupported extension what is the way forward?

Probably stop using iPXE if I am not willing to write the support if I even get through all the confusion to conclude that this is a feature missing in iPXE.

NiKiZe, 2026-03-05:
> Enabling debug as instructed provides no information because the information does not fit on the screen, as already reported.

Already tried to give suggestion on how to get the data, error page updated to include what was already suggested here, but also other ways.

> 
> The problem was diagnosed to be with intermediate certificate, not the server certificate. That means examining the server certificate will likely NOT show the extenstion that iPXE complains about.

Yes the full chain needs to be investigated, did it say to **only** check the server cert?

> 
> Also if I as a user get unsupported extension what is the way forward?
> 
> Probably stop using iPXE if I am not willing to write the support if I even get through all the confusion to conclude that this is a feature missing in iPXE.

Lets read what the page currently says again (fully aware of that this wasn't the case when the issue was created):
> Certificate have an extension that is marked as critical that iPXE don't support. Consider if this extension should be critical and if not change it and reissue certificate. If it really needs to be critical then some development will be needed in iPXE, the certificate or at least extension needs to be sent to a developer. (You need to enabled DEBUG as described above to see exactly what fails)

So:
### Consider if this extension should be critical and if not change it and reissue certificate.

To use iPXE, or any client, the server needs to be configured in a compatible way, this includes cipers, **and certificates**.
Or have the wanted features being developed, as I suggested earlier, open an issue that state requested feature.

One example here is letsencrypt which gave certificates that was not compatible, re-issue was needed.

There is even the option to run a fully self signed chain, so self signed ca certificate and cross-sign the problematic certificate with the issue, and have iPXE working with that as embeded/trusted chain.

The now updated error page should cover the initial difficulties around this issue.