ipxe/ipxe #1088: make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 bin-arm64-efi/ipxe.efi EMBED=boot.ipxe -C ipxe/src && cp ipxe/src/bin-x86_64-efi/ipxe.efi /var/lib/tftpboot/arm_uefi/

lx775289883, 2023-11-29:
[BUILD] bin-arm64-efi/arbel.ids.o
  [BUILD] bin-arm64-efi/qib7322.ids.o
  [BUILD] bin-arm64-efi/golan.ids.o
  [BUILD] bin-arm64-efi/xhci.ids.o
  [BUILD] bin-arm64-efi/uhci.ids.o
  [BUILD] bin-arm64-efi/ehci.ids.o
  [BUILD] bin-arm64-efi/rtl8185.ids.o
  [BUILD] bin-arm64-efi/rtl8180.ids.o
  [BUILD] bin-arm64-efi/ath5k.ids.o
  [BUILD] bin-arm64-efi/ath9k.ids.o
  [AR] bin-arm64-efi/blib.a
  [VERSION] bin-arm64-efi/version.ipxe.efi.o
  [LD] bin-arm64-efi/ipxe.efi.tmp
bin-arm64-efi/blib.a(config.o)：在函数‘__requiring_symbol__’中：
config.c:(.tbl.requiring_symbols+0x0)：对‘obj_video_subr’未定义的引用
make: *** [bin-arm64-efi/ipxe.efi.tmp] 错误 1
rm bin-arm64-efi/version.ipxe.efi.o
make: 离开目录“/home/ipxe/src”

mcb30, 2023-11-29:
Please temporarily set your locale to English, redo your test, and post the output that you get

lx775289883, 2023-11-30:
thanks。
make: Entering directory `/home/ipxe/src'
  [VERSION] bin-arm64-efi/version.ipxe.efi.o
  [LD] bin-arm64-efi/ipxe.efi.tmp
bin-arm64-efi/blib.a(config.o): In function `__requiring_symbol__':
config.c:(.tbl.requiring_symbols+0x0): undefined reference to `obj_video_subr'
make: *** [bin-arm64-efi/ipxe.efi.tmp] Error 1
rm bin-arm64-efi/version.ipxe.efi.o
make: Leaving directory `/home/ipxe/src'

mcb30, 2023-12-28:
You have defined `CONSOLE_DIRECT_VGA` in `config/console.h`.  Don't do that.

Build with the unmodified source tree and default configuration and everything will be fine.

closed 2023-12-28 by mcb30

lx775289883, 2023-12-29:
this is a new error

make: Entering directory `/home/ipxe/src'
  [DEPS] crypto/certstore.c
  [DEPS] crypto/privkey.c
make: Leaving directory `/home/ipxe/src'
make: Entering directory `/home/ipxe/src'
  [BUILD] bin-arm64-efi/__umoddi3.o
  [BUILD] bin-arm64-efi/__udivdi3.o
  [BUILD] bin-arm64-efi/__moddi3.o
  [BUILD] bin-arm64-efi/__divmoddi4.o
  [BUILD] bin-arm64-efi/icc.o
  [BUILD] bin-arm64-efi/__divdi3.o
  [BUILD] bin-arm64-efi/implicit.o
  [BUILD] bin-arm64-efi/__udivmoddi4.o
  [BUILD] bin-arm64-efi/null_nap.o
  [BUILD] bin-arm64-efi/gdbstub.o
  [BUILD] bin-arm64-efi/pc_kbd.o
  [BUILD] bin-arm64-efi/list.o
  [BUILD] bin-arm64-efi/uuid.o
  [BUILD] bin-arm64-efi/gdbudp.o
  [BUILD] bin-arm64-efi/memmap_settings.o
In file included from core/memmap_settings.c:27:0:
core/memmap_settings.c: In function ‘memmap_settings_fetch’:
include/errno.h:261:2: error: invalid 'asm': invalid operand prefix '%c'
  __asm__ ( ".section \".einfo\", \"\", " PROGBITS_OPS "\n\t" \
  ^
include/errno.h:514:16: note: in expansion of macro ‘__einfo_error’
 #define ENOENT __einfo_error ( EINFO_ENOENT )
                ^
core/memmap_settings.c:179:13: note: in expansion of macro ‘ENOENT’
     return -ENOENT;
             ^
include/errno.h:261:2: error: invalid 'asm': invalid operand prefix '%c'
  __asm__ ( ".section \".einfo\", \"\", " PROGBITS_OPS "\n\t" \
  ^
include/errno.h:514:16: note: in expansion of macro ‘__einfo_error’
 #define ENOENT __einfo_error ( EINFO_ENOENT )
                ^
core/memmap_settings.c:179:13: note: in expansion of macro ‘ENOENT’
     return -ENOENT;
             ^
core/memmap_settings.c: At top level:
cc1: error: unrecognized command line option "-Wno-address-of-packed-member" [-Werror]
cc1: error: unrecognized command line option "-Wno-dangling-pointer" [-Werror]
cc1: all warnings being treated as errors
make: *** [bin-arm64-efi/memmap_settings.o] Error 1
make: Leaving directory `/home/ipxe/src'

mcb30, 2023-12-29:
You seem to be using an extremely old version of gcc.  What do you get from `aarch64-linux-gnu-gcc --version`?