ipxe/ipxe #1091: Issue with Mellanox CX4 seemingly caused by 8b14652

nshalman, 2023-12-01:
I was running into weird boot failures on Mellanox Connect-X cards using the latest builds of iPXE including issues using netboot.xyz.

I have bisected the issue down to 8b14652e506d99499cfbeaed0df07d6a83ec029e

```console
$ git bisect log
git bisect start
# status: waiting for both good and bad commits
# good: [9062544f6a0c69c249b90d21a08d05518aafc2ec] [efi] Disable EFI watchdog timer when shutting down to boot an OS
git bisect good 9062544f6a0c69c249b90d21a08d05518aafc2ec
# status: waiting for bad commit, 1 good commit known
# bad: [115707c0edebad65f87525fed583fef73880016d] [iphone] Add missing va_start()/va_end() around reused argument list
git bisect bad 115707c0edebad65f87525fed583fef73880016d
# good: [f07630c74f3d67906baf820f080b5d0e5ad49ca4] [vlan] Support automatic VLAN device creation
git bisect good f07630c74f3d67906baf820f080b5d0e5ad49ca4
# good: [aa368ba529e13f07658e16507eb8686ef1c07423] [tls] Pass I/O buffer to received record handlers
git bisect good aa368ba529e13f07658e16507eb8686ef1c07423
# good: [c832580f197dac013edb72ce031570d66a9448f6] [efi] Pass more detailed driver information to veto methods
git bisect good c832580f197dac013edb72ce031570d66a9448f6
# good: [6701d91c5033a3804a3bb5d49a8f9f2212b901b4] [netdevice] Stop link block timer when device is closed
git bisect good 6701d91c5033a3804a3bb5d49a8f9f2212b901b4
# good: [eeb7cd56e54e2bc649626988872c170fba37c163] [netdevice] Remove netdev_priv() helper function
git bisect good eeb7cd56e54e2bc649626988872c170fba37c163
# good: [8cbf248198f3bc66c52b2340b4decf293af8af47] [vmware] Use driver-private data to hold GuestInfo settings block
git bisect good 8cbf248198f3bc66c52b2340b4decf293af8af47
# good: [56cc61a168820c7cbbe23418388129ec11699a8c] [eap] Define a supplicant model for EAP and EAPoL
git bisect good 56cc61a168820c7cbbe23418388129ec11699a8c
# bad: [ff0f860483e344f1af633f94696ff7bc1854611f] [libc] Use wall clock time as seed for the (non-cryptographic) RNG
git bisect bad ff0f860483e344f1af633f94696ff7bc1854611f
# bad: [8b14652e506d99499cfbeaed0df07d6a83ec029e] [eapol] Send EAPoL-Start packets to trigger EAP authentication
git bisect bad 8b14652e506d99499cfbeaed0df07d6a83ec029e
# first bad commit: [8b14652e506d99499cfbeaed0df07d6a83ec029e] [eapol] Send EAPoL-Start packets to trigger EAP authentication
```

Reverting that commit seems to fix the issue.

I'm happy to help further debug the issue including providing an environment for testing any proposed fixes.

<details>

<summary>Output captured from the console of a machine demonstrating the issue</summary>

```
iPXE> chain http://OBSCURED:8000/ipxe.efi
http://OBSCURED:8000/ipxe.efi... ok
iPXE initialising devices...ok

iPXE 1.0.0+ -- Open Source Network Boot Firmware -- https://ipxe.org/
Features: DNS HTTP HTTPS NFS TFTP VLAN EFI Menu
Welcome to iPXE Stress Test!
Configuring (net0 98:03:9b:89:d9:36)..................... ok
https://OBSCURED/image.tar.gz................ Connection timed out (https://ipxe.org/4c116092)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
```

Verbose report from the Dell BIOS
```
PowerEdge R6415 - BIOS 1.19.0
A system restart is required. The system detected an exception during the UEFI
[... 999 lines cut ...]
 85  00  0 1022:1455 13-00-00 00  Other Controller
 85  00  1 1022:1468 10-80-00 00  Other Controller
 C0  00  0 1022:1450 06-00-00 00  Bridge Controller
 C0  00  2 1022:1451 08-06-00 00  Base System Controller
 C0  01  0 1022:1452 06-00-00 00  Bridge Controller
 C0  01  1 1022:1453 06-04-00 00  Bridge decodes buses C1-C1
 C0  02  0 1022:1452 06-00-00 00  Bridge Controller
 C0  03  0 1022:1452 06-00-00 00  Bridge Controller
 C0  03  1 1022:1453 06-04-00 00  Bridge decodes buses C2-C2
 C0  03  2 1022:1453 06-04-00 00  Bridge decodes buses C3-C3
 C0  03  3 1022:1453 06-04-00 00  Bridge decodes buses C4-C4
 C0  03  4 1022:1453 06-04-00 00  Bridge decodes buses C5-C5
 C0  04  0 1022:1452 06-00-00 00  Bridge Controller
 C0  07  0 1022:1452 06-00-00 00  Bridge Controller
 C0  07  1 1022:1454 06-04-00 00  Bridge decodes buses C6-C6
 C0  08  0 1022:1452 06-00-00 00  Bridge Controller
 C0  08  1 1022:1454 06-04-00 00  Bridge decodes buses C7-C7
 C1  00  0 1000:0097 01-07-00 02  SAS Controller
 C6  00  0 1022:145A 13-00-00 00  Other Controller
 C6  00  2 1022:1456 10-80-00 00  Other Controller
 C7  00  0 1022:1455 13-00-00 00  Other Controller
 C7  00  1 1022:1468 10-80-00 00  Other Controller
Log Size: 43,962
```
</details>

nshalman, 2023-12-01:
The tested revert was specifically fbc3b4a104698658202c2a83217ca8722453bf49
https://github.com/ipxe/ipxe/compare/master...nshalman:ipxe:fbc3b4a104698658202c2a83217ca8722453bf49

NiKiZe, 2023-12-01:
This sounds like a duplicate of #1048 which should be fixed in current master. Can you verify which commit you have checked out?

nshalman, 2023-12-01:
I may not have tested on the latest master. Thank you for the pointer.

nshalman, 2023-12-01:
My test fails on the latest commit of master (98dd25a3bb2d3aafa71f088cbabf89418a783132)

```
http://147.28.150.231:8000/ipxe.efi... ok
iPXE initialising devices...ok

iPXE 1.0.0+ -- Open Source Network Boot Firmware -- https://ipxe.org
Features: DNS HTTP HTTPS NFS TFTP VLAN EFI Menu
Welcome to iPXE Stress Test Embedded Script!
Configuring (net0 98:03:9b:89:d9:36)..................... ok
https://artifacts.platformequinix.com/images/ubuntu/22_04/fe3f18eead9ab1bf6a333294198cdb6cdf918290/image.tar.gz.................. Connection timed out (https://ipxe.org/4c116092)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
flexboot_nodnic_ports_register_dev: port register_dev failed (Status = -336093320)
flexboot_nodnic_probe: flexboot_nodnic_ports_register_dev failed (Status = -336093320)
```

nshalman, 2023-12-05:
Just confirming that additional testing confirms that Mellanox CX4 cards are having trouble once booted into the latest commit of iPXe (98dd25a3bb2d3aafa71f088cbabf89418a783132) but the problems go away if I apply my revert commit (fbc3b4a104698658202c2a83217ca8722453bf49)

What additional debugging information would be of use for tracking down the issue?

NiKiZe, 2023-12-05:
What is the card connected to, and what do you see on the wire?

ech68, 2023-12-19:
I can also verify that reverting that commit fixes failure to boot with Mellanox CX5 nics as well

ech68, 2024-01-10:
I tested the latest version again, as I saw some more commits related to eapol went in, but this is still broken.

It appears that the code in eapol.c, where it says "Ignore non-EAPol devices" isn't ignoring these Mellanox cards, because if I just add another unconditional "return 0;" before the "Initialize structure" comment, then my hosts w/ Mellanox boot interfaces will work.

Rinaldo-lsw, 2024-01-17:
Hello,

I'm working for a relatively big hosting company and we also noticed that iPXE is broken for a while on Mellanox cards.

As an example we have new HP RL300 ARM servers and these chassis have an onboard Mellanox card.
`Mellanox Technologies MT2894 Family [ConnectX-6 Lx]`

This issue is not limited to this specific model, we also have 25GbE+ Mellanox cards that are acting in the same way.

We are still on commit `cac3a584dc8acea1522669f1ed16e0979fb92252` which works for Mellanox cards. 
However, anything after will break PXE boot.

Cornelicorn, 2024-02-15:
To add a data point to the reports: We had the same issue with `Mellanox Technologies MT27710 Family [ConnectX-4 Lx]` and "fixed" it by adding an`#undef NET_PROTO_EAPOL` to our build config.

Smithx10, 2024-02-20:
Ran into this issue with Mellanox CX5 and CX6, rebasing https://github.com/ipxe/ipxe/commit/fbc3b4a104698658202c2a83217ca8722453bf49 this onto main got them booting again.

nshalman, 2024-02-20:
@Smithx10 can you try the workaround suggested by @Cornelicorn and report back if it helped as it's a much less invasive workaround to tweak a define than backing the code out entirely. I haven't had a chance to test for myself.

> To add a data point to the reports: We had the same issue with `Mellanox Technologies MT27710 Family [ConnectX-4 Lx]` and "fixed" it by adding an`#undef NET_PROTO_EAPOL` to our build config.

stappersg, 2024-02-21:
From the `iPXE` _IRC channel_:
```text
21:00 < Redacted> I'm trying to boot a Mellanox ConnectX5 card and
  ran into  Configuring (net2 a0:88:c2:6b:7f:44).................. No
  configuration methods succeeded (https://ipxe.org/040ee119)
21:00 < Redacted> in both bios and uefi 
21:00 < Redacted> Is there some gotcha with these Mellanox cards ?
21:06 < stappers> https://ipxe.org/040ee119
21:14 < Redacted> @stappers  think I might be hitting
https://github.com/ipxe/ipxe/issues/1091 ?
21:24 < stappers> Keep thinking and act upon the better thoughts, at
  least try to do.
21:55 < Redacted> Interesting, rolling back to
  https://github.com/ipxe/ipxe/tree/8f1514a00450119b04b08642c55aa674bdf5a4ef
  worked, Im applying this
  https://github.com/ipxe/ipxe/commit/fbc3b4a104698658202c2a83217ca8722453bf49
  and seeing what happens
21:58  * stappers is in UTC+1 and goes sleeping
22:36 < Redacted> Yea,  just confirmed,  mellanox worked after rebasing
  that commit onto main
```
I as non mellanox hardware owner, are with the mellanox hardware owners: _Somebody else should provide a merge request_

nshalman, 2024-02-21:
> I as non mellanox hardware owner, are with the mellanox hardware owners: _Somebody else should provide a merge request_

I don't think my revert commit is a good solution. I believe @mcb30 is working on a better solution.
Of the short term fixes I can currently think of, changing the default for `NET_PROTO_EAPOL` to be undefined is one option, assuming that that workaround works.

I am going to update my description of this bug to include the suggestion that folks attempt https://github.com/ipxe/ipxe/issues/1091#issuecomment-1946152615 before patching the source.

As I have said before, I haven't had time to test that myself, but it seems very likely to me that it is a much simpler workaround.

danmcd, 2024-02-22:
Adding a comment so I can watch.  I have a downstream and was planning on updating with `master` soon.  Would very much like this fixed before I accept the merge. (And apparently I have the guilty commit in two releases of our downstream.)

NiKiZe, 2024-03-17:
Could we get confirmation if this is fixed by the merge of #1174, thanks

danmcd, 2024-03-18:
> Could we get confirmation if this is fixed by the merge of #1174, thanks

I'm going to ask our ops team to try it out on an affected box.  We have, in the interim, removed EAPOL support from Triton's downstream of ipxe, since we don't use it anyway currently.  See TritonDataCenter/ipxe#25 .

ech68, 2024-03-18:
Still failed for me when building with this latest change and eapol enabled
again, booting from snponly.

mcb30, 2024-03-18:
> Still failed for me when building with this latest change and eapol enabled again, booting from snponly.
> […](#)

If you are using snponly then there's a good chance that the underlying SNP driver provided by Mellanox has the same bug, since Mellanox uses a shared driver codebase for both iPXE and their UEFI SNP driver.  There's nothing we can do about the bug being present in the underlying SNP driver.

Please try using `ipxe.efi` instead of `snponly.efi` so that the updated iPXE driver (including the fix) is used to drive the hardware instead.

ech68, 2024-03-18:
With ipxe.efi this does work for my test host, though we can’t use that
bootloader due to issues that occur when many different nics are installed
in a system.

mcb30, 2024-03-18:
> With ipxe.efi this does work for my test host, though we can’t use that bootloader due to issues that occur when many different nics are installed in a system.

Thank you for testing.

Your result indicates that the issue is fixed in iPXE, so I will close this issue now.  If you want to continue using snponly.efi, you will need to contact your UEFI BIOS vendor to get a BIOS update that includes the equivalent fix in the BIOS-provided SNP driver.

You can also open a separate issue to cover whatever problem you are seeing that prevents you from using ipxe.efi when many different NICs are installed.

danmcd, 2024-03-18:
> Your result indicates that the issue is fixed in iPXE, so I will close this issue now. If you want to continue using snponly.efi, you will need to contact your UEFI BIOS vendor to get a BIOS update that includes the equivalent fix in the BIOS-provided SNP driver.
> 

Triton Data Center needs snponly.efi (and undionly.kpxe for BIOS) as well, so our testing would likely fail as well.  (To that our, our downstream will maintain exclusing EAPOL for now.)

mcb30, 2024-03-18:
The 18-byte packet will be zero-padded to 60 bytes on the wire anyway (64 bytes including the Ethernet FCS), since that is the minimum length Ethernet packet.

We could possibly work around the underlying SNP driver bug by pointlessly zero-padding the packet to 60 bytes ourselves.  That would be sufficient to avoid the underlying bug in the SNP driver (assuming that it is using code identical to that fixed in commit https://github.com/ipxe/ipxe/commit/c11734eee).

@ech68 could you please retest snponly.efi built from https://github.com/ipxe/ipxe/pull/1177 ?

ech68, 2024-03-18:
On Mon, Mar 18, 2024 at 11:40 AM Michael Brown ***@***.***>
wrote:

> @ech68 <https://github.com/ech68> could you please retest snponly.efi
> built from #1177 <https://github.com/ipxe/ipxe/pull/1177> ?
>
Same failure mode unfortunately.

mcb30, 2024-03-18:
> On Mon, Mar 18, 2024 at 11:40 AM Michael Brown ***@***.***> wrote: @ech68 <https://github.com/ech68> could you please retest snponly.efi built from #1177 <#1177> ?
> Same failure mode unfortunately.

Thanks for testing.  Does `ifstat` report the driver as SNP or NII when you are using snponly.efi?

ech68, 2024-03-18:
On Mon, Mar 18, 2024 at 12:41 PM Michael Brown ***@***.***>
wrote:

> Thanks for testing. Does ifstat report the driver as SNP or NII when you
> are using snponly.efi?
>
NII

>

mcb30, 2024-03-18:
> > Thanks for testing. Does ifstat report the driver as SNP or NII when you are using snponly.efi?
>
> NII

Thanks.  I've generalised the PR to cover both SNP and NII, and force-pushed PR #1177.  Could you please retest with this commit?

mcb30, 2024-03-18:
> > Thanks. I've generalised the PR to cover both SNP and NII, and
> > force-pushed PR #1177 <https://github.com/ipxe/ipxe/pull/1177>. Could you
> > please retest with this commit?
>
> with that update, it works!

Fantastic, thank you! Could you let me know your name and email for the commit log testing credit?

ech68, 2024-03-18:
On Mon, Mar 18, 2024 at 1:13 PM Michael Brown ***@***.***>
wrote:

> Fantastic, thank you! Could you let me know your name and email for the
> commit log testing credit?
>
Eric Hagberg, ***@***.***

>

mcb30, 2024-03-18:
> > Fantastic, thank you! Could you let me know your name and email for the commit log testing credit?
>
> Eric Hagberg, \*\*\*@\*\*\*.\*\*\*

I think there may be some kind of automated censorship system at work here. :upside_down_face:

stappersg, 2024-03-18:
> > > Fantastic, thank you! Could you let me know your name and email for the commit log testing credit?
> > 
> > 
> > Eric Hagberg, ***@***.***
> 
> I think there may be some kind of automated censorship system at work here. 🙃

In https://github.com/ipxe/ipxe/pull/1177/commits is an email address ( `mcb30 AT ipxe . org` )  Eric, please mail to that address directly to by-pass the automated censorship.

mcb30, 2024-03-18:
Merged with the Tested-by credit: thank you!

closed 2024-03-18 by mcb30

sburette, 2024-05-21:
Hi,

The `undionly` driver does not manage to establish DHCP through Mellanox ConnectX-5 on machine booting in BIOS mode.

As per DHCP server logs, the DHCP request is never received (guessing it is never sent out of machine).

Setting `#undef  NET_PROTO_EAPOL` fixes the issue.

Would it be possible to have similar fix in `undi` driver, as the ones in `snp` and `nii` ?

Here are the iPXE logs when compiling with `DEBUG=undinet:3,undi:3,dhcp:3,dhcppkt:3,dhcpmgmt:3`

```
DHCP 0x21984 DHCPDISCOVER
DHCP 0x21984 deferring discovery
DHCP 0x21984 entering discovery state
..DHCP 0x21984 DHCPDISCOVER
.DHCP 0x21984 entering request state
DHCP 0x21984 DHCPREQUEST to <DHCP_SERVER_IP> for <MACHINE_IP>
.DHCP 0x21984 DHCPREQUEST to <DHCP_SERVER_IP> for <MACHINE_IP>
DHCP 0x21984 DHCPREQUEST to <DHCP_SERVER_IP> for <MACHINE_IP>
.DHCP 0x21984 DHCPREQUEST to <DHCP_SERVER_IP> for <MACHINE_IP>
..DHCP 0x21984 DHCPREQUEST to <DHCP_SERVER_IP> for <MACHINE_IP>
....DHCP 0x21984 DHCPREQUEST to <DHCP_SERVER_IP> for <MACHINE_IP>
........ No configuration methods succeeded (https://ipxe.org/040ee119)
UNDINIC 0x21314 closed
```

Shall I open a new issue about the UNDI driver case ?

NiKiZe, 2024-05-21:
> Hi,
>
> The `undionly` driver does not manage to establish DHCP through Mellanox ConnectX-5 on machine booting in BIOS mode.
>
> As per DHCP server logs, the DHCP request is never received (guessing it is [...]

Your best fix might be to get updated firmware, or use ipxe.pxe instead.

Refs: https://github.com/ipxe/ipxe/issues/1091#issuecomment-2004025556 
https://github.com/ipxe/ipxe/issues/1091#issuecomment-2004138664

andres-moya, 2025-09-13:
Hello. I just hit the same issue. Cloned latest ipxe and 100G ConnectX-5. Cards. 
Had to use ipxe.pxe, that will scan all MB interfaces and because ipxe staged it will loop over 4 interfaces 2 times. 

Finally found about NET_PROTO_EAPOL. Ah, NICs are connected to Z9332 Dell switch. Maybe something with its configuration 
too.

undionly was able to send DHCP discover, get reply, and stuck on sending DHCP requests. Server never see that 3rd packet. 

Disabling NET_PROTO_EAPOL fixed issue.

I will try to play with firmware versions on NICs as well. Maybe need upgrade or downgrade.

PR #1177 (2024-03-18): [snp] Pad transmit buffer length to work around buggy vendor drivers
The Mellanox/Nvidia UEFI SNP driver is built from the same codebase as the iPXE driver, and appears to contain the bug that was fixed in commit c11734e ("[golan] Use ETH_HLEN for inline header size").  This results in identical failures when using the SNP interface (via e.g. snponly.efi) to drive a Mellanox card while EAPoL is enabled.

Work around the underlying UEFI SNP driver bug by padding transmit I/O buffers to the minimum Ethernet frame length before passing them to the underlying SNP driver's Transmit() function.

Thi [... cut]