ipxe/ipxe #1097: Waiting for link-up on net0... Connection timed out

asdil12, 2023-12-13:
[image]

It was built using:
```
make EMBED=boot.ipxe bin/ipxe.pxe bin-x86_64-efi/ipxe.efi
```

boot.ipxe:
```
#!ipxe

echo Welcome to NFSLinux iPXE

dhcp

chain --replace --autofree http://${next-server}/tftpboot/menu.ipxe ||
chain --replace --autofree tftp://${next-server}/menu.ipxe ||
```

Booting works fine in a VM (that actually uses iPXE for booting my iPXE bootloader)
and on a different UEFI hardware machine.

But on this old BIOS-PXE Asus Laptop it seems to timeout for some reason - even though the Intel UNDI PXE was able to boot into iPXE.

asdil12, 2023-12-13:
It seems to work if using undionly.kpxe instead of ipxe.pxe.

NiKiZe, 2023-12-13:
What does ifstat show before and after your dhcp attempt?

asdil12, 2023-12-13:
[image]

NiKiZe, 2023-12-13:
The only nic it finds is a wireless card AR9485
 By using undionly.kpxe instead it uses the undo stack from bios PXE ROM.

Which nic is it you are trying to boot, and what is the pciid?

asdil12, 2023-12-13:
I'm trying to boot from the onboard wired ethernet:
```
03:00.0 Ethernet controller [0200]: Qualcomm Atheros QCA8172 Fast Ethernet [1969:10a0] (rev 10)
        Subsystem: ASUSTeK Computer Inc. Device [1043:200f]
        Flags: bus master, fast devsel, latency 0, IRQ 19
        Memory at f7c00000 (64-bit, non-prefetchable) [size=256K]
        I/O ports at e000 [size=128]
        Capabilities: [40] Power Management version 3
        Capabilities: [58] Express Endpoint, MSI 00
        Capabilities: [c0] MSI: Enable- Count=1/16 Maskable+ 64bit+
        Capabilities: [d8] MSI-X: Enable+ Count=16 Masked-
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [180] Device Serial Number ff-6a-b3-be-74-d0-2b-ff
        Kernel driver in use: alx
        Kernel modules: alx
```

kcinickgx, 2024-06-20:
try net1
```
ifopen net1
dhcp net1
```
the screenshot says the card is on 02, the quote says 03, maybe is detecting 2 cards, the wireless first (net0), the wired second (net1)