ipxe/ipxe #1099: NETDEV net0 link is down

claell, 2023-12-14:
When trying to boot an iPXE ISO image, we get the following output (already using debug options):

[image]

https://ipxe.org/err/380861 says:

> check that the cable is plugged in, and that any link lights are on

This should be the case.

The other link (https://ipxe.org/err/3c6540) points towards a driver problem?

Thus, I am writing here, to hopefully get a quick fix.

Am happy to provide more information like specific device info, etc.

iPXE version is the latest from GitHub master.

NiKiZe, 2023-12-14:
Get to an iPXE shell and run ifstat, post output.
Also helpful if you have pciid of your NIC(s)

claell, 2023-12-14:
How can I get into such a shell?

NiKiZe, 2023-12-14:
Build without embedded script and press ctrl-b, or use shell in embedded script.

claell, 2023-12-14:
Yup, found out that my ISO is probably with embedded script, so I used the prebuilt version from https://ipxe.org/download.

Below the output.

[image]

For PCI ID, I didn't find a quick way to see that on Windows. Not sure whether one of the screenshots from device manager contain the relevant information?

[image]
[image]
[image]

Also note, that we use a separate NIC, not the onboard card.
[image]

NiKiZe, 2023-12-14:
So your machine has one intel 219LM-7 which is found by iPXE native drivers, and one Marvell 1D6A 07B1 which isn't supported by native drivers. Your simplest workaround is to plug your cable in to the supported NIC ;)

For this nic to be supported development needs to funded... However consider boot with efi instead, and use SNP.efi build of iPXE to use any NIC provided network stack instead.

claell, 2023-12-14:
Alright. We already thought that this might be a solution, at least as workaround. But then, we couldn't have 10 Gigabit Ethernet.

Are there already Linux drivers available for that NIC? (i.e., how much effort is needed to support it, or how much funding for development)?

We'll also look into EFI. Will that then use the UEFI provided network interface (not an expert on this)?

NiKiZe, 2023-12-14:
Development of iPXE drivers is done from scratch, hardware manuals for the NIC (chip) is required to do the work. See iPXE.org for commercial support.

If the NIC has a ROM that provides an efi network stack, then that will be used in EFI mode if no native support is available.

claell, 2023-12-18:
Thanks for your quick support! We'll look into it and then decide, which way is the best.

claell, 2023-12-20:
Adding further information for reference (after contacting commercial support): #370 seems to be the relevant PR for this.

claell, 2023-12-20:
And this issue might then be a duplicate of #294?