ipxe/ipxe #1103: Latest commit 3fc1b40 fails to compile with GCC 8.3.1 on Centos 7.9.

threepistons, 2023-12-20:
I've got a laptop with an Intel I219-LM network adaptor (pci id 8086:1a1e) that runs iPXE's `dhcp` command but won't chainload the menu and I suspect that I need to upgrade the ipxe binary on my build server to support the network adaptor. Other, older laptops will chainload the same menu as expected.

Compiling commit 3fc1b40 gives the following output:

```[user@unixboot src]$ cd ~ ; rm -rf ipxe/ ; git clone https://github.com/ipxe/ipxe.git ; cd ipxe/src ; make bin-x86_64-efi/ipxe.efi EMBED=/home/campus/user/autobooter.ipxe
Cloning into 'ipxe'...
remote: Enumerating objects: 59385, done.
remote: Counting objects: 100% (4716/4716), done.
remote: Compressing objects: 100% (466/466), done.
remote: Total 59385 (delta 4288), reused 4428 (delta 4244), pack-reused 54669
Receiving objects: 100% (59385/59385), 19.08 MiB | 17.44 MiB/s, done.
Resolving deltas: 100% (44469/44469), done.
  [PARSEROM]
[[many lines omitted to keep the character count below 65536]]
  [BUILD] bin-x86_64-efi/xhci.ids.o
  [BUILD] bin-x86_64-efi/uhci.ids.o
  [BUILD] bin-x86_64-efi/ehci.ids.o
  [BUILD] bin-x86_64-efi/hvm.ids.o
  [BUILD] bin-x86_64-efi/rtl8185.ids.o
  [BUILD] bin-x86_64-efi/rtl8180.ids.o
  [BUILD] bin-x86_64-efi/ath5k.ids.o
  [BUILD] bin-x86_64-efi/ath9k.ids.o
  [AR] bin-x86_64-efi/blib.a
  [HOSTCC] util/elf2efi64
util/elf2efi.c: In function ‘process_reloc’:
util/elf2efi.c:831:30: error: ‘R_X86_64_GOTPCRELX’ undeclared (first use in this function); did you mean ‘R_X86_64_GOTPCREL’?
   case ELF_MREL ( EM_X86_64, R_X86_64_GOTPCRELX ) :
                              ^~~~~~~~~~~~~~~~~~
util/elf2efi.c:80:46: note: in definition of macro ‘ELF_MREL’
 #define ELF_MREL( mach, type ) ( (mach) | ( (type) << 16 ) )
                                              ^~~~
util/elf2efi.c:831:30: note: each undeclared identifier is reported only once for each function it appears in
   case ELF_MREL ( EM_X86_64, R_X86_64_GOTPCRELX ) :
                              ^~~~~~~~~~~~~~~~~~
util/elf2efi.c:80:46: note: in definition of macro ‘ELF_MREL’
 #define ELF_MREL( mach, type ) ( (mach) | ( (type) << 16 ) )
                                              ^~~~
util/elf2efi.c:832:30: error: ‘R_X86_64_REX_GOTPCRELX’ undeclared (first use in this function); did you mean ‘R_X86_64_GOTPCREL’?
   case ELF_MREL ( EM_X86_64, R_X86_64_REX_GOTPCRELX ) :
                              ^~~~~~~~~~~~~~~~~~~~~~
util/elf2efi.c:80:46: note: in definition of macro ‘ELF_MREL’
 #define ELF_MREL( mach, type ) ( (mach) | ( (type) << 16 ) )
                                              ^~~~
util/elf2efi.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-dangling-pointer’ [-Werror]
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:1442: util/elf2efi64] Error 1
```

threepistons, 2023-12-20:
`git reset --hard HEAD~2` prior to compiling removes the error messages and allows compilation to finish, so the problem been introduced since 337880d.

threepistons, 2023-12-20:
The failure to chainload turns out to be DNS not working so well in the building I'm in.  Replacing `chain tftp://server.domain/menu.ipxe` with `chain tftp://ipaddress/menu.ipxe` solved that problem.

NiKiZe, 2023-12-20:
Thanks for the report!

When was gcc 8.3 released?
If you need to build with old versions of gcc you might need to stay on older commits.
You should be able to use a Docker image or other VM with a never set of build tools to build current master.

It might be harder and harder to stay compatible, not sure how long back we really should (or must?) stay compatible?

threepistons, 2023-12-20:
https://gcc.gnu.org/gcc-8 says May 2019.  The GCC shipped with my CentOS 7 system was 4.something, I had to use https://www.softwarecollections.org/en/scls/rhscl/devtoolset-8/ to get 8.3 and there isn't a later devtoolset release.

Would an iPXE compiled on an Ubuntu system work on a CentOS 7 server?

mcb30, 2023-12-20:
The `R_X86_64_GOTPCRELX` et al errors should be resolved by a trivial patch to `elf2efi.c` to add extra constants to the `/* Provide constants missing on some platforms */` section.  I'll push a PR for this later this evening, if I get time.

threepistons, 2023-12-20:
> Would an iPXE compiled on an Ubuntu system work on a CentOS 7 server?

That's one of my stupider questions of today. Of course it will, because the laptop under build is executing the iPXE binary.

mcb30, 2023-12-21:
> The `R_X86_64_GOTPCRELX` et al errors should be resolved by a trivial patch to `elf2efi.c` to add extra constants to the `/* Provide constants missing on some platforms */` section. I'll push a PR for this later this evening, if I get time.

OK, this part at least should be solved in commit https://github.com/ipxe/ipxe/commit/9e92c3989 (and in wimboot in commit https://github.com/ipxe/wimboot/commit/47f6298)

mcb30, 2023-12-21:
With the current master (commit https://github.com/ipxe/ipxe/commit/9e92c3989) I now get a successful build in a CentOS 7.9 container:
```console
[root@7a3b91da0507 src]# cat /etc/redhat-release 
CentOS Linux release 7.9.2009 (Core)
[root@7a3b91da0507 src]# gcc --version
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
```