ipxe/ipxe #1107: Certificates issued by Digicert GeoTrust Cloud DV are not trusted

ciorisgithub, 2023-12-21:
Digicert uses GeoTrust TLS RSA CA G1 as intermediate certificate and DigiCert Global Root G2 as root certificate. Can those be added to the https://ca.ipxe.org/? Adding those manually using TRUST=... and CERT=... works as expected, but it would be good to be added so they are automatically trusted.

Thanks

NiKiZe, 2023-12-21:
iPXE CA is based on Mozilla, are they missing there?

ciorisgithub, 2023-12-21:
I couldn't find them in there.... unless I'm missing something...

Paper-Dragon, 2023-12-22:
issue by https://letsencrypt.org/zh-cn/certificates/ also it,sometime can use, 
here https://github.com/ipxe/ipxe/discussions/1105

Paper-Dragon, 2023-12-22:
> issue by https://letsencrypt.org/zh-cn/certificates/ also it,sometime can use, here #1105

[image]

NiKiZe, 2023-12-22:
I can't read that... This is not an issue for many others, are you sure this isn't due to some kind of TLS interception with an invalid certificate?

What does ssllabs say? https://www.ssllabs.com/ssltest/analyze.html?d=netboot.geekery.cn&latest

iPXE CA is unlikely to do anything that diverges from Mozilla, if it is missing there (unlikely) then it is there it needs to be added.

Opening multiple issues about the same actuall issue is wasting resources. even if well intended.

Paper-Dragon, 2023-12-23:
Have you solved it?

closed 2024-02-15 by mcb30