ipxe/ipxe #1110: Unrecognised relocation type 22  in  build in loong64 !

jackiehe, 2023-12-27:
when i build  snponly.efi in loong64,  when finish have error.  i test all  loongarch64 operations and cross compile.  please help!

  make bin-loong64-efi/snponly.efi ARCH=loong64
  [PARSEROM]
  [DEPS] core/version.c
  [DEPS] arch/loong64/core/setjmp.S
  [DEPS] drivers/net/ath/ath9k/ath9k_xmit.c
  [DEPS] drivers/net/ath/ath9k/ath9k_recv.c
  [DEPS] drivers/net/ath/ath9k/ath9k_main.c
....
  [BUILD] bin-loong64-efi/ath9k.ids.o
  [AR] bin-loong64-efi/blib.a
  [HOSTCC] util/elf2efi64
  [VERSION] bin-loong64-efi/version.snponly.efi.o
  [LD] bin-loong64-efi/snponly.efi.tmp
  [FINISH] bin-loong64-efi/snponly.efi
Unrecognised relocation type 22
make: *** [Makefile.efi:34：bin-loong64-efi/snponly.efi] 错误 1
rm bin-loong64-efi/version.snponly.efi.o

mcb30, 2023-12-27:
I am able to successfully cross-compile `bin-loong64-efi/snponly.efi` using:
```
make CROSS=loongarch64-linux-gnu- bin-loong64-efi/snponly.efi
```

Relocation type 22 is `R_LARCH_SOP_PUSH_PCREL`.  Given the presence of `PCREL`, this is almost certainly a PC-relative relocation that could be handled via:
```diff
--- a/src/util/elf2efi.c
+++ b/src/util/elf2efi.c
@@ -140,6 +140,9 @@
 #ifndef R_LARCH_64
 #define R_LARCH_64 2
 #endif
+#ifndef R_LARCH_SOP_PUSH_PCREL
+#define R_LARCH_SOP_PUSH_PCREL 22
+#endif
 #ifndef R_LARCH_B26
 #define R_LARCH_B26 66
 #endif
@@ -849,6 +852,7 @@ static void process_reloc ( struct elf_file *elf, const Elf_Shdr *shdr,
                case ELF_MREL ( EM_AARCH64, R_AARCH64_LDST32_ABS_LO12_NC ) :
                case ELF_MREL ( EM_AARCH64, R_AARCH64_LDST64_ABS_LO12_NC ) :
                case ELF_MREL ( EM_AARCH64, R_AARCH64_LDST128_ABS_LO12_NC ) :
+               case ELF_MREL ( EM_LOONGARCH, R_LARCH_SOP_PUSH_PCREL ):
                case ELF_MREL ( EM_LOONGARCH, R_LARCH_B26):
                case ELF_MREL ( EM_LOONGARCH, R_LARCH_PCALA_HI20 ):
                case ELF_MREL ( EM_LOONGARCH, R_LARCH_PCALA_LO12 ):
```

However, it would be useful to check that this relocation is being generated sanely.  Could you please show your output from running:
```
loongarch64-linux-gnu-objdump -dr bin-loong64-efi/blib.a | grep -C30 R_LARCH_SOP_PUSH_PCREL
```

jackiehe, 2023-12-27:
thank you for you help,  after i patch the /src/util/elf2efi.c and rebuild,   the error change to  "Unrecognised relocation type 23"   ！ i upload the blib.a objdump .

[dumpblib.zip](https://github.com/ipxe/ipxe/files/13779363/dumpblib.zip)

make bin-loong64-efi/snponly.efi EMBED=myscript.ipxe CROSS=loongarch64-linux-gnu- ARCH=loong64
  [FINISH] bin-loong64-efi/snponly.efi
Unrecognised relocation type 23
make: *** [Makefile.efi:34：bin-loong64-efi/snponly.efi] error 1

jackiehe, 2023-12-27:
i check the LoongArch ELF ABI specification,  the   elf2efi.c  lost many ELF relocation types, please patch the elf2efi.c  complete.

https://loongson.github.io/LoongArch-Documentation/LoongArch-ELF-ABI-EN.html

22   R_LARCH_SOP_PUSH_PCREL  Push PC-relative offset push (S - PC + A)
23 R_LARCH_SOP_PUSH_ABSOLUTE Push constant or absolute address push (S + A)
32 R_LARCH_SOP_SUB Stack top operation opr2 = pop (), opr1 = pop (), push (opr1 - opr2)
33 R_LARCH_SOP_SL Stack top operation opr2 = pop (), opr1 = pop (), push (opr1 << opr2)
34 R_LARCH_SOP_SR Stack top operation  opr2 = pop (), opr1 = pop (), push (opr1 >> opr2)
40 R_LARCH_SOP_POP_32_S_10_12  Instruction imm-field relocation opr1 = pop (), (*(uint32_t *) PC) [21 ... 10] = opr1 [11 ... 0] with check 12-bit signed overflow
45 R_LARCH_SOP_POP_32_S_0_10_10_16_S2 Instruction imm-field relocation opr1 = pop (), (*(uint32_t *) PC) [9 ... 0] = opr1 [27 ... 18],(*(uint32_t *) PC) [25 ... 10] = opr1 [17 ... 2]with check 28-bit signed overflow and 4-bit aligned

jackiehe, 2023-12-27:
please use this new version of cross compile tools chain for test!

http://www.loongnix.cn/zh/toolchain/GNU/

download link:  http://ftp.loongnix.cn/toolchain/gcc/release/loongarch/gcc8/loongson-gnu-toolchain-8.3-x86_64-loongarch64-linux-gnu-rc1.2.tar.xz

code:  http://ftp.loongnix.cn/toolchain/gcc/release/loongarch/gcc8/loongson-gnu-toolchain-8.3-src-loongarch64-linux-gnu-rc1.2.tar

mcb30, 2023-12-27:
On 27/12/2023 15:32, jackiehe wrote:
> please use this new version of cross compile tools chain for test!
>
> http://www.loongnix.cn/zh/toolchain/GNU/
> <http://www.loongnix.cn/zh/toolchain/GNU/>
>
> download link: [...]

GCC 8 is extremely old.  I am getting a successful compile with GCC 13.2.1.

You will need at least GCC 13 to be able to compile iPXE for LoongArch64.

Please verify that you can still reproduce the problem with an 
up-to-date toolchain.

Thanks,

Michael

jackiehe, 2023-12-27:
ok, thank you .   i will test in new  GCC 13 in real  LoongArch64  computer to test.   but  the  cross compile tools chain  have old version only .    when i complete,  i will tell you .  thank you .

mcb30, 2023-12-27:
In case it helps: I am cross-compiling on a standard Fedora 39 installation, where the cross-compiling GCC 13 is available from the standard distro package repository via just:
```
dnf install gcc-loongarch64-linux-gnu
```

yetist, 2024-01-08:
To build ipxe for LoongArch requires gcc13+, we dont support the old relocation types in upstream code.

mcb30, 2024-01-08:
> To build ipxe for LoongArch requires gcc13+, we dont support the old relocation types in upstream code.

Thanks.  I'll mark this as closed.

closed 2024-01-08 by mcb30