ipxe/ipxe #1118: Initial make fails

leon47331, 2024-01-09:
Hi!
I installed all the packages and dependencies and was going to do "make".
It aborted with the following error:
```
  [...]
  [AR] bin/blib.a
  [HOSTCC] util/zbin
  [VERSION] bin/version.ipxe.dsk.o
  [LD] bin/ipxe.dsk.tmp
  [BIN] bin/ipxe.dsk.bin
  [ZINFO] bin/ipxe.dsk.zinfo
  [ZBIN] bin/ipxe.dsk.zbin
  [FINISH] bin/ipxe.dsk
  [VERSION] bin/version.ipxe.lkrn.o
  [LD] bin/ipxe.lkrn.tmp
  [BIN] bin/ipxe.lkrn.bin
  [ZINFO] bin/ipxe.lkrn.zinfo
  [ZBIN] bin/ipxe.lkrn.zbin
  [FINISH] bin/ipxe.lkrn
  [GENFSIMG] bin/ipxe.iso
util/genfsimg: could not find isolinux.bin
make: *** [arch/x86/Makefile.pcbios:68: bin/ipxe.iso] Fehler 1
rm bin/ipxe.dsk.zinfo bin/ipxe.lkrn.bin bin/ipxe.lkrn.zinfo bin/ipxe.lkrn.zbin bin/ipxe.dsk.zbin bin/version.ipxe.dsk.o bin/ipxe.dsk.bin bin/version.ipxe.lkrn.o

```

What exactly am I doing wrong?

NiKiZe, 2024-01-09:
You do not have the isolinux package installed as described at https://ipxe.org/download#source_code

Either install that package if you want to build Isos, or use a different make command that does not include creating Isos.

A good tip is to always search for existing issues, before creating new ones.

closed 2024-01-09 by NiKiZe

leon47331, 2024-01-09:
I haven't found another issue, that's why I created one...
But still thank you.

NiKiZe, 2024-01-09:
Ok well https://github.com/search?q=repo%3Aipxe%2Fipxe+isolinux&type=issues shows some #775 has the same message with explanation. Several of the results has the same error message with follow up explanations.

Easiest might be to use your package manager and install isolinux. But how will you use iPXE in the end? Do you want the legacy pcbios iso?

leon47331, 2024-01-09:
Thank you for clearing that up
Maybe I wasn't looking enough

I've ended up re-installing isolinux via Terminal which was working fine.
I wanna boot off of several operating systems and use WinPE

Do you have the legacy pcbios iso? I would take that.