ipxe/ipxe #1123: iPXE unable to operate Intel 82599ES NIC in EFI mode

andrewbonney, 2024-01-12:
Hi. I'm encountering an issue which looks similar to https://github.com/ipxe/ipxe/issues/503. In this case it's with an 82599ES based NIC on a Supermicro motherboard. The board is running the latest available BIOS and is set up for mixed legacy/EFI operation. The NIC works correctly in legacy mode, but in EFI mode iPXE cannot find the NIC.

Upon loading an iPXE EFI build via TFTP the output reads 'No more network devices', and issuing 'ifstat' at the shell gives a blank output.

Using an iPXE build with Intel and EFI debug enabled indicates that the existing drivers could not be disconnected, and the PCI device could not be opened:
[image]

Using an iPXE SNP build doesn't indicate a disconnection issue, and the NII driver attaches. This later attempts to DHCP but fails at that stage. Finally, when attempting to re-attach the old drivers upon exit the system hangs.
[image]

If I use a UEFI shell to attempt driver disconnection I receive a 'Status = Not Found' error for each interface (2x dual NIC cards), but reviewing 'devtree' after issuing these commands the devices have indeed disappeared, so it seems the disconnect commands are successful:
[image]

Can you suggest any further debug steps I could take, or any other information I could provide?

Thanks

NiKiZe, 2024-01-12:
I assume this is with ipxe.efi? If so have you tried snponly.efi?

andrewbonney, 2024-01-12:
Hi. The first output is with ipxe.efi. The second is with snp.efi. I haven't tried snponly directly, but I assume that would do the same as snp.efi but just for the interface it was launched against?

NiKiZe, 2024-01-12:
So the NICs SNP implementation is potentially broken? Could you test disabling the NICs firmware and see if you then get different result?

andrewbonney, 2024-01-12:
It appears that with the firmware disabled the standard ipxe.efi works correctly for all NICs, where previously it couldn't see any of them. The firmware was only enabled on one NIC but appears to have prevented them all working.

Does this indicate a NIC firmware issue which can't be resolved via any tweak to ipxe?

NiKiZe, 2024-01-12:
It indeed indicates a firmware issue that should be reported to the manufacturer.

iPXE might be able to work around parts of the issue. But since it isn't working with snponly then it might have some way to big issues.

Just to be sure, verify that you have latest firmware. If you can, document which versions of the firmware.

andrewbonney, 2024-01-12:
This is occurring with the latest Intel BootUtil flashed image (28.3). I've got an older version from Supermicro which I'll try shortly and report back.

andrewbonney, 2024-01-17:
Just to update this, we see the same behaviour having flashed the NIC's EEPROM and boot firmware using Supermicro's versions.

andrewbonney, 2024-01-17:
I've just been pointed at https://github.com/ipxe/ipxe/discussions/553#discussioncomment-7758663 which I'll try out as a fix

NiKiZe, 2024-01-17:
What is Supermicros supports response on this?
Since SNP isn't working the efi driver really should be fixed, and it also works without the efi ROM enabled.

andrewbonney, 2024-01-17:
Supermicro note that the specific NIC is many years old, having last had an update in 2018, so they won't be able to support it via new updates. If we purchase any new hardware with the same chip (in a newer NIC) and have the same issue I'll follow up with them further.

I have just retried with snponly.efi having previously tried snp.efi. This loads over TFTP with the boot ROM enabled, and can find the interface, but then fails when attempting to configure the interface.

However, if I modify the veto patch I linked to so that it matches the driver version identified from the EFI shell, this does appear to work for ipxe.efi, again via TFTP with the boot ROM enabled. This does at least mean we have a workaround.