ipxe/ipxe #1140: Can't build iPXE due to error: call to ‘build_assert_805’ declared with attribute error

g3rhard, 2024-02-01:
Hello, moved subject from discussion to issues:
```
[BUILD] bin/aes.o
In file included from <command-line>:
crypto/aes.c: In function ‘aes_gcm_setkey’:
include/assert.h:82:10: error: call to ‘build_assert_805’ declared with attribute error: build_assert(( ( void * ) &context->raw ) == ( ( void * ) context->gcm.raw_ctx )) failed
82 | C2 ( build_assert, LINE ) ();
| ^
./include/compiler.h:46:21: note: in definition of macro ‘_C1’
46 | #define _C1( x, y ) x ## y
| ^
include/assert.h:82:4: note: in expansion of macro ‘_C2’
82 | C2 ( build_assert, LINE ) ();
| ^~~
include/ipxe/gcm.h:92:2: note: in expansion of macro ‘build_assert’
92 | build_assert ( ( ( void * ) &context->gcm ) == ctx );
| ^~~~~~~~~~~~
crypto/aes.c:805:1: note: in expansion of macro ‘GCM_CIPHER’
805 | GCM_CIPHER ( aes_gcm, aes_gcm_algorithm,
| ^~~~~~~~~~
crypto/aes.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-dangling-pointer’ [-Werror]
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:964: bin/aes.o] Error 1
```

For build I use latest Ubuntu Focal docker image.

Links:
* https://forums.fogproject.org/topic/17207/ipxe-binary-compile-error/2
* https://github.com/ipxe/ipxe/discussions/1126

mcb30, 2024-02-10:
May be fixed as of commit https://github.com/ipxe/ipxe/commit/94b39fbe9 - please retest

g3rhard, 2024-02-12:
Thanks, from my side can confirm - it works as expected.

vvs-, 2024-03-25:
The same problem happened compiling on NixOS with the latest `master`. Only reverting `94b39fbe9298160b034c93ca06deb39a907e3b3f` helped in my case.

mcb30, 2024-03-25:
> The same problem happened compiling on NixOS with the latest ipxe `master`. Only reverting [94b39fb](https://github.com/ipxe/ipxe/commit/94b39fbe9298160b034c93ca06deb39a907e3b3f) helped in my case.

So you're saying that commit https://github.com/ipxe/ipxe/commit/94b39fbe9298160b034c93ca06deb39a907e3b3f which **fixes** the problem with gcc 4.8.5 is the commit which **causes** the exact same problem for you?  This seems very unlikely.  Which version of gcc are you using?

vvs-, 2024-03-25:
gcc version 12.2.0 (GCC)

Without reverting I'm getting consistent `build_asserts` in gcm.c

vvs-, 2024-03-25:
```
  [BUILD] bin/gcm.o
In file included from <command-line>:
crypto/gcm.c: In function ‘gcm_setiv’:
include/assert.h:82:31: error: call to ‘build_assert_474’ declared with attribute warning: build_assert(&context->hash < &context->ctr) failed [-Werror=attribute-warning]
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                               ^
./include/compiler.h:46:21: note: in definition of macro ‘_C1’
   46 | #define _C1( x, y ) x ## y
      |                     ^
include/assert.h:82:25: note: in expansion of macro ‘_C2’
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                         ^~~
crypto/gcm.c:474:9: note: in expansion of macro ‘build_assert’
  474 |         build_assert ( &context->hash < &context->ctr );
      |         ^~~~~~~~~~~~
include/assert.h:82:31: error: call to ‘build_assert_475’ declared with attribute warning: build_assert(&context->len < &context->ctr) failed [-Werror=attribute-warning]
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                               ^
./include/compiler.h:46:21: note: in definition of macro ‘_C1’
   46 | #define _C1( x, y ) x ## y
      |                     ^
include/assert.h:82:25: note: in expansion of macro ‘_C2’
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                         ^~~
crypto/gcm.c:475:9: note: in expansion of macro ‘build_assert’
  475 |         build_assert ( &context->len < &context->ctr );
      |         ^~~~~~~~~~~~
include/assert.h:82:31: error: call to ‘build_assert_476’ declared with attribute warning: build_assert(&context->ctr < &context->key) failed [-Werror=attribute-warning]
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                               ^
./include/compiler.h:46:21: note: in definition of macro ‘_C1’
   46 | #define _C1( x, y ) x ## y
      |                     ^
include/assert.h:82:25: note: in expansion of macro ‘_C2’
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                         ^~~
crypto/gcm.c:476:9: note: in expansion of macro ‘build_assert’
  476 |         build_assert ( &context->ctr < &context->key );
      |         ^~~~~~~~~~~~
include/assert.h:82:31: error: call to ‘build_assert_477’ declared with attribute warning: build_assert(( ( void * ) &context->raw_cipher ) > ( ( void * ) &context->key )) failed [-Werror=attribute-warning]
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                               ^
./include/compiler.h:46:21: note: in definition of macro ‘_C1’
   46 | #define _C1( x, y ) x ## y
      |                     ^
include/assert.h:82:25: note: in expansion of macro ‘_C2’
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                         ^~~
crypto/gcm.c:477:9: note: in expansion of macro ‘build_assert’
  477 |         build_assert ( ( ( void * ) &context->raw_cipher ) >
      |         ^~~~~~~~~~~~
include/assert.h:82:31: error: call to ‘build_assert_479’ declared with attribute warning: build_assert(( ( void * ) context->raw_ctx ) > ( ( void * ) &context->key )) failed [-Werror=attribute-warning]
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                               ^
./include/compiler.h:46:21: note: in definition of macro ‘_C1’
   46 | #define _C1( x, y ) x ## y
      |                     ^
include/assert.h:82:25: note: in expansion of macro ‘_C2’
   82 |                         _C2 ( build_assert_, __LINE__ ) ();                  \
      |                         ^~~
crypto/gcm.c:479:9: note: in expansion of macro ‘build_assert’
  479 |         build_assert ( ( ( void * ) context->raw_ctx ) >
      |         ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:964: bin/gcm.o] Error 1
```

mcb30, 2024-03-27:
@vvs- Could you please try https://github.com/ipxe/ipxe/tree/gcmassert to see if that form of the assertions fixes the build issues in your environment?

vvs-, 2024-03-28:
Thanks, I confirm: it now works for me again.

Let's hope that gcc doesn't break it in the future.

mcb30, 2024-03-29:
Merged as https://github.com/ipxe/ipxe/commit/37850e0e8

closed 2024-03-29 by mcb30