ipxe/ipxe #1141: Feature request: support for /32 DHCP allocations

XioNoX, 2024-02-01:
Hi,

Thanks for iPXE, I'm slowly getting familiar with it.

On Linux it's possible to assign a /32 v4 IP to an interface (usually for VMs) with a specific route for outbound traffic. This allows to efficiently use IP space, without having to rely on tricks like proxy ARP.
For example on the VM side I can have `inet 10.192.24.3/32 scope global ens13`
As well as 
```
# ip route
default via 10.192.24.1 dev ens13 onlink 
10.192.24.1 dev ens13 scope link
```
Then on the hypervisor side `inet 10.192.24.1/32 scope link tap0`
and `10.192.24.3 dev tap0 proto static scope link`
(Note that the gw IP doesn't have to be from the same prefix as the VM IP)
This works very well.
It's also possible to have ISC-DHCP hand out DHCP offers compatible with that setup, see for example this [blogpost](https://blog.fhrnet.eu/2020/03/07/dhcp-server-on-a-32-subnet/).

The problem I'm facing is at the iPXE step. When DHCP offers such IP config to the client, iPXE accepts (see the /32 IP) it but then can't fetch the debian-installer files.
[image]

You can find some more of my troubleshooting steps (and DHCP exchange on this task [comment](https://phabricator.wikimedia.org/T300152#9503098).

Please let me know if I'm doing something wrong or if it's indeed a bug/missing feature. Happy to help, but I don't know C enough to send a PR :( 

Cheers,

NiKiZe, 2024-02-01:
This has been up for discussion before. (But can't find the post)
It is not a bug, limited ability to implement it in small code size.
Is there any RFC that defines this? I would agree that in a perfect world iPXE should have support for /32 on legacy IPv4, on the other hand, is it worth it? Why not use IPv6 with possibly NAT64 and DNS64 instead?

MalteJ, 2024-02-23:
+1

I can confirm the problem with IPv6: When the RA does not contain an ICMPv6 prefix information, iPXE should configure with the address from DHCPv6 with a /128 prefix length and add the router as advertised by the RA. Instead iPXE configures the DHCPv6 assigned IP with a /64 subnet and the default route is not configured.

CC: @mcb30

wfleurant, 2024-03-21:
additional information here. case is when iPXE boots off of say, a /24 and then the vlan switches from a /24 network to one handing out /32 networks. from iPXE shell you then run `dhcp` and you see a /32 network and gateway are installed. so at this point routing does not work.
a hack is to change netmask eg; `set net0/netmask 255.255.0.0`

mcb30, 2024-06-26:
Support added in #1243, tested with a /32 handed out by ISC dhcpd using
```
shared-network "slash32" {
	subnet 10.251.251.1 netmask 255.255.255.255 {}
	subnet 10.42.69.23 netmask 255.255.255.255 {
		option subnet-mask 255.255.255.255;
		option routers 10.251.251.1;
		range dynamic-bootp 10.42.69.23;
		...
	}
}
```
(i.e. DHCP host is 10.251.251.1, iPXE gets given 10.42.69.3/255.255.255.255)

closed 2024-06-26 by mcb30

mcb30, 2024-06-27:
> I can confirm the problem with IPv6: When the RA does not contain an ICMPv6 prefix information, iPXE should configure with the address from DHCPv6 with a /128 prefix length and add the default route as advertised by the RA. Instead iPXE configures the DHCPv6 assigned IP with a /64 subnet and the default route is not configured.

This problem (not technically related to the issue title) should have been fixed via #1245

PR #1243 (2024-06-26): [ipv4] Support small subnets with no directed broadcast address
In a small subnet (with a /31 or /32 subnet mask), all addresses within the subnet are valid host addresses: there is no separate network address or directed broadcast address.

The logic used in iPXE to determine whether or not to use a link-layer broadcast address will currently fail in these subnets.  In a /31 subnet, the higher of the two host addresses (i.e. the address with all host bits set) will be treated as a broadcast address.  In a /32 subnet, the single valid host address will be treated as a broadcast address.

Fix b [... cut]