ipxe/ipxe #1151: "No more network devices" on a supposedly supported network card

solskogen, 2024-02-14:
According to lspci on Linux it's a Realtek RTL8111/8168/8411 PCI Express (rev 15)

solskogen, 2024-02-15:
Ok, partly fixed by using snponly.efi, and not ipxe.efi.

NiKiZe, 2024-02-15:
iPXE does not have native drivers for every NIC,
You can reach out for commercial support if it is important, but for efi SNP/NII should work just fine.

With that said please post pciid (lspci -nn), and also search existing issues.

closed 2024-02-15 by NiKiZe

solskogen, 2024-02-15:
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)

NiKiZe, 2024-02-15:
Yes that does have native drivers, but it is a buggy NIC, check existing issues.
Detailed debug output is required to do any more here.