ipxe/ipxe #1161: Feature Request:Add multiple target which using qemu-img(1) as convert

naY9yjoS6ZqhOd35sIFH, 2024-02-24:
https://github.com/ipxe/ipxe/commit/dfbbc16ae392e8f4ab62e15a99cdfb32a06af557
as this commit,vhd pcbios target first intro to ipxe build target
but why not vhd efi or vmdk efi using qemu-img(1) too?

NiKiZe, 2024-02-24:
Should we add every target image format?
The simple answer is no.

So why was this added? Probably because it was needed in one of the test pipelines, and using the already existing makefile was the easiest path in that scenario.

However for most other scenarios you are better of converting from .usb yourself, or just use the raw .usb ad is.

In exactly which scenarios do you think more images are needed, and why should it be in the iPXE makefile?

naY9yjoS6ZqhOd35sIFH, 2024-02-28:
i see,i have written script to handle this pipeline

closed 2024-02-28 by NiKiZe