ipxe/ipxe #1164: Chelsio NICs don't play nice with NII

danmcd, 2024-02-29:
It's very possible this is known and exclusively a Chelsio problem. This modern FW update page:

    https://service.chelsio.com/store2/T5//PXE%20Boot%20Option%20ROM//Boot//PXE_2.4.0.0/Release%20Notes.txt
    
mentions under Errata:

```
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
ERRATA (Open Issues)
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

- 42108: Link timeout takes more than 5 minutes when peer side link is 
         administratively down. 

- 40457: Need support for iPXE.
<rest snipped>
```

I'm happy to help gather data, as I have the HW here.  I'm using a downstream, but even with the most recent merge (which is under test) I cannot get the right thing to happen.  The only thing I see from nii default DEBUG messages is:

```
NII NII-0000:5e:00.0 using UNDI v3.1 at 0x683a9b20 entry 0x688dfa98 impl 0x4000c5fa
NII NII-0000:5e:00.0 has memory BAR 0 and no I/O BAR
NII NII-0000:5e:00.0 registered as net0 for PciRoot(0x3)/Pci(0x0,0x0)/Pci(0x0,0x0)/MAC(0007434F76A0,0x0)
```

I also notice there's a discussion: #1050 related to this.

NiKiZe, 2024-02-29:
So you have NII interface to the NIC, and that might be a bad driver.

Does network boot work in any other scenario other than iPXE?
Do you get any errors when trying to use the NIC? (You didn't explain what isn't working, I expect error messages)

You will need current master, since we don't know what is changed in some other tree.

And as stated in the linked discussion, if you can't show this is a iPXE bug, please don't create issues.

danmcd, 2024-02-29:
(Sorry for not reading the discussion more carefully.)

Will close.  I have a WIP for updating with master.  I'm 90% certain it'll fail, and upon reading snoops I'm pretty sure the comment from Chelsio's release notes should be in the discussion and not here.