ipxe/ipxe #1175: ARM64 build failed

nbanb, 2024-03-17:
Hello 

I'm trying to build ipxe  on ARM64  to boot ARM64 linux image after from internet.

The trouble I encounered is when building, I saw some errors which let me think 'make' is trying to build i8086-32bit image instead of building a correct ARM image.

As you can imagine, build failed.

```bash
19:31:50 nba@14RV-FSRV-07-ws-build:~/ipxe/src$ make
  [BUILD] bin/__divdi3.o
gcc: error: unrecognized command-line option ‘-mpreferred-stack-boundary=2’
gcc: error: unrecognized command-line option ‘-mregparm=3’
gcc: error: unrecognized command-line option ‘-mrtd’
gcc: error: unrecognized command-line option ‘-m32’
make: *** [Makefile.housekeeping:964: bin/__divdi3.o] Error 1

19:39:45 nba@14RV-FSRV-07-ws-build:~/ipxe/src$ make bin/ipxe.iso
  [BUILD] bin/__divdi3.o
gcc: error: unrecognized command-line option ‘-mpreferred-stack-boundary=2’
gcc: error: unrecognized command-line option ‘-mregparm=3’
gcc: error: unrecognized command-line option ‘-mrtd’
gcc: error: unrecognized command-line option ‘-m32’
make: *** [Makefile.housekeeping:964: bin/__divdi3.o] Error 1

19:39:49 nba@14RV-FSRV-07-ws-build:~/ipxe/src$  # CPU detail 

19:39:49 nba@14RV-FSRV-07-ws-build:~/ipxe/src$ lscpu
Architecture:            aarch64
  CPU op-mode(s):        32-bit, 64-bit
  Byte Order:            Little Endian
CPU(s):                  1
  On-line CPU(s) list:   0
Vendor ID:               ARM
  Model name:            Cortex-A72
    Model:               1
    Thread(s) per core:  1
    Core(s) per cluster: 1
    Socket(s):           -
    Cluster(s):          1
    Stepping:            r0p1
    BogoMIPS:            50,00
    Flags:               fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
NUMA:                    
  NUMA node(s):          1
  NUMA node0 CPU(s):     0
Vulnerabilities:         
  Gather data sampling:  Not affected
  Itlb multihit:         Not affected
  L1tf:                  Not affected
  Mds:                   Not affected
  Meltdown:              Not affected
  Mmio stale data:       Not affected
  Retbleed:              Not affected
  Spec rstack overflow:  Not affected
  Spec store bypass:     Vulnerable
  Spectre v1:            Mitigation; __user pointer sanitization
  Spectre v2:            Vulnerable
  Srbds:                 Not affected
  Tsx async abort:       Not affected

`

Do someone know a way to build ipxe under arm64 ?

Thanks for help
Kind regards
nbanba

NiKiZe, 2024-03-17:
Yes, bin, the default make target, is i386 pcbios, please specify the correct build target for your platform.
Also note that ipxe.efi won't build, use snponly.efi or your specific native NIC driver.

You will also find other similar (if not duplicate) discussions and issues about ARM64 using the search function.

https://ipxe.org/appnote/buildtargets

closed 2024-03-17 by NiKiZe

mcb30, 2024-03-17:
The default when you type `make` is to build the i386 BIOS build of iPXE.  For arm64, you almost certainly want something like:
```
make bin-x86_64-efi/ipxe.efi
```

nbanb, 2024-03-17:
Thanks for answer and help !

nbanb, 2024-03-24:
The solution was : 

make bin-arm64-efi/ipxe.efi
sudo apt install mtools xorriso genisoimage
make bin-arm64-efi/ipxe.iso

Thanks again for help
Kind regards
nbanba