ipxe/ipxe #1179: Support for EC certificates

haarp, 2024-03-25:
Hello,

there's an issue when using HTTPS, `DOWNLOAD_PROTO_HTTPS` certain kinds of certificates and ipxe as of today's 390bce9.

On servers using the EC(256) certificates, the TLS handshake will fail. iPXE will log `Operation not permitted (https://ipxe.org/410de18f)` and `received fatal alert 40`. If the server is switched to using RSA certificates, TLS works as expected.

My crypto knowledge is limited, but the [crypto page](https://ipxe.org/crypto) reveals that only the RSA pubkey algorithm is supported, which seems to be the issue.

Unfortunately EC certs are becoming the norm. Tools like [lego](https://go-acme.github.io/lego/usage/cli/options/) use EC256 by default. It seems some servers like nginx allow running both RSA and EC certs in parallel, but that's not always an option, requires cooperation of admins and software, and understanding why iPXE fails in this manner in the first place. And either way, EC is on the way to replace RSA eventually.

I'd like to request that iPXE gain support for EC. I know this is far from trivial, but probably the best solution of dealing with this.

Thanks a lot! and thanks for making iPXE!

mcb30, 2024-03-25:
iPXE does not advertise support for EC server certificates.  We do have support for elliptic curves, but only for key exchange (i.e. the ECDHE variants of the cipher suites), and the only supported curve is x25519.

I'm happy to add support for EC certificates as soon as anyone wants to fund the development work.  (To preempt the inevitable future pull requests: please note that the cost of reviewing a contributed PR is **more** than the cost of doing the development work, because code review is always more time consuming than code creation.)

martafolf, 2024-03-27:
The lack of support for EC certs caused me some headaches today as I was trying to fetch a resource from a domain which was automatically enrolled with one. 

You mention that you're looking for funding for the development work and would prefer to do it yourself instead of others, would you be able to estimate that would look like how much funding would be needed for this to be implemented?

mcb30, 2024-03-28:
> You mention that you're looking for funding for the development work and would prefer to do it yourself instead of others, would you be able to estimate that would look like how much funding would be needed for this to be implemented?

I've replied directly by email.

gchmurka123, 2024-06-11:
Today I also had problems with EC256.
The `repo.almalinux.org` forced https (Let's Encrypt), now is unable to get initrd/kernel from http protocol.

AdamKearn, 2024-12-01:
Hi guys - has there been any development on implanting this?
I am trying to chainload from a HTTPS URL and getting the same error.

mcb30, 2024-12-01:
> Hi guys - has there been any development on implanting this?
> I am trying to chainload from a HTTPS URL and getting the same error.

Yes, the work is in progress. A whole batch of precursor commits have been pushed recently, speeding up some of the large integer arithmetic operations so that elliptic curves without special forms (eg NIST P-256) can be realistically supported.

AdamKearn, 2024-12-01:
> Yes, the work is in progress. A whole batch of precursor commits have been pushed recently, speeding up some of the large integer arithmetic operations so that elliptic curves without special forms (eg NIST P-256) can be realistically supported.

Thanks I really appreciate your work on this.   I am using Cloudflare to provide SSL currently and their free tier only supports ECDSA certs so once this has been implemented it would allow this to work.   Thanks again for your time.  :)

omniproc, 2024-12-13:
I believe the TLS ciphers negotiation is bugged. In my case the NGINX server was configured to offer both, RSA and EC based algorithms. When performing e.g. `imgfetch` running a recent master (commit 24db39f) I would get presented the mentioned `Operation not permitted (https://ipxe.org/410de18f)` error.

Please note that the list of server-offered ciphers included multiple which are supported as of today according to [crypto](https://ipxe.org/crypto). Example NGINX config that breaks:

Failing config
```ini
ssl_protocols TLSv1.2 TLSv1.3;
ssl_ciphers DHE-RSA-AES128-GCM-SHA256:DHE-RSA-AES256-GCM-SHA384:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-RSA-AES256-GCM-SHA384;
ssl_ecdh_curve  secp521r1:secp384r1:prime256v1;
```

To get a working config I had to either enable weaker ciphers that do not use ECDH at all, e.g. 
```ini
ssl_protocols TLSv1.2 TLSv1.3;
ssl_ciphers AES128-SHA256:AES256-SHA256:DHE-RSA-AES128-GCM-SHA256:DHE-RSA-AES256-GCM-SHA384:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-RSA-AES256-GCM-SHA384;
ssl_ecdh_curve  secp521r1:secp384r1:prime256v1;
```

OR add the X25519 ECDH curve explicitly:

```ini
ssl_protocols TLSv1.2 TLSv1.3;
ssl_ciphers DHE-RSA-AES128-GCM-SHA256:DHE-RSA-AES256-GCM-SHA384:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-RSA-AES256-GCM-SHA384;
ssl_ecdh_curve  X25519:secp521r1:secp384r1:prime256v1;
```

So as you can see even when a non EHCD cipher was offered, e.g. DHE-RSA-AES128-GCM-SHA256, as long as it was offered alonger with a EHCD cipher, e.g. ECDHE-RSA-AES128-GCM-SHA256, and X25519 was not allowed, the request would fail. Only [quite weak](https://ciphersuite.info/cs/TLS_RSA_WITH_AES_128_CBC_SHA256/) ciphers like AES128-SHA256 (this maps to TLS_RSA_WITH_AES_128_CBC_SHA256) would actually force iPXE to ignore ECDH ciphers. I guess this has to do with the current negotiation logic.

mcb30, 2024-12-16:
> So as you can see even when a non EHCD cipher was offered, e.g. DHE-RSA-AES128-GCM-SHA256, as long as it was offered alonger with a EHCD cipher, e.g. ECDHE-RSA-AES128-GCM-SHA256, and X25519 was not [...]

In the ClientHello, iPXE offers support for ECDHE and a list of named curves that currently includes only x25519.

It seems as though nginx is selecting ECDHE even when it has no curves in common with the client, and then failing because it can't select a curve to use from the empty set.  I've also seen this failure mode on the TLS-over-USB implemented on the iPhone, so I'm guessing it's a bug in a library that both of them use.

The server-side bug will be hidden once iPXE implements the NIST P-256 curve.  Until then, you'll have to work around the server bug by modifying the server side configuration (e.g. to allow x25519).

omniproc, 2024-12-16:
Yes, we're currently working around it as mentioned. I just wanted to add it to the discussion here to make you aware of those issues since I'm pretty sure I won't be the last one to run into this.

covers1624, 2025-02-26:
Is there any update on this? 

ECDSA is extremely popular these days it seems, some CDN's are only providing ciphers using it and its variations.
<details>
<summary>ssl labs cipers report of a particular cdn we wish to use</summary>

[image]

</details>

joernheissler, 2025-09-17:
> In my case the NGINX server was configured to offer both, RSA and EC based algorithms.

My new root chain is EC based and that workaround does not work:
I tried adding a self-signed RSA cert to nginx (parallel to the EC cert), but some other HTTP clients prefer RSA over EC and get served the wrong cert which they can't verify.

I also tried adding an RSA cert signed with my EC chain. That makes the other clients happy, but ipxe fails with https://ipxe.org/err/3c00e1 despite me adding that cert as trusted (CERT=... and TRUST=...).

Guess what I'll do for now is add a long-lived self-signed RSA cert with a different SNI.

closed 2026-09-09 by mcb30