ipxe/ipxe #1181: [ new feature ? ] send 'tput init' to terminal from ipxe

nbanb, 2024-03-30:
Dear all,

Here is a case which cause me a problem : 

I have made an ipxe menu (in ipxe script) which let people boot linux from internet directly on Debian netboot repository.
The menu is working perfectly the first time,  and let launch Debian console installer (or other installer) using the arrows keys and enter key after selecting the option.
The installer resize the terminal and load a 'dialog' interface.
If for any reason, I kill the installer and exit from it, the computer is rebooting and loading ipxe menu again.
But after this reboot, arrows keys are broken and are not working anymore on ipxe menu.

I'm accessing the server running ipxe through a serial console.
If I logout the serial console and send 'tput init' in the current terminal and connecting again the serial console, arrows keys started to work again.

So it's a terminal issue affecting xterm, not an ipxe issue, but it's affecting ipxe normal behaviour.

What I would like to so is to make ipxe sending to the terminal 'tput init' before loadnig the ipxe menu (that way, after rebooting the terminal would be reset to it's default behaviour and let arrows key work in every situation.

I did try to start my ipxe script with things like : 

```bash
console
```

or 
```bash
echo -n \033[H\033[2J\033[3J
echo -n \033c
```

or 
```bash
set rst:hex 30:33:33:63:0a
set rst ${rst:string}
echo ${rst}
```

or 
```bash
set esc:hex 1b 
set cls ${esc:string}[2J 
set rst ${esc:string}[3J
echo ${cls}
echo ${rst}
echo ${esc}
```

but I did not succeed... 

Does someone know a way to make ipxe send 'tput init' command to the terminal before loading the menue ?

Thanks and kind regards
nbanba

mcb30, 2024-04-02:
As a quick test, running:
```shell
tput init | od -t x1 -A n -w255 | tr -d " "
```
gives the output string `1b5b21701b5b3f333b346c1b5b346c1b3e1b5b3f36396c`

You can reproduce this output from iPXE using something like:
```
set tputinit:hexraw 1b5b21701b5b3f333b346c1b5b346c1b3e1b5b3f36396c
echo ${tputinit:string}
```

nbanb, 2024-04-03:
Thanks for help and answer

There is a quickest way to get this string : 

```bash
$ tput init | xxd -ps
1b5b21701b5b3f333b346c1b5b346c1b3e1b5b3f36396c
```

But unfortunatly it don't work.

I think I will have a deeper look on why a linux installer is able to send command which are interpreted by the terminal and not iPXE during the same session.

But thanks again for help
Regards
nbanba