ipxe/ipxe #1183: Provide checksums for downloads on boot.ipxe.org

TomTucka, 2024-04-03:
I'd like to verify the the items I'm downloading from boot.ipxe.org are what they say there are and have been built by ipxe.org. Are you able to start providing checksums or a signature file?

NiKiZe, 2024-04-03:
It is always recommended to build your own, the ones at boot.ipxe.org is only a convenience for those that really can't build.

If we get to a level where .efi versions are signed, it is likely that these will be published on GitHub, maybe even in a similar way that wimboot is, that way you will have digital signatures on them.

It would be nice if the boot.ipxe.org builds where done thru GitHub actions, and that way the extra signature step as well as reproducible builds could be there.

TomTucka, 2024-04-03:
I'm happy to do that work if it would be taken in

mcb30, 2024-04-03:
We already have GitHub Actions that build a variety of targets, so the logical thing to do is probably to allow builds on the master branch to publish to boot.ipxe.org (or to make boot.ipxe.org URLs redirect to fetch the latest commit's corresponding build artifact on GitHub, if that is simpler).

There is some precedent in the https://github.com/ipxe/wimboot repository for using an SSH key provided as a GitHub Actions secret to access files (Windows images) that happen to be hosted on boot.ipxe.org, so a similar approach could be viable.

This would definitely require a separate job within build.yml, with dependencies on all three of `x86`, `arm64`, and `arm32`, since we need binaries from all of them to create the combined `ipxe.iso` image.

GitHub Actions are remarkably tedious to debug and to alter, and so my main goal has always been to keep them as simple as possible.  As a rough goal: if you can implement the relevant "combining and publishing" job using no more lines of code than are currently used for the current `arm64` job, then it has a chance of being mergeable.

Signing is a separate issue.  I haven't looked in to what support exists within the GitHub Actions infrastructure for performing signing operations.  For example: does it allow us to provide a secret password to an external RSA token, so that we can issue a trusted certificate for the RSA token's public key and use the GitHub Actions secret to allow signing of binaries with that key (without the token's private key ever being exposed)?  I have no idea what kinds of facilities are provided, sorry.

NiKiZe, 2024-04-03:
Signing might be possible thru scard over network interfaces, but not sure if that is possible in gh actions.

For the rest, publishing to gh pages might be possible. The first step here could be to publish artifacts from each job. (There should be an upload artifacts action available)