ipxe/ipxe #1186: Sanboot --drive 0 fails

hac3ru, 2024-04-04:
Hello,

I just got an issue with Proxmox VMs when trying to boot from iPXE using sanboot:

- When I boot from PXE "manually", I mean, I press ESCAPE to show me the boot menu, pick Network Boot, I drop to shell and run: sanboot -d 0, it boots fine from the first HDD

- however, when I set the BIOS to boot automatically from PXE then HDD, it boots fine into the iPXE, drops me to shell and I run: sanboot -d 0 just to get an error:

```
Booting from SAN device 0x00
EFIBLK 0x00 cannot locate block i/O: Error 0x7f22208e (https://ipxe.org/7f22208e)
Any clue what's going on?
Boot from SAN device 0x00 failed: Error 0x7f22208e (https://ipxe.org/7f22208e)
```

P.S. I built ipxe from the master branch with DEBUG=efi_block

NiKiZe, 2024-04-04:
First check the URL in error message https://ipxe.org/7f22208e and follow the suggestions.

This is a newish feature, see https://github.com/ipxe/ipxe/commit/b1c13cc43ece3008f7012cf736fc943d5bb89131 maybe your firmware don't expose the disk thru efi when in "PXE mode" ?

hac3ru, 2024-04-04:
I checked that link already, but found nothing useful, unfortunately.

Weirdly enough, I created a new VM and it now works. Safe to say, not a fix, but I'm glad it works at the moment.

NiKiZe, 2024-04-04:
If you have the steps to reproduce, preferably with info about qemu version and command line or virsh dumpxml maybe someone can dig into it, otherwise if it can't be reproduced consider closing this issue. Thanks.

hac3ru, 2024-04-05:
Okay, unfortunately I can't seem to be able to reproduce it... I guess I'll close this and reopen it if I find a way to reliably reproduce the issue.
P.S. just as a curiosity: Why is the 1.21.1 version released in 2020 but the development is still active? Why not release 1.22.0 or something?

NiKiZe, 2024-04-05:
The current active version is git master notice the +
Se other discussions (and issues) on this topic, adding a new version number for every commit makes no sense ;)

closed 2024-07-04 by zube[bot]