ipxe/ipxe #1188: iPXE on Synology NAS ds1515+ - NICs link down.

hetii, 2024-04-08:
Hi. 
As I test ipxe on Synology NAS ds1515+ I notice two issues.

I compile my ipxe by:
`make bin-x86_64-efi/ipxe.efi DEBUG=intel,intelx,netdevice EMBED=SynoBootLoader.ipxe`

First is about networking cards.
All my interfaces claim that all links are in down state (all leds in RJ45 sockets are off):
  
  ```
   iPXE initialising devices...
  INTEL 0x79a05a58 MAC+PHY reset (08100241/80382780 was 08100241/80382780)
  INTEL 0x79a05a58 has autoloaded MAC address 00:a0:c9:00:00:00
  NETDEV net0 registered (phys 0000:00:14.0 hwaddr 00:a0:c9:00:00:00)
  INTEL 0x79a05a58 link status is 80382780
  NETDEV net0 link is down: Down (https://ipxe.org/38086193)
  INTEL 0x79a067b8 MAC+PHY reset (08100241/80382784 was 08100241/80382784)
  INTEL 0x79a067b8 has autoloaded MAC address 00:a0:c9:00:00:01
  NETDEV net1 registered (phys 0000:00:14.1 hwaddr 00:a0:c9:00:00:01)
  INTEL 0x79a067b8 link status is 80382784
  NETDEV net1 link is down: Down (https://ipxe.org/38086193)
  INTEL 0x79a08598 MAC+PHY reset (08100241/80382788 was 08100241/80382788)
  INTEL 0x79a08598 has autoloaded MAC address 00:a0:c9:00:00:02
  NETDEV net2 registered (phys 0000:00:14.2 hwaddr 00:a0:c9:00:00:02)
  INTEL 0x79a08598 link status is 80382788
  NETDEV net2 link is down: Down (https://ipxe.org/38086193)
  INTEL 0x79a0a418 MAC+PHY reset (08100241/8038278c was 08100241/8038278c)
  INTEL 0x79a0a418 has autoloaded MAC address 00:a0:c9:00:00:03
  NETDEV net3 registered (phys 0000:00:14.3 hwaddr 00:a0:c9:00:00:03)
  INTEL 0x79a0a418 link status is 8038278c
  NETDEV net3 link is down: Down (https://ipxe.org/38086193)
  ok
```
Opening interfaces don't change anything:

````
iPXE> ifopen net0
NETDEV net0 opening
INTEL 0x79a05a58 ring 03800 is at [7b749000,7b749100)
INTEL 0x79a05a58 ring 02800 is at [7b748000,7b748100)
INTEL 0x79a05a58 link status is 80382780
iPXE> INTEL 0x79a05a58 link status is 80382780

iPXE> ifopen net1
NETDEV net1 opening
INTEL 0x79a067b8 ring 03800 is at [7b747000,7b747100)
INTEL 0x79a067b8 ring 02800 is at [7b746000,7b746100)
INTEL 0x79a067b8 link status is 80382784
iPXE> INTEL 0x79a067b8 link status is 80382784
ifopen net2
NETDEV net2 opening
INTEL 0x79a08598 ring 03800 is at [7b745000,7b745100)
INTEL 0x79a08598 ring 02800 is at [7b744000,7b744100)
INTEL 0x79a08598 link status is 80382788
iPXE> INTEL 0x79a08598 link status is 80382788
ifopen net3
NETDEV net3 opening
INTEL 0x79a0a418 ring 03800 is at [7b743000,7b743100)
INTEL 0x79a0a418 ring 02800 is at [7b742000,7b742100)
INTEL 0x79a0a418 link status is 8038278c
iPXE> INTEL 0x79a0a418 link status is 8038278c
ifstat
net0: 00:a0:c9:00:00:00 using i354 on 0000:00:14.0 (Ethernet) [open]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net1: 00:a0:c9:00:00:01 using i354 on 0000:00:14.1 (Ethernet) [open]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net2: 00:a0:c9:00:00:02 using i354 on 0000:00:14.2 (Ethernet) [open]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net3: 00:a0:c9:00:00:03 using i354 on 0000:00:14.3 (Ethernet) [open]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
````

We can see the MACa are OUIs one, but even when I set correct one nothing change.

On other hand I compile pure kernel 6.8.4 and load igb drivers and all my NICs works fine, 
so this seams to be issue in intel driver implementation inside ipxe:

```
/lib/modules # insmod dca.ko
[   78.060729] dca: module verification failed: signature and/or required key missing - tainting kernel
[   78.071692] dca service started, version 1.12.1
/lib/modules # insmod  i2c-algo-bit.ko
/lib/modules # insmod igb.ko
[   92.618261] igb: Intel(R) Gigabit Ethernet Network Driver
[   92.624310] igb: Copyright (c) 2007-2014 Intel Corporation.
[   92.986075] igb 0000:00:14.0: added PHC on eth0
[   92.991166] igb 0000:00:14.0: Intel(R) Gigabit Ethernet Network Connection
[   92.998927] igb 0000:00:14.0: eth0: PBA No: 002100-000
[   93.004675] igb 0000:00:14.0: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[   93.370011] igb 0000:00:14.1: added PHC on eth1
[   93.375091] igb 0000:00:14.1: Intel(R) Gigabit Ethernet Network Connection
[   93.382852] igb 0000:00:14.1: eth1: PBA No: 002100-000
[   93.388599] igb 0000:00:14.1: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[   93.754345] igb 0000:00:14.2: added PHC on eth2
[   93.759448] igb 0000:00:14.2: Intel(R) Gigabit Ethernet Network Connection
[   93.767223] igb 0000:00:14.2: eth2: PBA No: 002100-000
[   93.772983] igb 0000:00:14.2: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
[   94.138483] igb 0000:00:14.3: added PHC on eth3
[   94.143581] igb 0000:00:14.3: Intel(R) Gigabit Ethernet Network Connection
[   94.151350] igb 0000:00:14.3: eth3: PBA No: 002100-000
[   94.157110] igb 0000:00:14.3: Using MSI-X interrupts. 4 rx queue(s), 4 tx queue(s)
/lib/modules # ip link set dev eth0 up
/lib/modules # [  130.605793] igb 0000:00:14.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
```
Now leds on eth0 are on.

I can analyze kernel drivers vs ipxe driver but if you had any tips what I should check first on ipxe side let me know.

B.R.

mcb30, 2024-04-08:
Try connecting a pair of ports on the NIC directly with a cable (i.e. **without** any network switch in between) and use the `lotest` command (enabled via `#define LOTEST_CMD` in `config/general.h`) to send packets from one port to another.  This should eliminate any possible issues related to your specific network setup, as a starting point.

hetii, 2024-04-08:
@mcb30 HI.

I checked different wires configuration and NICs without success:

I used normal cable not a crossover one if that matter.

Btw as I wrote before under linux all works fine, so this should not be related to my switches or wires.

```

 Type 'exit' to get the back to the menu
iPXE> ifstat
net0: 00:a0:c9:00:00:00 using i354 on 0000:00:14.0 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net1: 00:a0:c9:00:00:01 using i354 on 0000:00:14.1 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net2: 00:a0:c9:00:00:02 using i354 on 0000:00:14.2 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
net3: 00:a0:c9:00:00:03 using i354 on 0000:00:14.3 (Ethernet) [closed]
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Down (https://ipxe.org/38086193)]
iPXE> ifopen net0
NETDEV net0 opening
INTEL 0x79a05a58 ring 03800 is at [7b748000,7b748100)
INTEL 0x79a05a58 ring 02800 is at [7b747000,7b747100)
INTEL 0x79a05a58 link status is 80382780
iPXE> INTEL 0x79a05a58 link status is 80382780
ifopen net1
NETDEV net1 opening
INTEL 0x79a067b8 ring 03800 is at [7b746000,7b746100)
INTEL 0x79a067b8 ring 02800 is at [7b745000,7b745100)
INTEL 0x79a067b8 link status is 80382784
iPXE> INTEL 0x79a067b8 link status is 80382784
lotest
Usage:

  lotest [-m|--mtu <mtu>] [-b|--broadcast] <sending interface> <receiving interface>

See https://ipxe.org/cmd/lotest for further information
iPXE> lotest net0 net1
Waiting for link-up on net0......................................................... Operation canceled (https://ipxe.org/0b072095)
Test failed: Operation canceled (https://ipxe.org/0b072095)
iPXE> lotest net2 net3
NETDEV net2 opening
INTEL 0x79a08598 ring 03800 is at [7b744000,7b744100)
INTEL 0x79a08598 ring 02800 is at [7b743000,7b743100)
INTEL 0x79a08598 link status is 80382788
NETDEV net3 opening
INTEL 0x79a0a418 ring 03800 is at [7b742000,7b742100)
INTEL 0x79a0a418 ring 02800 is at [7b741000,7b741100)
INTEL 0x79a0a418 link status is 8038278c
INTEL 0x79a08598 link status is 80382788
Waiting for link-up on net2...INTEL 0x79a0a418 link status is 8038278c
........................... Operation canceled (https://ipxe.org/0b072095)
Test failed: Operation canceled (https://ipxe.org/0b072095)
            
iPXE> lotest net0 net1
Waiting for link-up on net0.................... Operation canceled (https://ipxe.org/0b072095)
Test failed: Operation canceled (https://ipxe.org/0b072095)
iPXE> lotest net1 net2
Waiting for link-up on net1........................ Operation canceled (https://ipxe.org/0b072095)
Test failed: Operation canceled (https://ipxe.org/0b072095)
iPXE> lotest net2 net3
Waiting for link-up on net2............................. Operation canceled (https://ipxe.org/0b072095)
Test failed: Operation canceled (https://ipxe.org/0b072095)
````

Grzegorz-Rey, 2024-04-08:
Additionally I play with  `src/drivers/net/intel.c` and `drivers/net/intelx.c`
where for `intel_poll()` I added recheck by intel_check_link ( netdev );
[image]
and even modify `intel_check_link()` to always set `netdev_link_up ( netdev );`
like:
[image]
but even when I open device and try get ip the link is dead:
[image]

I think that something is not ok in lower PHY layer and the NICs are not initialized properly or something missing in the driver. 
Also the intel driver use some timers to probe links and maybe it's some timing issue in sequence of checking PHY state.

Let me know if I can do anything to solve it.

stappersg, 2024-04-11:
> Let me know if I can do anything to solve it.

* Write more documentation about the project
* Make a _merge request_ that fixes [#1189](https://github.com/ipxe/ipxe/issues/1189)
* Keep that _MR_ in good condition
* Enable others to join the quest

hetii, 2024-04-12:
@stappersg 

The goal of this project is to be able to run iPXE on mentioned NAS and boot experimental systems that I want to prepare for it.

The given solution by me for https://github.com/ipxe/ipxe/issues/1189 is wrong, so make no sense to prepare merge request for broken code.

I mean, it works as a workaround to be able to have input, but should not be considered as a solution, as it block main thread and the main root cause why bit UART_LSR_DR is not set most of the time, is still unknown for me.

Current task about NIC is another story, as you can see from logs the card is detected but somehow link is not up.

I try already different settings about autospeed detection but I have very limited knowledge about writing such complex drivers.

Idea came to my mind, to strip linux igb driver (as it works) to very basic code were my NICs report link to be up and try to port that part to ipxe, but not sure if/when I be able to do it.

If we have someone onboard who already know intel drivers and know what can be check regarding registers and initialization part of it, then I be more then happy to spend some hours to check things up...

mcb30, 2024-04-15:
@hetii I was going to suggest trying the `INTEL_NO_PHY_RST` flag, but your original post shows the debug message
```
INTEL 0x79a05a58 MAC+PHY reset (08100241/80382780 was 08100241/80382780)
```
which indicates that the link was down when iPXE started execution (i.e. it's not the reset that's causing the link to drop).

I'd suggest checking the Linux driver to see what it does differently between i350 and i354.

Are the NIC ports connected to completely standard RJ-54 Gigabit Ethernet switch ports with autonegotiation enabled, or is there some kind of unusual network setup?

hetii, 2024-04-15:
@mcb30 Hi, 
I have a standard router Zbtlink ZBT-WE1326 connected to nic0 and in addition nic2 + nic3 are connected together, so all that interfaces goes up under linux without issue as soon as I call `ip link set up` on them.