ipxe/ipxe #1197: iPXE ISO 'breaking' Dell iDRAC LOM NIC

champtar, 2024-04-19:
I'm building a simple iPXE ISO using as follow
```
ISOVERSION=$(git show -s --format=%cs_%h)
make bin/ipxe.lkrn bin-x86_64-efi/ipxe.efi
./util/genfsimg -o ipxe-shell_${ISOVERSION}.iso -s script-shell.ipxe bin/ipxe.lkrn bin-x86_64-efi/ipxe.efi
```
script-shell.ipxe content
```
#!ipxe

echo iPXE provisioning ISO
shell
```

Also enabling a bunch of options:
```
--- a/src/config/general.h
+++ b/src/config/general.h
@@ -35,7 +35,7 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
  */
 
 #define        NET_PROTO_IPV4          /* IPv4 protocol */
-//#define NET_PROTO_IPV6       /* IPv6 protocol */
+#define        NET_PROTO_IPV6          /* IPv6 protocol */
 #undef NET_PROTO_FCOE          /* Fibre Channel over Ethernet protocol */
 #define        NET_PROTO_STP           /* Spanning Tree protocol */
 #define        NET_PROTO_LACP          /* Link Aggregation control protocol */
@@ -56,7 +56,7 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 
 #define        DOWNLOAD_PROTO_TFTP     /* Trivial File Transfer Protocol */
 #define        DOWNLOAD_PROTO_HTTP     /* Hypertext Transfer Protocol */
-#undef DOWNLOAD_PROTO_HTTPS    /* Secure Hypertext Transfer Protocol */
+#define        DOWNLOAD_PROTO_HTTPS    /* Secure Hypertext Transfer Protocol */
 #undef DOWNLOAD_PROTO_FTP      /* File Transfer Protocol */
 #undef DOWNLOAD_PROTO_SLAM     /* Scalable Local Area Multicast */
 #undef DOWNLOAD_PROTO_NFS      /* Network File System Protocol */
@@ -148,21 +148,21 @@ FILE_LICENCE ( GPL2_OR_LATER_OR_UBDL );
 #define LOGIN_CMD              /* Login command */
 #define SYNC_CMD               /* Sync command */
 #define SHELL_CMD              /* Shell command */
-//#define NSLOOKUP_CMD         /* DNS resolving command */
-//#define TIME_CMD             /* Time commands */
+#define NSLOOKUP_CMD           /* DNS resolving command */
+#define TIME_CMD               /* Time commands */
 //#define DIGEST_CMD           /* Image crypto digest commands */
 //#define LOTEST_CMD           /* Loopback testing commands */
-//#define VLAN_CMD             /* VLAN commands */
+#define VLAN_CMD               /* VLAN commands */
 //#define PXE_CMD              /* PXE commands */
-//#define REBOOT_CMD           /* Reboot command */
-//#define POWEROFF_CMD         /* Power off command */
+#define REBOOT_CMD             /* Reboot command */
+#define POWEROFF_CMD           /* Power off command */
 //#define IMAGE_TRUST_CMD      /* Image trust management commands */
-//#define PCI_CMD              /* PCI commands */
-//#define PARAM_CMD            /* Request parameter commands */
-//#define NEIGHBOUR_CMD                /* Neighbour management commands */
-//#define PING_CMD             /* Ping command */
-//#define CONSOLE_CMD          /* Console command */
-//#define IPSTAT_CMD           /* IP statistics commands */
+#define PCI_CMD                /* PCI commands */
+#define PARAM_CMD              /* Form parameter commands */
+#define NEIGHBOUR_CMD          /* Neighbour management commands */
+#define PING_CMD               /* Ping command */
+#define CONSOLE_CMD            /* Console command */
+#define IPSTAT_CMD             /* IP statistics commands */
 //#define PROFSTAT_CMD         /* Profiling commands */
 //#define NTP_CMD              /* NTP commands */
 //#define CERT_CMD             /* Certificate management commands */
```

I'm then booting the resulting ISO using iDRAC virtual CD
I'm not using the iDRAC dedicated network port but the shared port (shared LOM), and I pretty often (but not 100% of the time) completely loose connectivity to the iDRAC, here the screenshot (from iDRAC) when it freeze
[image]

I'm reproducing the issue with both Intel and Broadcom NICs (testing on 3 Dell servers)

When booting via ISO, is ipxe.efi fully loaded in RAM before execution ?
Without knowing anything about iPXE internals, maybe it reset the NIC, temporarily preventing reads from the virtual CD, so iPXE can't be loaded and can't finish resetting the NIC

I'm happy to run any debug version to diagnose this, just tell me what to do :)

NiKiZe, 2024-04-19:
iPXE loads it's own native NIC driver.
You could try snponly.efi
Also don't modify config until you have your basics, and then only modify the configs you actually need, and test them one by one to make sure it gives the change you expect. And do put the modifications only in config/local.

champtar, 2024-04-20:
Something I forgot to say, if I http boot ipxe.efi I don't have that issue.
Doing some more test now that I have another KVM connected to the server,
as soon as I ifopen the interface I get iDRAC access back.
ipxe.efi via http boot does ifopen the NIC to fetch autoexec.ipxe so that would explain why it work.

> iPXE loads it's own native NIC driver. You could try snponly.efi

Will do (or maybe snp.efi)

> Also don't modify config until you have your basics, and then only modify the configs you actually need, and test them one by one to make sure it gives the change you expect. And do put the modifications only in config/local.

Thanks for the config/local tip, I'll clean that up

mcb30, 2024-04-20:
iPXE is completely loaded from the virtual CD/DVD before it starts execution, so the loss of connection to the virtual CD/DVD will not cause any problem.

Linux kernel drivers tend to include extra convoluted logic to interoperate with the (generally very hacked-together) OEM remote access solutions such as iDRAC.  iPXE generally has much simpler drivers that don't include this extra logic, not least because it's so difficult to test without access to the hardware in question.  It seems very plausible that iPXE is leaving the link closed in a situation in which the OEM expects the driver to make special allowances for the existence of iDRAC by bringing the link back up.

The easiest workaround is probably for you to use an embedded script that performs an immediate `ifopen` so that your connectivity returns.

NiKiZe, 2024-04-20:
When iPXE starts the NICs are reset, bringing the link down.

In regards to being loaded to ram I would not make assumptions it's up to firmware.

But the issue here is probably reading file: after iPXE initialization, however this should be done before iPXE does any NIC reset.

Once ifopen/DHCP is done and bring the link up iDRAC should be able to inject and catch it's packets, but I don't think iPXE can do much here unless we resort to dirty hacks... But you should probably make sure that your script does an ifopen as soon as possible?

champtar, 2024-04-22:
Thanks both, I can either do an ifopen on all interfaces when using ipxe.efi, or use snp.efi, so this is ok for now

> not least because it's so difficult to test without access to the hardware in question

Even with access to Dell servers, they support multiple shared NICs (Intel / Broadcom, multiple models), so this would be a lot of work for little reward

closed 2024-04-22 by champtar