ipxe/ipxe #1199: Network interface ordering

champtar, 2024-04-22:
Not sure if this is expected, the interfaces are a bit out of order (both when using ipxe.efi and snp.efi)
NIC is `Broadcom NetXtreme Gigabit Ethernet (BCM5720)`
https://www.dell.com/en-sg/shop/dell-quad-port-1-gigabit-broadcom-5720-network-daughter-card-cuskit/apd/540-bbhg/wifi-and-networking

[image]
[image]

NiKiZe, 2024-04-22:
Just as in linux, the enumeration order is undefined. Specificly it enumerates drivers, and than each NIC for that driver.

Could you explain what kind of issue this is causing you?

champtar, 2024-04-22:
> Just as in linux, the enumeration order is undefined. Specificly it enumerates drivers, and than each NIC for that driver.

In Linux you have Consistent Network Device Naming so enumeration order doesn't matter that much
what surprises me here is that it doesn't scan PCI buses in order (19 ends up before 18)

> Could you explain what kind of issue this is causing you?

I was just expecting to have port 1 == net0, I can work around this,
I just wanted to be sure this is expected in this case (here we only have 1 driver)

NiKiZe, 2024-04-22:
In linux the order is defined by the order of loaded drivers, which is mostly alphabetical as long as you keep them in kernel and not modules... The (un)predictable network names just creates more problems and makes this very defined behaviour unpredictable between hosts /rant

Using snponly.efi when chained gives you one nic which it was booted from, other than that we can't be sure of ordering, you might be able to grab dmi data to get port designators and sort on that, but that would need to be in a script to not break any already existing assumptions on net ordering.

habetsm-xilinx, 2024-04-25:
The PCI enumeration order will also vary between different BIOS vendors.

closed 2024-09-24 by zube[bot]